celsworth/lxp-bridge #2: Standing ovation !

teicors, 2021-06-03:
GREAT JOB !!
But I need a little modification.. 
Basically my "farm" is composed with three inverters and I need the identification of the inverter inside the packet mqtt: I'm trying to modify your code, but I have no idea how fix it. Can you help me ?

celsworth, 2021-06-03:
Hey, thanks for the kind words :)

I must emphasise that I wasn't expecting anyone to be using this yet, its still very experimental.

Can you confirm what you need - every single MQTT packet should describe the inverter that sent it? I guess we could do this in the topics, maybe something like:

`lxp/$serial/hold/12` => `12`
`lxp/$serial/inputs/1` => `{ "soc": 96, ... }`

I assume you're running 3 copies of this, one connected to each inverter? Do you need to control them individually? Would it also be helpful to add the ability to specify the serial when sending commands, so you'd publish to something like `lxp/$serial/ac_charge => true` or whatever.

I can certainly work on this, give me a few days and I'm sure I'll have something knocked up.. I'll update this issue when I do :)

teicors, 2021-06-03:
Hi Chris,
I'm running (now) a single instance only for test purpose. My system is three-phase so I have three inverters and I figure out who is the "owner" of 
`{"max_chg_curr":66.0,"max_dischg_curr":66.0,"charge_volt_ref":53.5,"dischg_cut_volt":40.0,"bat_status_0":0,"bat_status_1":0,"bat_status_2":0,"bat_status_3":0,"bat_status_4":0,"bat_status_5":0,"bat_status_6":0,"bat_status_7":0,"bat_status_8":0,"bat_status_9":0,"bat_status_inv":0,"bat_count":0,"bat_capacity":0}`
Cheers,
Angelo

teicors, 2021-06-03:
Perfect ! :1st_place_medal: 
[2021-06-03T14:35:27.605 DEBUG rumqttc::state] Publish. Topic = lxp/XX1043999/inputs/1, Pkid = 2, Payload Size = 584
Now I can start to play with the mqttwarn, the db, etc...

celsworth, 2021-06-03:
Ok all done, quicker than I thought :)

PR #3 (2021-06-03): Use datalog in MQTT packet topics
Closes #2

This changes the outgoing MQTT topics from `lxp/hold/1` to `lxp/$datalog/hold/1`. This lets people with multiple inverters distinguish which inverter is sending the messages.
closed 2021-06-03 by celsworth