celsworth/lxp-bridge #7: Decode bit registers

celsworth, 2021-06-12:
Some of the outputs in `lxp/inputs/*` could be a bit more useful because they actually represent a range of values stored in individual bits.

Problem is I'm not sure what all of them mean.

`lxp/inputs/1 {status}` - I do actually have documentation for this one so I can decode all these no problem

## battery status bits

`lxp/inputs/3 {bat_status_*}` - these are largely unknown. All I know is that usually `bat_status_5=192` and `bat_status_inv=3`. This may just mean that the charging/discharging FETs are enabled.

A potentially useful datapoint is that when the BMS requested a full battery charge, these two *changed* to `bat_status_5=200` and `bat_status_inv=35`. I'll try documenting the bits below and if anyone has anything to add please do share.

### bat_status_5

```
1 << 7 = charging or discharging enabled? guessing as these are usually on for me
1 << 6 = charging or discharging enabled? guessing as these are usually on for me
1 << 5 = ?
1 << 4 = ?
1 << 3 = full charge request
1 << 2 = ?
1 << 1 = ?
1 << 0 = ?
```

### bat_status_inv

```
1 << 7 = ?
1 << 6 = ?
1 << 5 = full charge request
1 << 4 = ?
1 << 3 = ?
1 << 2 = ?
1 << 1 = charging or discharging enabled? guessing as these are usually on for me
1 << 0 = charging or discharging enabled? guessing as these are usually on for me
```

lupine, 2022-11-24:
Hi @celsworth 

Thanks for this project - I'm just getting set up with it after switching my inverter from a Solax to an LXP Hybrid (I'd actually put something similar-ish, only more rubbish, together for the Solax at https://code.ur.gs/lupine/solax2mqtt !)

I'm just converting my existing grafana dashboards over, and looking for documentation on the status field. Above, you said:

> lxp/inputs/1 {status} - I do actually have documentation for this one so I can decode all these no problem

Is there any chance you could share those docs / fill out the issue description with them? A feature of my grafana dashboard is value mapping the status from numeric to text values, so I could see at a glance if the inverter was "OK", "Waiting", "On fire", etc, and I've love to replicate that with the LXP too.

celsworth, 2022-11-24:
Hi, yep I've actually added them to the wiki recently, see here: https://github.com/celsworth/lxp-bridge/wiki/Inputs#status

I'll add this link above as well.

lupine, 2022-11-24:
Ah, amazing, thanks - that's exactly what I need.

Yes, I worked at Bytemark - I think I left shortly before you arrived. Small world!

lupine, 2022-11-25:
OK, I managed to get a reasonable status display into grafana, using the statusmap plugin: https://grafana.com/grafana/plugins/flant-statusmap-panel/?tab=installation . The existing values are being pushed (via the influx wire format) into VictoriaMetrics, whose MetricsQL has a `bitmap_and` function :sweat_smile: 

Here's my interpretation of each bit, with a zero value treated as `idle`:

[image]

The bitfield queries are 8 separate queries like:

```
clamp_max(bitmap_and(inputs_status, 0x04), 1)
```

(I'm treating the FAULT and GRID-OFF fields differently so I get red if they switch on, but that's a detail)

joshs85, 2023-08-18:
I think this defines everything you need: https://eg4electronics.com/wp-content/uploads/2023/06/EG4-18KPV-12LV-Modbus-Protocol.pdf

celsworth, 2023-08-18:
Yes! We've been discussing that document in various other issues actually. It renders a lot of my early guesswork and reverse engineering redundant, but it is nice that I got most of it right ;)

I'm actually surprised Signature Solar have published it so readily. Historically Luxpower have been extremely cagey about releasing this sort of information. It was impossible to get it for my LXP ACS3600 years ago when I bought it.

lupine, 2023-08-18:
That is my new favourite PDF :heart:

joshs85, 2023-08-23:
I'm especially interested in being able to set all of the bits in hold registers 21 and 110 via the gui.

celsworth, 2023-08-28:
> I'm especially interested in being able to set all of the bits in hold registers 21 and 110 via the gui.

All of them? Some of them are quite niche I thought and generally don't need tweaking once set for your region. I thought we'd covered most of the common ones like setting AC Charge on/off, Charge Priority, Force Discharge.

Is there any other particular ones you'd want as a matter of priority? I'm not so sure that adding the MQTT topics for, say, GFCI, or Neutral Detect, or Fast Zero Export, is really worth it?