celsworth/lxp-bridge #18: Stopping to retrieve the data

teicors, 2021-06-28:
Hi Chris,
for the second time, the software stopped to get the data from my inverter without errors!
I resumed the fault with a restart of the soft.
Inside the log there were nothing, but I noticed the error at midnight.

[2021-06-27T23:59:21.174 DEBUG rumqttc::state] Publish. Topic = lxp/BA1043xxx/inputs/3, Pkid = 33, Payload Size = 336
[2021-06-28T00:00:12.322 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 51145 millisecs,
            last outgoing request before 51147 millisecs
[2021-06-28T00:01:12.325 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 60000 millisecs,
            last outgoing request before 60002 millisecs
[....}
[2021-06-28T08:53:13.430 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59999 millisecs,
            last outgoing request before 60000 millisecs
[2021-06-28T08:54:13.432 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59999 millisecs,
            last outgoing request before 60000 millisecs
[2021-06-28T08:55:13.433 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59998 millisecs,
            last outgoing request before 60000 millisecs
[2021-06-28T08:56:13.436 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59999 millisecs,
            last outgoing request before 60001 millisecs
[2021-06-28T08:57:13.437 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59998 millisecs,
            last outgoing request before 60001 millisecs
[2021-06-28T08:58:13.440 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59998 millisecs,
            last outgoing request before 60001 millisecs
[2021-06-28T08:59:13.441 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59998 millisecs,
            last outgoing request before 60001 millisecs
[2021-06-28T09:00:13.444 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59999 millisecs,
            last outgoing request before 60001 millisecs
[2021-06-28T09:01:13.445 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59998 millisecs,
            last outgoing request before 60000 millisecs

celsworth, 2021-06-28:
Hi @teicors!

I actually put in a fix yesterday that *might* help this:

https://github.com/celsworth/lxp-bridge/commit/fc0e856da7b00c1dc8ca7991ebc07b5923841d55#diff-54bf69d35a5fb81ebde24c309b9c94c16747a015d731ff613a3cf1cbd3d375a3R411-R415

It still needs a bit more work but I did notice that it was possible for the loop that handles packets from the inverter to silently exit if it encountered a failure. This change fixes that temporarily by just logging the error instead.

Can you try the current master and see how that goes for you?

teicors, 2021-06-28:
got it!

celsworth, 2021-06-28:
If it happens again I have one other idea; it is possible that the TCP connection to the inverter is dropping but lxp-bridge never notices because it only sends to the inverter when you send it a command (with MQTT `lxp/cmd/..` topics). I should look at enabling TCP keepalives so we detect when the connection drops, then reconnect.

celsworth, 2021-06-28:
@teicors I've now added TCP keepalives in #20 so try current master, especially if it happens again.

celsworth, 2021-07-06:
I'm hoping this is fixed now I've released 0.2.0 so going to close, but feel free to reopen if it happens again - might have to add a bunch more debug logging if it does.

I've had the app running for days consistently without any loss of reception now so it's looking good :)

closed 2021-07-06 by celsworth

teicors, 2021-07-08:
Until now, no errors at all ...

teicors, 2021-07-14:
Here the proof of evidence that you found an right solution

[2021-07-12T00:02:58.517 ERROR lxp_bridge::lxp::inverter] inverter BA10439990: Connection reset by peer (os error 104)
[2021-07-12T00:02:58.519 INFO lxp_bridge::lxp::inverter] inverter BA10439990: reconnecting in 5s
[2021-07-12T00:03:03.522 INFO lxp_bridge::lxp::inverter] connecting to inverter BA10439990 at 10.10.20.243:8000
[2021-07-12T00:03:03.531 INFO lxp_bridge::lxp::inverter] inverter BA10439990: connected!
[2021-07-12T00:03:20.297 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59999 millisecs,
            last outgoing request before 60001 millisecs
[2021-07-12T00:03:43.551 ERROR lxp_bridge::lxp::inverter] inverter BA10439998: Connection reset by peer (os error 104)
[2021-07-12T00:03:43.552 INFO lxp_bridge::lxp::inverter] inverter BA10439998: reconnecting in 5s
[2021-07-12T00:03:45.598 ERROR lxp_bridge::lxp::inverter] inverter BA10439992: Connection reset by peer (os error 104)
[2021-07-12T00:03:45.599 INFO lxp_bridge::lxp::inverter] inverter BA10439992: reconnecting in 5s
[2021-07-12T00:03:48.554 INFO lxp_bridge::lxp::inverter] connecting to inverter BA10439998 at 10.10.20.241:8000
[2021-07-12T00:03:48.561 INFO lxp_bridge::lxp::inverter] inverter BA10439998: connected!
[2021-07-12T00:03:50.602 INFO lxp_bridge::lxp::inverter] connecting to inverter BA10439992 at 10.10.20.242:8000
[2021-07-12T00:03:50.607 INFO lxp_bridge::lxp::inverter] inverter BA10439992: connected!
[2021-07-12T00:04:20.299 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59999 millisecs,
            last outgoing request before 60001 millisecs
[2021-07-12T00:05:20.301 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59999 millisecs,
            last outgoing request before 60001 millisecs
[2021-07-12T00:06:20.302 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59999 millisecs,
            last outgoing request before 60000 millisecs
[2021-07-12T00:06:46.883 DEBUG lxp_bridge::lxp::packet_decoder] 19 bytes in: [161, 26, 2, 0, 13, 0, 1, 193, 66, 65, 49, 48, 52, 51, 48, 48, 50, 48, 0]
[2021-07-12T00:06:46.884 DEBUG lxp_bridge::lxp::inverter] inverter BA10439990: RX Heartbeat(Heartbeat { datalog: BA10430020 })
[2021-07-12T00:06:46.886 DEBUG lxp_bridge::coordinator] RX: Heartbeat(Heartbeat { datalog: BA10439990 })