celsworth/lxp-bridge #22: Add WriteParam functions

celsworth, 2021-08-01:


kaitlinsm, 2022-10-23:
Did i read that right - `param 0` is the cadence of publishing the stats from all the sensors in the inverter, and defaults to 5 minutes?

This will publish to both the local interface AND the luxpower server interface, so setting this to a lower value will cause problems with clogging up their servers?

celsworth, 2022-10-23:
Yes, exactly. When I first got my inverter, this register was set to 120 (2 minutes). Later changed to 3 minutes, then again changed to 5 minutes, so I think they've been decreasing the frequency of all inverters to reduce load on their servers over time.

If you did change this register to a low value then it could be a good idea to firewall your inverter off from their servers so it can't send data back. This would prevent their web interface working with your inverter at all though, of course, including firmware updates etc.. but some people might not think isolating their kit from China is a bad thing anyway :)

It's been on my list for some time to try changing this value, and isolating my own inverter from them, just to see if everything keeps working properly locally, just never got round to it yet.

kaitlinsm, 2022-10-23:
Yeah, I was aware that these massive chunks of stuff phone home before purchasing one, but kind of "oh well". I've noticed the increased duration between readings too and wasn't surprised, as I think a *lot* of these inverters are being sold in the UK at the moment, at least.

I've had a bit of a read through the code/docs in this repo but I'm not completely familiar with it - can I just confirm something?

In the docs, it seems that publishing 
```
lxp/cmd/{datalog}/read/inputs/1
lxp/cmd/{datalog}/read/inputs/2
lxp/cmd/{datalog}/read/inputs/3
```

should cause every avaiblable sensor value to be published in response, and anyone subscribed to the topics will get the update.

Will this also cause stuff to get sent to china, or is this a local mqtt-driven workaround? basically (sort of bash-esque)

```
while true; do
  mosquitto_pub --topic lxp/cmd/{datalog}/read/inputs/1
  sleep 1
  mosquitto_pub --topic lxp/cmd/{datalog}/read/inputs/2
  sleep 1;
  mosquitto_pub --topic lxp/cmd/{datalog}/read/inputs/3
  sleep 30;
done;
```

celsworth, 2022-10-23:
Yes that will cause the responses to get sent to China; it's a broadcast system whereby every connection to the inverter sees all the same responses, no matter which connection originally sent the request.

You can probably verify this by publishing those topics then checking the Data History log in the Lux web portal - more than likely they'll log it as an additional data row.

kaitlinsm, 2022-10-23:
Yep - verified that, that's put an additional data point in place. So doing this in a loop is probably a really bad idea!

PR #117 (2022-11-25): Add WriteParam functionality
This has been verified to work ~but the result topic is always a `FAIL`~ (now fixed) because WriteParam apparently doesn't return the new value of the register. It returns 0 on success and 1 on fail.

This is weird, if anything I could have understood it being the other way around, ie return 1 on success to mean 1 register has been updated, and 0 on failure to mean no registers were updated. 🤷 

I'm tempted to just do away with `result` topics, I doubt anyone uses them and wouldn't even notice. The preferred way to check if a set has succeeded should be to do a sub [... cut]
closed 2022-11-25 by celsworth