celsworth/lxp-bridge #24: New issue

teicors, 2021-08-23:
Hi Chris,
last week in the log raised an erorr:

Connection reset by peer (os error 104)

I resumed to catch the data from the inverter when a cronjob restarted the job at midnight.
No significant info about the error, only I found some "error 104" starting from the afternoon, but the night no one data was fetched into my database.

celsworth, 2021-08-29:
Hi teicors!

So just to confirm, when you saw the error 104 in your logs, lxp-bridge exited? It *should* catch this sort of error and simply reconnect. I'll see if I can work out why but I've not witnessed anything like this myself and I've had it running for weeks.

A mitigation is to start lxp-bridge using the systemd script so it restarts automatically when it exits, an example is provided at https://github.com/celsworth/lxp-bridge/blob/master/systemd/lxp-bridge.service

teicors, 2021-08-30:
Hi Chris,
the lxp-bridge was still running: as you suppose the "error 104" was in the log and the soft still running and I saw (unfortunately I lost the log) trying to reconnect the inverters, but without luck: the issue started at 18;52 and around 00:30 the soft can reconnect the invrters. Before I said a cronjob, but it wasn't true, the software itself was able to reconnect.
The same tonight, but only an error and after the lxp was able to reconnect the inverters. 
No more info, sorry !

celsworth, 2021-08-31:
Can you see if Luxpower's own portal was logging data for the inverter during the time period that lxp-bridge couldn't reconnect?

It sounds to me like lxp-bridge is doing everything right - when the connection is lost it tries to reconnect until the inverter is back. That can happen sometimes, it's only WiFi after all.. if it happens often maybe a WiFi repeater nearer your inverter might help.

If you manage to see the logs while lxp-bridge can't reconnect then checking if you can even ping the inverter IP or see if Luxpower's portal is logging data still, might be useful. My guess is the inverter is dropping off your LAN entirely :/

teicors, 2021-08-31:
Sorry !!
As you suggested, I checked if the data on Luxpower portal was present.... and also on they portal are missed (from 16:59 to 22:57 utc)! So I supposing a general failure of the interface.
Your sw works as expected ! ;)

celsworth, 2021-08-31:
👌

closed 2021-08-31 by celsworth