celsworth/lxp-bridge #41: Crashes with "Channel closed" after a few MQTT packets sent

m1geo, 2022-02-18:
Possibly related to #30

I see lxp-bridge crash with "Channel closed" after one or two MQTT packets, exit-code 255.

```
george@ubuntu-vm:~/lxp-bridge$ /home/george/lxp-bridge/target/debug/lxp-bridge -c /home/george/lxp-bridge/config.yaml
[2022-02-18T05:17:58.882 INFO lxp_bridge] lxp-bridge 0.6.0-dev starting
[2022-02-18T05:17:58.883 INFO lxp_bridge::lxp::inverter] connecting to inverter 2222222222 at 192.168.0.4:8000
[2022-02-18T05:17:58.883 INFO lxp_bridge::mqtt] initializing mqtt at localhost:1883
[2022-02-18T05:17:58.886 INFO lxp_bridge::influx] influx disabled, skipping
[2022-02-18T05:17:58.887 DEBUG rumqttc::state] Subscribe. Topics = [Filter = lxp/cmd/all/#, Qos = AtMostOnce], Pkid = 1
[2022-02-18T05:17:58.887 DEBUG rumqttc::state] Subscribe. Topics = [Filter = lxp/cmd/2222222222/#, Qos = AtMostOnce], Pkid = 2
[2022-02-18T05:17:58.892 INFO lxp_bridge::lxp::inverter] inverter 2222222222: connected!
[2022-02-18T05:18:58.888 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 60000 millisecs,
            last outgoing request before 60000 millisecs
[2022-02-18T05:19:58.888 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 59999 millisecs,
            last outgoing request before 60000 millisecs
[2022-02-18T05:20:58.890 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 60001 millisecs,
            last outgoing request before 60001 millisecs
[2022-02-18T05:21:02.036 DEBUG lxp_bridge::lxp::packet_decoder] 19 bytes in: [removed]
[2022-02-18T05:21:02.036 DEBUG lxp_bridge::lxp::inverter] inverter 2222222222: RX Heartbeat(Heartbeat { datalog: 2222222222 })
[2022-02-18T05:21:02.036 DEBUG lxp_bridge::coordinator] RX: Heartbeat(Heartbeat { datalog: 2222222222 })
[2022-02-18T05:21:03.423 DEBUG lxp_bridge::lxp::packet_decoder] 117 bytes in: [removed]
[2022-02-18T05:21:03.423 DEBUG lxp_bridge::lxp::inverter] inverter 2222222222: RX TranslatedData(TranslatedData { datalog: 2222222222, device_function: ReadInput, inverter: 5555555555, register: 0, values: [removed]] })
[2022-02-18T05:21:03.423 DEBUG lxp_bridge::coordinator] RX: TranslatedData(TranslatedData { datalog: 2222222222, device_function: ReadInput, inverter: 5555555555, register: 0, values: [removed] })
[2022-02-18T05:21:03.424 DEBUG lxp_bridge::mqtt] publishing: lxp/2222222222/inputs/1 = {"status":32,"v_pv":26.4,"v_pv_1":16.2,"v_pv_2":9.8,"v_pv_3":0.4,"v_bat":53.5,"soc":98,"soh":100,"p_pv":0,"p_pv_1":0,"p_pv_2":0,"p_pv_3":0,"p_charge":60,"p_discharge":0,"v_ac_r":243.4,"v_ac_s":25.7,"v_ac_t":0.0,"f_ac":50.13,"p_inv":0,"p_rec":66,"pf":1.0,"v_eps_r":243.4,"v_eps_s":4015.3,"v_eps_t":10.6,"f_eps":50.13,"p_to_grid":0,"p_to_user":8067,"e_pv_day":0.0,"e_pv_day_1":0.0,"e_pv_day_2":0.0,"e_pv_day_3":0.0,"e_inv_day":0.3,"e_rec_day":6.4,"e_chg_day":6.1,"e_dischg_day":0.4,"e_eps_day":0.0,"e_to_grid_day":0.0,"e_to_user_day":45.9,"v_bus_1":378.9,"v_bus_2":324.6,"time":1645161663,"datalog":"2222222222"}
[2022-02-18T05:21:03.424 DEBUG rumqttc::state] Publish. Topic = lxp/2222222222/inputs/1, Pkid = 3, Payload Size = 610
[2022-02-18T05:21:04.921 DEBUG lxp_bridge::lxp::packet_decoder] 117 bytes in: [removed]
[2022-02-18T05:21:04.921 DEBUG lxp_bridge::lxp::inverter] inverter 2222222222: RX TranslatedData(TranslatedData { datalog: 2222222222, device_function: ReadInput, inverter: 5555555555, register: 40, values: [removed] })
[2022-02-18T05:21:04.921 DEBUG lxp_bridge::coordinator] RX: TranslatedData(TranslatedData { datalog: 2222222222, device_function: ReadInput, inverter: 5555555555, register: 40, values: [removed] })
[2022-02-18T05:21:04.921 DEBUG lxp_bridge::mqtt] publishing: lxp/2222222222/inputs/2 = {"e_pv_all":3437.3,"e_pv_all_1":1837.8,"e_pv_all_2":1599.5,"e_pv_all_3":0.0,"e_inv_all":3832.3,"e_rec_all":1491.9,"e_chg_all":2478.1,"e_dischg_all":2075.2,"e_eps_all":0.0,"e_to_grid_all":853.6,"e_to_user_all":6116.1,"t_inner":32,"t_rad_1":19,"t_rad_2":20,"t_bat":0,"runtime":25708850,"time":1645161664,"datalog":"2222222222"}
[2022-02-18T05:21:04.922 DEBUG rumqttc::state] Publish. Topic = lxp/2222222222/inputs/2, Pkid = 4, Payload Size = 325
[2022-02-18T05:21:06.441 DEBUG lxp_bridge::lxp::packet_decoder] 117 bytes in: [removed]
[2022-02-18T05:21:06.441 DEBUG lxp_bridge::lxp::inverter] inverter 2222222222: RX TranslatedData(TranslatedData { datalog: 2222222222, device_function: ReadInput, inverter: 5555555555, register: 80, values: [removed] })
[2022-02-18T05:21:06.441 DEBUG lxp_bridge::coordinator] RX: TranslatedData(TranslatedData { datalog: 2222222222, device_function: ReadInput, inverter: 5555555555, register: 80, values: [removed] })
[2022-02-18T05:21:06.442 ERROR lxp_bridge] channel closed
george@ubuntu-vm:~/lxp-bridge$ echo $?
255
george@ubuntu-vm:~/lxp-bridge$
```

Using mosquito_sub to watch MQTT, I see:

```
george@ubuntu-vm:~/lxp-bridge$ mosquitto_sub -t lxp/2222222222/inputs/+
{"status":32,"v_pv":26.4,"v_pv_1":16.2,"v_pv_2":9.8,"v_pv_3":0.4,"v_bat":53.5,"soc":98,"soh":100,"p_pv":0,"p_pv_1":0,"p_pv_2":0,"p_pv_3":0,"p_charge":60,"p_discharge":0,"v_ac_r":243.4,"v_ac_s":25.7,"v_ac_t":0.0,"f_ac":50.13,"p_inv":0,"p_rec":66,"pf":1.0,"v_eps_r":243.4,"v_eps_s":4015.3,"v_eps_t":10.6,"f_eps":50.13,"p_to_grid":0,"p_to_user":8067,"e_pv_day":0.0,"e_pv_day_1":0.0,"e_pv_day_2":0.0,"e_pv_day_3":0.0,"e_inv_day":0.3,"e_rec_day":6.4,"e_chg_day":6.1,"e_dischg_day":0.4,"e_eps_day":0.0,"e_to_grid_day":0.0,"e_to_user_day":45.9,"v_bus_1":378.9,"v_bus_2":324.6,"time":1645161663,"datalog":"2222222222"}
{"e_pv_all":3437.3,"e_pv_all_1":1837.8,"e_pv_all_2":1599.5,"e_pv_all_3":0.0,"e_inv_all":3832.3,"e_rec_all":1491.9,"e_chg_all":2478.1,"e_dischg_all":2075.2,"e_eps_all":0.0,"e_to_grid_all":853.6,"e_to_user_all":6116.1,"t_inner":32,"t_rad_1":19,"t_rad_2":20,"t_bat":0,"runtime":25708850,"time":1645161664,"datalog":"2222222222"}
```

Relevant parts of my config:

```
george@ubuntu-vm:~/lxp-bridge$ cat config.yaml
inverters:
- enabled: true
  host: 192.168.0.4
  port: 8000
  serial: 5555555555
  datalog: 2222222222

mqtt:
  enabled: true
  host: localhost
  port: 1883
  username:
  password:
  namespace: lxp
  homeassistant:
    enabled: false
    prefix: homeassistant

influx:
  enabled: false
  url: http://localhost:8086
  username:
  password:
  database: lxp
george@ubuntu-vm:~/lxp-bridge$

```

Finally, the mosquito (MQTT broker) log doesn't say anything useful other than "socket error" around 2 minutes after connecting:

```
1645161243: New connection from 127.0.0.1 on port 1883.
1645161243: New client connected from 127.0.0.1 as lxp-bridge (p2, c1, k60).
1645161366: Socket error on client lxp-bridge, disconnecting.
```

Happy to supply more details or unedited logs if helpful. Thanks!

celsworth, 2022-02-18:
Hi!

Thanks for the report. This seems to be related to Influx being disabled as I can reproduce if I do that. I'll see if I can work it out later today.

celsworth, 2022-02-18:
https://github.com/celsworth/lxp-bridge/pull/42 should fix this. There should be some binaries at https://github.com/celsworth/lxp-bridge/actions/runs/1863429792 shortly if you want to try it out. Or just git pull the [branch](https://github.com/celsworth/lxp-bridge/tree/fix-skipping-disabled-services) and rebuild.

davewinwood, 2022-02-18:
Hi Chris, I've been getting crashes also, so I'd like to update my system when this branch is merged back into the main branch. I would try and access it from the branch but can't seem to work out how to do that. I'm running on a dedicated RPi4 just for this item. It works well but I do have to keep manually restarting it when it fails. Great work though.

//Dave

m1geo, 2022-02-18:
> Hi Chris, I've been getting crashes also, so I'd like to update my system when this branch is merged back into the main branch. I would try and access it from the branch but can't seem to work out [...]

@davewinwood, you should just be able to go into the checked out folder and type:
> git pull
> git checkout fix-skipping-disabled-services

That'll get you the fix branch. Alternatively, just wait until it gets merged into main branch.

celsworth, 2022-02-18:
It's merged, thanks all. I'll get a new release out with that fix.

davewinwood, 2022-02-18:
Thanks Chris 😀