celsworth/lxp-bridge #43: Add mysql Support

zatie12, 2022-03-19:
Would be very useful, thanks for a great project.

celsworth, 2022-03-19:
Hi @zatie12, thanks for the suggestion :) Do you mean populating a table with the values, log-style? So adding a new row every 5 minutes with all the readings?

I think that sounds pretty doable, I'll see what I can do!

zatie12, 2022-03-19:
Hi @celsworth thanks for the response. Yes that would be ideal, all readings logged every 5 minutes to a table for long term archival and query purposes.

Thanks very much!

closed 2022-03-21 by celsworth

celsworth, 2022-03-24:
Hi @zatie12 just a little update here, after https://github.com/celsworth/lxp-bridge/pull/47 you'll need to change your config slightly as I added support for multiple databases.

So instead of this:

```
database:
  enabled: true
  url: mysql://foo
```

You now want this:

```
databases:
- enabled: true
  url: mysql://foo
```

Sorry about that but I expect you're the only one using this so far so hopefully you don't mind the change :) Better now than after a release!

This just means you can now log to multiple databases which could be handy for backup databases and stuff.

PR #44 (2022-03-21): Add database support
Adds support for storing input data to a database.

Postgres, MySQL, and SQLite are all supported.

Closes #43