celsworth/lxp-bridge #50: Documentation discrepancies

IMgoRt, 2022-04-02:
It is stated that:
`topic = lxp/cmd/{datalog}/set/ac_charge, payload = boolean`
forces the invertor to charge the batteries

However this actually only sets the AC charging flag to enabled (or disabled), if there are charge times set then the batteries do not charge until those active periods

celsworth, 2022-04-03:
Thanks for pointing this out, that's a good thing to explain better in the docs. Hopefully this helps - https://github.com/celsworth/lxp-bridge/commit/7c6d964d6bfcd83a9007cda24296547d0c17be90

I should probably think about reorganising all the docs as the README is getting pretty unwieldy but that's a good start I hope :)

closed 2022-04-03 by celsworth

IMgoRt, 2022-04-04:
> Thanks for pointing this out, that's a good thing to explain better in the docs. Hopefully this helps - [7c6d964](https://github.com/celsworth/lxp-bridge/commit/7c6d964d6bfcd83a9007cda24296547d0c17be90)
> 
> I should probably think about reorganising all the docs as the README is getting pretty unwieldy but that's a good start I hope :)

Thank you Chris, that should clear up any confusion