celsworth/lxp-bridge #52: Sample docker-compose.yml

IMgoRt, 2022-04-04:
Hi Chris,
For those of us who prefer a docker-compose.yml, the following replicates your command:

```
version: '3.4'

services:
  lxp-bridge:
    container_name: lxp-bridge:latest
    image: celsworth/lxp-bridge
    restart: unless-stopped
    init: true
    volumes:
    - type: bind
      source: ${PWD}/config.yaml
      target: /etc/config.yaml 
```

celsworth, 2022-04-04:
👍 thanks, added and referenced in the README in https://github.com/celsworth/lxp-bridge/commit/f540eb8bc94c1a60848694e7218f40f94ee54600 :)

closed 2022-04-04 by celsworth

celsworth, 2023-02-02:
Hi @IMgoRt!

I'm not sure I understand - the example docker-compose is still in the repo, and the content of this part of the the README has moved to https://github.com/celsworth/lxp-bridge/wiki/Getting-Started

Can you elaborate exactly whats missing? Its very possible I did miss something when writing the wiki!

IMgoRt, 2023-02-02:
> Hi @IMgoRt!
>
> I'm not sure I understand - the example docker-compose is still in the repo, and the content of this part of the the README has moved to [...]

Sorry Chris

I was looking for the items now contained in https://github.com/celsworth/lxp-bridge/wiki/MQTT

celsworth, 2023-02-02:
👍 so long as you found what you were looking for :)