celsworth/lxp-bridge #53: Home Assistant Setup?

chriscn, 2022-04-12:
Would it be possible with some help in getting the values into Home Assistant through MQTT? Could we have an option for autodiscovery?

davewinwood, 2022-04-12:
Hi,

I am successfully using lxp bridge with HA. I'd be happy to assist in getting yours working. I am having issues with the ac charge on off as I don't get a feedback message topic but it doesn't stop it working open ended. All the energy data gets populated which was my main aim and now I'm starting to adust ac charge timing depending on solar forecast etc.

Dave Winwood

celsworth, 2022-04-12:
> Would it be possible with some help in getting the values into Home Assistant through MQTT? Could we have an option for autodiscovery?

autodiscovery is already supported, but note that the messages that populate HA are only sent on lxp-bridge restart. The relevant parts are under `mqtt` / `homeassistant`. So you'll need to configure it like so:

```
mqtt:
  enabled: true
  host: localhost
  port: 1883
  username:
  password:
  homeassistant:
    enabled: true
    prefix: homeassistant
```

Does that work?

I should really do a better job of documenting all this, maybe its time to start a Github Wiki instead of cramming more stuff into the README.

chriscn, 2022-04-12:
After a lot of fiddling around, it wasn't working but my issue is that the `datalog:` value was set to the wrong value and the values werent populating.

Some wiki would be good.

closed 2022-04-12 by chriscn