celsworth/lxp-bridge #54: Multi platform build with `docker buildx`

chriscn, 2022-04-12:
Would it be possible to add `armv7` to your list of build options? So that I can run this on a Raspberry Pi.

celsworth, 2022-04-13:
Funnily enough, I tried messing around with that recently: https://github.com/celsworth/lxp-bridge/actions/runs/2083226430

Try the [arm-unknown-linux-gnueabihf](https://github.com/celsworth/lxp-bridge/suites/5907867605/artifacts/201423876) build from that page?

celsworth, 2022-04-13:
If it's a docker container specifically you want, I did try experimenting with that but ran into various bugs which meant an armv7 build didn't work. I did manage to get arm64 working but it took *2 hours* of CI time so I don't think its much of a goer anyway.

There's other options I can explore like cross-compiling (as is done for the above arm build) and then injecting that binary into a docker container and just tagging it with a manifest as arm. If that would be useful then I can give that a go at some point. Might be a few days.

celsworth, 2022-04-13:
Just noticed that the above might be 32-bit (armv6?) If so, try the armv7 from here: https://github.com/celsworth/lxp-bridge/actions/runs/2159537623

celsworth, 2022-04-14:
Just a smal update here - I've got multi-arch builds going, sort of. There's a test at https://hub.docker.com/layers/202743378/celsworth/lxp-bridge/test/images/sha256-6a8f4162d7dd3ef213052a3eda306a7bfd9d39a92f778346f480792cba125f91?context=repo

So if you'd like to try: `docker pull celsworth/lxp-bridge:test` and it should work on a Pi now.

chriscn, 2022-04-14:
Getting this error.
```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 1, kind: PermissionDenied, message: "Operation not permitted" }', library/std/src/sys/unix/time.rs:353:62
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
I've been able to install this using the systemd

celsworth, 2022-04-14:
huh thats unexpected. Not sure whats going on there, but the error does indicate the binary is running so the compilation looks good :)

I'll have to see if I can reproduce that to debug it, what hardware is this, and what OS/docker version?

celsworth, 2022-04-15:
Seems to be: https://users.rust-lang.org/t/chrono-strange-panic-permission-denied/68867/5

It's just not able to get the current time, are your containers particularly locked down? This thread suggests privileged mode but that seems a bit overkill, maybe there's some permission bit that can be toggled..

chriscn, 2022-04-15:
> I'll have to see if I can reproduce that to debug it, what hardware is this, and what OS/docker version?

Docker version 20.10.12, build e91ed57

Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

> It's just not able to get the current time, are your containers particularly locked down? This thread suggests privileged mode but that seems a bit overkill, maybe there's some permission bit that can be toggled..

As far as I am aware this is just a standard docker install, `docker run hello-world` seems to be running succesfully. I've tried it on another Raspberry Pi and getting the same result. 

It is on a Raspberry Pi at my father's house so isn't mission criticial since I can run the binary. Although for the future fixing the build would be good for the general user. :-)

celsworth, 2022-04-16:
I'll repurpose this issue into a bug to track the above error, and try to reproduce it. I have a couple of Pis lying about so should be able to knock something up.

celsworth, 2022-06-25:
Took a look at this this morning, but I'm not able to reproduce so far. Do you get the error immediately on starting it or does it have to do something in particular to trigger it?

Here's my diagnosis so far:

Raspbian Buster 32bit from `2022-04-04-raspios-bullseye-armhf-lite.img` installed on a Pi2, default configs, no tweaking except to enable SSH

```
chris@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux
```

Installed Docker with:

```
curl -fsSL https://get.docker.com -o get-docker.sh
sudo -s
sh get-docker.sh
```

```
root@raspberrypi:/home/chris# docker version

Client: Docker Engine - Community
 Version:           20.10.17
 API version:       1.41
[snip]

Server: Docker Engine - Community
 Engine:
  Version:          20.10.17
  API version:      1.41 (minimum version 1.12)
[snip]
```

```

root@raspberrypi:/home/chris# docker run --rm -it celsworth/lxp-bridge:test
Unable to find image 'celsworth/lxp-bridge:test' locally
test: Pulling from celsworth/lxp-bridge
a6b2963233cd: Pull complete
13cf88150845: Pull complete
b230fcc4eef7: Pull complete
Digest: sha256:2dab00831a9f474e9c69a7b447d3365874a53bb42bc0c223f6a2776b48151150
Status: Downloaded newer image for celsworth/lxp-bridge:test
[2022-06-25T08:35:34.343 INFO lxp_bridge] lxp-bridge 0.7.0-dev starting
[2022-06-25T08:35:34.345 ERROR lxp_bridge] error reading /etc/config.yaml: No such file or directory (os error 2)
```

At this stage that's basically what I'd expect since I didn't give it a config but it's gotten the correct time (albeit in UTC not BST but I haven't told docker my timezone so again expected).

It even works as my unprivileged user (after putting myself in the `docker` group):

```
chris@raspberrypi:~ $ sudo usermod -aG docker chris
<reboot>
chris@raspberrypi:~ $ docker run --rm celsworth/lxp-bridge:test
[2022-06-25T08:40:26.393 INFO lxp_bridge] lxp-bridge 0.7.0-dev starting
[2022-06-25T08:40:26.420 ERROR lxp_bridge] error reading /etc/config.yaml: No such file or directory (os error 2)
```

So might need more to go on with this 🤔

celsworth, 2022-06-25:
I've updated the docker `test` tag with current development build just on the offchance you'd like to try it again

celsworth, 2022-07-04:
I've now also tried this on a Pi4 running aarch64 and the recently released v0.7.0 which now has the multi-arch images

```
root@raspberrypi:~# uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

root@raspberrypi:~# docker version
Client: Docker Engine - Community
 Version:           20.10.17
 API version:       1.41
 Go version:        go1.17.11
 Git commit:        100c701
 Built:             Mon Jun  6 23:02:34 2022
 OS/Arch:           linux/arm64
 Context:           default
 Experimental:      true

Server: Docker Engine - Community
 Engine:
  Version:          20.10.17
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.17.11
  Git commit:       a89b842
  Built:            Mon Jun  6 23:01:01 2022
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.6.6
  GitCommit:        10c12954828e7c7c9b6e0ea9b0c02b01407d3ae1
 runc:
  Version:          1.1.2
  GitCommit:        v1.1.2-0-ga916309
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
root@raspberrypi:~#

root@raspberrypi:~# docker run --rm -it celsworth/lxp-bridge
[2022-07-04T11:26:00.511 INFO lxp_bridge] lxp-bridge 0.7.0 starting
[2022-07-04T11:26:00.511 ERROR lxp_bridge] error reading /etc/config.yaml: No such file or directory (os error 2)
```

I think I have to give up here unless I can get more detail on how this is happening.

closed 2022-07-04 by celsworth