celsworth/lxp-bridge #56: Ability to choose which sensors to expose to Home Assistant

chriscn, 2022-04-14:
Currently 24 sensors are exposed which is great and gives the end user a lot of information however the future of this project may be a nice bridge from the Inverter to Home Assistant specifically for the Energy feature. This could be slimed down to 3 or 4 sensors. I know for my use case I don't need the individual string for the array since I only have one.

Let me know your thoughts @celsworth I'm not good with rust but there may be a way to dynamically send them to Home Assistant?

celsworth, 2022-04-15:
Hi :)

I have to confess I'm not a user of HA, this feature was added by request, so I'm not all that familiar with the specifics of this. My understanding though is even if lxp-bridge is presenting 24 sensors to HA, until you add them to dashboards and suchlike, they're mostly "hidden" away and easy to ignore if you want to - so with that in mind how valuable would this ability to choose which get exposed be? I'm not sure I see a great benefit?

However that said I'm not opposed to doing it, my main concerns would be simplicity of configuration and maintaining backwards compatibility with older configurations. My first idea is something like:

```
mqtt:
  homeassistant:
    enabled: true
    sensors: e_pv_all, v_bat, soc
```

So here the new `sensors` is a comma separated string of keys to announce to HA. By default (if this key is missing) then it would announce all of them which covers the backwards compatibility.

Does that sound like something that would fit your needs? And as mentioned above would be grateful to hear why its needed!

celsworth, 2022-04-15:
Experimental branch here: https://github.com/celsworth/lxp-bridge/pull/58

Implemented as described above. You can use `all` to get the default behaviour or omit the new sensors key entirely. Otherwise, list the keys you want. Note there's no sanity checking of the keys so it won't warn you about typos or anything, they'll be ignored. No spaces between commas either (though I'll strip those shortly, easy improvement to make).

chriscn, 2022-04-15:
Thank you so much for this, if you enable the wiki I'll start putting together some information on a Home Assistant setup since I believe that is where large percentage of the usecase for this will come from. As mentioned you don't use Home Assistant but thought you might like to see an example of what iis possible and was super easy to setup with your software so thank you once again. 

[image]
[image]

PR #58 (2022-04-16): Allow selecting which sensors are announced to HA
Closes #56
closed 2022-04-16 by celsworth