celsworth/lxp-bridge #59: `sudo make install` compare and strip are mutually exclusive

chriscn, 2022-04-15:
First, sorry for all the issues.
```shell
pi@pi-mqtt:~/lxp-bridge $ make
    Finished dev [unoptimized + debuginfo] target(s) in 2.19s
pi@pi-mqtt:~/lxp-bridge $ make install
    Finished dev [unoptimized + debuginfo] target(s) in 0.50s
install: options --compare (-C) and --strip are mutually exclusive
Try 'install --help' for more information.
make: *** [Makefile:10: install] Error 1
```

Also getting the issue when running `sudo make install`
```shell
sudo make install
make: cargo: Command not found
make: *** [Makefile:4: build] Error 127
```

Running `sudo install -C target/debug/lxp-bridge /usr/local/bin` works fine and installs it in the location. Not sure where the Makefile is failing.

celsworth, 2022-04-15:
Ah oops. The `Makefile` is kinda new, I'd barely tested it. Looks like I'll just remove the `-C`. Also hadn't thought about the issue with `sudo`, `cargo` isn't in the path since it usually lives in `~/.cargo/bin` I guess.

Maybe I'm best just removing the `Makefile` and sticking to basic cargo instructions in the README? Basically just revert some of https://github.com/celsworth/lxp-bridge/commit/966d99e72792a7180a67bf62846e0bbfc1b16927

chriscn, 2022-04-15:
I think instructions in the READMEs are good, would you be able to enable the Wiki and I can start porting some documentation to there if you'd like? Personally I use docker for everything since it is just the way the worlds going now. 

No worries it is a cracking bit of software which I am very much a fan of 😀

celsworth, 2022-04-15:
Done and opened up editing to everyone for now :)

chriscn, 2022-04-15:
> I think the README should be much slimmer, basically more of a "how to get started" then refer to the wiki for pretty much everything else. That'll be an ongoing project, I'll start moving stuff over soon.

Completely agree, what do you think is the best way of distrubuting the project? I would argue that Docker is often the easiest but does require fixing #54

celsworth, 2022-06-13:
Closing this as the original issue with `make` is fixed

closed 2022-06-13 by celsworth