celsworth/lxp-bridge #64: TranslatedData::decode mismatch

KhyalKara, 2022-06-12:
I'm getting the following error `TranslatedData::decode mismatch`, when receiving from inputs every five minutes.

```sh
[2022-06-12T11:52:51.592 DEBUG lxp_bridge::lxp::packet_decoder] 19 bytes in: [161, 26, 5, 0, 13, 0, 1, 193, 66, 65, 49, 49, 53, 48, 48, 52, 56, 52, 0]
[2022-06-12T11:52:51.593 DEBUG lxp_bridge::lxp::inverter] inverter <SN>: RX Heartbeat(Heartbeat { datalog: <SN> })
[2022-06-12T11:52:51.593 DEBUG lxp_bridge::coordinator] RX: Heartbeat(Heartbeat { datalog: <SN> })
[2022-06-12T11:52:52.820 DEBUG lxp_bridge::lxp::packet_decoder] 291 bytes in: [161, 26, 5, 0, 29, 1, 1, 194, 66, 65, 49, 49, 53, 48, 48, 52, 56, 52, 15, 1, 1, 4, 50, 48, 52, 51, 48, 50, 50, 52, 48, 49, 0, 0, 254, 16, 0, 0, 0, 0, 0, 121, 15, 247, 1, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 74, 9, 0, 0, 0, 0, 141, 19, 0, 0, 0, 0, 120, 0, 0, 0, 73, 9, 212, 1, 193, 124, 140, 19, 0, 0, 0, 0, 0, 0, 8, 2, 0, 0, 0, 0, 0, 0, 0, 0, 9, 0, 8, 0, 1, 0, 0, 0, 0, 0, 55, 0, 118, 15, 232, 13, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 232, 1, 0, 0, 177, 1, 0, 0, 61, 0, 0, 0, 41, 0, 0, 0, 0, 0, 0, 0, 217, 10, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 24, 0, 25, 0, 39, 0, 24, 0, 0, 0, 75, 52, 30, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 34, 0, 0, 0, 136, 19, 23, 2, 0, 0, 0, 0, 1, 8, 0, 16, 16, 220, 255, 42, 48, 18, 255, 171, 14, 131, 240, 96, 157, 16, 2, 0, 1, 0, 50, 0, 76, 255, 0, 0, 245, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3, 16, 246, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 1, 0, 0, 50, 48, 52, 51, 48, 50, 50, 52, 48, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 179, 19]
[2022-06-12T11:52:52.821 ERROR lxp_bridge::lxp::inverter] inverter <SN>: TranslatedData::decode mismatch: values.len()=255, value_length_byte=2
[2022-06-12T11:52:52.821 INFO lxp_bridge::lxp::inverter] inverter <SN>: reconnecting in 5s
[2022-06-12T11:52:57.826 INFO lxp_bridge::lxp::inverter] connecting to inverter <SN> at 192.168.0.101:8000
[2022-06-12T11:52:57.836 INFO lxp_bridge::lxp::inverter] inverter <SN>: connected!
[2022-06-12T11:53:20.442 DEBUG rumqttc::state] Pingreq,
            last incoming packet before 60001 millisecs,
            last outgoing request before 60001 millisecs
```

Any suggestions ? Thx

celsworth, 2022-06-12:
Hey, thanks for the report. On first glance this looks very interesting; the third integer in your debug log is `5` which is the protocol number; this is new on me, so far I've only seen protocols 1 and 2. So this could be something new to handle. I'll have a dig into this later and see if I can make it work.

KhyalKara, 2022-06-12:
Thanks for the quick reply. I'm very interested in this project, unfortunately my rust knowledge is subpar. If you need anymore logs or other information let me know.

celsworth, 2022-06-12:
Hey, so I'm working on decoding this and what would be really handy is a screenshot or CSV dump of the values from the LuxPower Web Portal from the same time as this packet was sent. So by that I mean after you login to the Lux web interface, then click Data at the top, then History Data, you get a table of values. Hopefully there's one that corresponds to the packet above at 11:52am this morning. It would be super helpful to get that row of data, because it means I can marry up what I'm seeing in the decoded data, with what it actually is.

I think I've gotten most of the way there, but I'm seeing some slightly odd things, for example..

```
"t_inner":24,"t_rad_1":25,"t_rad_2":39,"t_bat":24
```

That's unusual because the inner temp is usually higher than the external radiators. And unless you have a lead acid battery connected, then that shouldn't be reading anything but its at 24c as well. Also I see a voltage of 396V on your PV_3 string, but no power coming from it. Things like that. So in order to check I'm decoding this properly, being able to see what the values *should* be is an immense help.

Feel free to email them directly to me if you don't want to post here, chris@cae.me.uk :)

celsworth, 2022-06-13:
Hey,

Thanks for the extra info over email! I have a branch you can test if you're happy to build from git over in #65. There should be some binaries courtesy of Github's CI attached to it shortly as well. It's a WIP and I want to do a lot of tidying up but it should at least not crash now and give you some output data over MQTT/Influx for starters.

KhyalKara, 2022-06-13:
Thanks Chris.
Seems to be working perfectly.

```sh
{"status":16,"v_pv":398.7,"v_pv_1":0.0,"v_pv_2":0.0,"v_pv_3":398.7,"v_bat":50.7,"soc":100,"soh":0,"p_pv":0,"p_pv_1":0,"p_pv_2":0,"p_pv_3":0,"p_charge":0,"p_discharge":0,"v_ac_r":226.5,"v_ac_s":0.0,"v_ac_t":0.0,"f
_ac":49.91,"p_inv":0,"p_rec":0,"pf":0.0,"v_eps_r":226.2,"v_eps_s":46.8,"v_eps_t":3193.7,"f_eps":49.84,"p_to_grid":0,"p_to_user":622,"e_pv_day":0.0,"e_pv_day_1":0.0,"e_pv_day_2":0.0,"e_pv_day_3":0.0,"e_inv_day":0.0,"e_rec_day":0.9,"e_chg_d
ay":0.7,"e_dischg_day":0.1,"e_eps_day":0.0,"e_to_grid_day":0.0,"e_to_user_day":5.1,"v_bus_1":398.4,"v_bus_2":358.4,"e_pv_all":0.0,"e_pv_all_1":0.0,"e_pv_all_2":0.0,"e_pv_all_3":0.0,"e_inv_all":0.0,"e_rec_all":50.6,"e_chg_all":44.8,"e_disc
hg_all":6.2,"e_eps_all":4.1,"e_to_grid_all":0.0,"e_to_user_all":290.1,"t_inner":24,"t_rad_1":23,"t_rad_2":35,"t_bat":24,"runtime":2062817,"max_chg_curr":0.0,"max_dischg_curr":50.0,"charge_volt_ref":53.5,"dischg_cut_volt":0.0,"bat_status_0
":0,"bat_status_1":2049,"bat_status_2":4096,"bat_status_3":56336,"bat_status_4":11007,"bat_status_5":4656,"bat_status_6":44031,"bat_status_7":33550,"bat_status_8":24816,"bat_status_9":4253,"bat_status_inv":2,"bat_count":1,"bat_capacity":5
0,"bat_current":-2.0,"bms_event_1":0,"bms_event_2":505,"max_cell_voltage":0.0,"min_cell_voltage":0.0,"max_cell_temp":0.0,"min_cell_temp":0.0,"bms_fw_update_state":0,"cycle_count":4099,"vbat_inv":50.5,"time":1655110925,"datalog":"<SN>"}
```