celsworth/lxp-bridge #74: MQTT commands failing to the Lux ACS3600

zatie12, 2022-06-24:
Hi Chris

I'm just getting around to setting up MQTT. Currently running 0.7.0-dev. I seem to be having trouble issuing commands, which all get returned with a FAIL message. For example:  (I've replaced the actual datalog number with _datalog_, although I'm not sure if it's particularly private). I'm most likely doing something wrong!

In another terminal:

`$ mosquitto_pub -t lxp/cmd/_datalog_/read/inputs/1 -m ""`

lxp-bridge log:

```
Jun 24 09:08:58 miranda lxp-bridge[54494]: [2022-06-24T09:08:58.987 DEBUG lxp_bridge::mqtt] RX: Message { topic: "cmd/_datalog_/read/inputs/1", payload: "" }
Jun 24 09:08:58 miranda lxp-bridge[54494]: [2022-06-24T09:08:58.987 DEBUG lxp_bridge::coordinator] parsed command ReadInputs1(Inverter { enabled: true, host: "10.0.100.165", port: 8000, serial: BAxxxxxxxx, datalog: _datalog_ })
Jun 24 09:09:09 miranda lxp-bridge[54494]: [2022-06-24T09:09:09.991 DEBUG lxp_bridge::mqtt] publishing: lxp/result/_datalog_/read/inputs/1 = FAIL
Jun 24 09:09:09 miranda lxp-bridge[54494]: [2022-06-24T09:09:09.991 DEBUG rumqttc::state] Publish. Topic = lxp/result/_datalog_/read/inputs/1, Pkid = 59, Payload Size = 4
```

I can see the normal 5-minute updates coming through with mosquitto_sub. Everything else (Infliux, MySQL) has been working great. Thanks as always for this great project!

celsworth, 2022-06-24:
Yeah I tend to obscure my serial numbers as well but I agree, I don't think they're that sensitive, there's nothing anyone could do with them.

So I just tried this on current master and got a successful response so this is probably either a misconfiguration or perhaps you have a newer firmware that does something different. Is that the entire log? It's not getting any response at all which can sometimes mean the serial/datalog numbers don't match, as the inverter will only respond if we send the right ones. A complete successful log should look like this:

```
Jun 24 09:35:59 vps lxp-bridge[2707336]: [2022-06-24T09:35:59.069 DEBUG lxp_bridge::mqtt] RX: Message { topic: "cmd/all/read/inputs/1", payload: "" }
Jun 24 09:35:59 vps lxp-bridge[2707336]: [2022-06-24T09:35:59.071 DEBUG lxp_bridge::coordinator] parsed command ReadInputs1(Inverter { enabled: true, host: "192.168.0.15", port: 8000, serial: XXXXXXXXXX, datalog: XXXXXXXXXX })
Jun 24 09:36:00 vps lxp-bridge[2707336]: [2022-06-24T09:36:00.102 DEBUG lxp_bridge::lxp::packet_decoder] 117 bytes in: [161, 26, 2, 0, 111, 0, 1, 194, FF, FF, FF, FF, FF, FF, FF, FF, FF, FF, 97, 0, 1, 4, FF, FF, FF, FF, FF, FF, FF, FF, FF, FF, 0, 0, 80, 16, 0, 0, 0, 0, 0, 0, 0, 242, 1, 85, 0, 0, 47, 0, 0, 0, 0, 0, 0, 0, 0, 254, 1, 137, 9, 0, 16, 0, 0, 145, 19, 203, 1, 0, 0, 179, 0, 232, 3, 137, 9, 0, 10, 80, 112, 145, 19, 0, 0, 0, 0, 67, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21, 0, 3, 0, 3, 0, 28, 0, 0, 0, 0, 0, 24, 0, 124, 14, 188, 11, 19, 100]Jun 24 09:36:00 vps lxp-bridge[2707336]: [2022-06-24T09:36:00.103 DEBUG lxp_bridge::lxp::inverter] inverter XXXXXXXXXX: RX TranslatedData(TranslatedData { datalog: XXXXXXXXXX, device_function: ReadInput, inverter: XXXXXXXXXX, register: 0, values: [16, 0, 0, 0, 0, 0, 0, 0, 242, 1, 85, 0, 0, 47, 0, 0, 0, 0, 0, 0, 0, 0, 254, 1, 137, 9, 0, 16, 0, 0, 145, 19, 203, 1, 0, 0, 179, 0, 232, 3, 137, 9, 0, 10, 80, 112, 145, 19, 0, 0, 0, 0, 67, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21, 0, 3, 0, 3, 0, 28, 0, 0, 0, 0, 0, 24, 0, 124, 14, 188, 11] })
Jun 24 09:36:00 vps lxp-bridge[2707336]: [2022-06-24T09:36:00.104 DEBUG lxp_bridge::coordinator] RX: TranslatedData(TranslatedData { datalog: XXXXXXXXXX, device_function: ReadInput, inverter: XXXXXXXXXX, register: 0, values: [16, 0, 0, 0, 0, 0, 0, 0, 242, 1, 85, 0, 0, 47, 0, 0, 0, 0, 0, 0, 0, 0, 254, 1, 137, 9, 0, 16, 0, 0, 145, 19, 203, 1, 0, 0, 179, 0, 232, 3, 137, 9, 0, 10, 80, 112, 145, 19, 0, 0, 0, 0, 67, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21, 0, 3, 0, 3, 0, 28, 0, 0, 0, 0, 0, 24, 0, 124, 14, 188, 11] })
Jun 24 09:36:00 vps lxp-bridge[2707336]: [2022-06-24T09:36:00.105 DEBUG lxp_bridge::mqtt] publishing: lxp/XXXXXXXXXX/inputs/1 = {"status":16,"v_pv":0.0,"v_pv_1":0.0,"v_pv_2":0.0,"v_pv_3":0.0,"v_bat":49.8,"soc":85,"soh":0,"p_pv":0,"p_pv_1":0,"p_pv_2":0,"p_pv_3":0,"p_charge":0,"p_discharge":510,"v_ac_r":244.1,"v_ac_s":409.6,"v_ac_t":0.0,"f_ac":50.09,"p_inv":459,"p_rec":0,"pf":1.0,"v_eps_r":244.1,"v_eps_s":256.0,"v_eps_t":2875.2,"f_eps":50.09,"p_to_grid":67,"p_to_user":0,"e_pv_day":0.0,"e_pv_day_1":0.0,"e_pv_day_2":0.0,"e_pv_day_3":0.0,"e_inv_day":2.1,"e_rec_day":0.3,"e_chg_day":0.3,"e_dischg_day":2.8,"e_eps_day":0.0,"e_to_grid_day":0.0,"e_to_user_day":2.4,"v_bus_1":370.8,"v_bus_2":300.4,"time":1656059760,"datalog":"XXXXXXXXXX"}
Jun 24 09:36:00 vps lxp-bridge[2707336]: [2022-06-24T09:36:00.106 DEBUG lxp_bridge::mqtt] publishing: lxp/result/XXXXXXXXXX/read/inputs/1 = OK
Jun 24 09:36:00 vps lxp-bridge[2707336]: [2022-06-24T09:36:00.106 DEBUG rumqttc::state] Publish. Topic = lxp/XXXXXXXXXX/inputs/1, Pkid = 52, Payload Size = 607
Jun 24 09:36:00 vps lxp-bridge[2707336]: [2022-06-24T09:36:00.108 DEBUG rumqttc::state] Publish. Topic = lxp/result/XXXXXXXXXX/read/inputs/1, Pkid = 53, Payload Size = 2
```

Note that your serial numbers are also encoded into the data packets (I've replaced the relevant parts here with `FF`) so if you prefer you can email some full logs directly to me, chris at cae.me.uk

If that is your full log then another thing I could do is add a bit more debugging of packets we send to the inverter, then you could email me the output of that, might yield a few more clues.

zatie12, 2022-06-24:
Thanks, those were the only 4 lines returned. I've double checked the number against the Lux website, seems to match up. I'll email you the logs with the new build now.

closed 2022-06-24 by celsworth