celsworth/lxp-bridge #77: Issue during the compiling

teicors, 2022-06-24:
Hi,
I did a git pull and after a cargo build, but I got this error:
```

   Compiling reqwest v0.11.11
   Compiling rinfluxdb v0.1.0 (https://gitlab.com/celsworth/rinfluxdb.git?rev=f3f5b23e#f3f5b23e)
   Compiling sqlx-macros v0.6.0
   Compiling sqlx v0.6.0
error: /home/monitor/Software/lxp-bridge/target/debug/deps/libsqlx_macros-85f8a997a4f69fd9.so: undefined symbol: EVP_CIPHER_block_size
  --> /home/monitor/.cargo/registry/src/github.com-1ecc6299db9ec823/sqlx-0.6.0/src/lib.rs:71:1
   |
71 | pub extern crate sqlx_macros;
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0432]: unresolved imports `self::encode::Encode`, `self::decode::Decode`, `super::Decode`, `super::Encode`, `super::FromRow`, `super::Type`
   --> /home/monitor/.cargo/registry/src/github.com-1ecc6299db9ec823/sqlx-0.6.0/src/lib.rs:118:9
    |
118 | pub use self::encode::Encode;
    |         ^^^^^^^^^^^^^^^^^^^^
...
129 | pub use self::decode::Decode;
    |         ^^^^^^^^^^^^^^^^^^^^
...
143 |     pub use super::Decode;
    |             ^^^^^^^^^^^^^
144 |     pub use super::Encode;
    |             ^^^^^^^^^^^^^
145 |     pub use super::Executor;
146 |     pub use super::FromRow;
    |             ^^^^^^^^^^^^^^
...
150 |     pub use super::Type;
    |             ^^^^^^^^^^^

For more information about this error, try `rustc --explain E0432`.
error: could not compile `sqlx` due to 2 previous errors
```

Cheers,
Angelo

celsworth, 2022-06-24:
Huh that's a weird one. So it's a compilation error in https://github.com/launchbadge/sqlx which is the database library I use. I can't see any issues there about this; might be something they've recently changed.

I have recently updated this library from 0.4 to 0.6. Could you try changing the bottom line in Cargo.toml:

```
sqlx = { version = "~0.6", features = ["runtime-tokio-native-tls", "any", "postgres", "mysql", "sqlite", "chrono"] }
```

Back to 0.5 (0.4 doesn't seem to build anymore, I think I started using some 0.5-specific features). Does that work? If so maybe its something we can report to their repo.

Might also just be worth doing a `rustup update` to get the latest Rust compiler. What distribution are you building on?

teicors, 2022-06-24:
only the rust outdated.. I did a `rustup update` and it was fixed !!
Many thanks for your support !!

Angelo

celsworth, 2022-06-24:
Amazing, we like nice easy fixes :) Worth doing a `rustup update` once every few months to keep up to date. They tend to release a new compiler every month, though most of them shouldn't break compilation to that extent I guess.

Anyway happy to help :)

closed 2022-06-24 by celsworth