celsworth/lxp-bridge #85: Home Assistant not receiving updates after restart

jeremypoulter, 2022-07-24:
First of all thank you for your work on this, however I do have one issue. After restarting Home Assistant, e.g. after an update, Home Assistant stoops receiving data. Can still see the data being sent via MQTT and Home Assistant resumes updating after a restart of lxp-bridge.

At time of writing I am using Home Assistant 2022.7.6, Mosquitto Addon 6.1.2 and lxp-bridge running via the Docker image (sha256:5f6d276966725366118531f6b8f929c26bb3203531558a0d1ee99432e1df1734) but have seen the same behavior with previous versions.

Are the Home Assistant auto discovery values sent with persistence?

celsworth, 2022-07-25:
Hi!

Thanks for the report! I hadn't realised this was happening as I don't use HA myself.

The discovery packets are not currently sent with persistence - I see from https://community.home-assistant.io/t/mqtt-discovery-to-retain-or-not/310734/2 that they should be set indeed.

I've merged #86 which turns this on, can you try the `dev` docker tag?

jeremypoulter, 2022-07-25:
Will give it a try

celsworth, 2022-07-27:
Hey :) Just wanted to check if this is resolved with the recent change so I can close this?

jeremypoulter, 2022-07-27:
Just tried the dev docker image and I look to still be getting data after restarting. Thanks for your help

PR #86 (2022-07-25): Publish MQTT discovery packets with retain bit set
Hopefully fixes #85
closed 2022-07-27 by celsworth