celsworth/lxp-bridge #87: Error : not handled

teicors, 2022-08-21:
Hi Chris,
looking at the log, today I noticed these errors:
```
[2022-08-21T19:12:49.564 ERROR lxp_bridge::lxp::inverter] inverter BA10xxxxxx: not handled: tcp_function=196
[2022-08-21T19:12:49.564 INFO lxp_bridge::lxp::inverter] inverter BA10xxxxxx: reconnecting in 5s
[2022-08-21T19:12:54.566 INFO lxp_bridge::lxp::inverter] connecting to inverter BA10xxxxxx at 10.10.20.241:8000
[2022-08-21T19:12:54.572 INFO lxp_bridge::lxp::inverter] inverter BA10xxxxxx: connected!
```
and also few
```
[2022-08-21T19:18:35.096 ERROR lxp_bridge::lxp::inverter] inverter BA10xxxxxy: Connection reset by peer (os error 104)
[2022-08-21T19:18:35.097 INFO lxp_bridge::lxp::inverter] inverter BA10xxxxxy: reconnecting in 5s
[2022-08-21T19:18:40.099 INFO lxp_bridge::lxp::inverter] connecting to inverter BA10xxxxxy at 10.10.20.243:8000
[2022-08-21T19:18:40.113 INFO lxp_bridge::lxp::inverter] inverter BA10xxxxxy: connected!
```
for all my LuxPower.
Can you try to look at the issue ?

Many thanks,
Angelo

celsworth, 2022-08-25:
Hey, sorry for the slow reply, I'm vacationing at the moment with poor Internet :)

So the first log is interesting, that tcp_function=196 corresponds to a WriteParam packet, which essentially means that Lux are changing a setting in the datalogger. The only parameter I'm aware of that can be set with function 196 is the interval between data packets, which is currently 5 minutes, so maybe they're changing it. How many of these do you see?, and how often? It would be really interesting to see the DEBUG loglines that correspond with those packets, they're probably logged just above what you show. Feel free to email them to me chris@github.cae.me.uk if you like :) Then I can perhaps add some more support for these types of packets as they're currently mostly unknown to me.
I could probably improve lxp-bridge not to reconnect on these, and just ignore them. The reconnection is a bit paranoid.

The second log seems to be poor connectivity to the inverter; os error 104 just means "Connection reset by peer" which is pretty vague and just means that lxp-bridge was unable to talk to the inverter anymore on the existing TCP connection, so it reconnected. Is it on the edge of wifi connectivity maybe?

celsworth, 2022-08-29:
I've merged a commit that adds a bit more logging around these so hopefully I can work out what they're trying to do with these packets. Could you update to the latest dev build and run that for a while? If you get another tcp_function=196 packet please email me the debug log :)

closed 2022-08-31 by celsworth

teicors, 2022-09-02:
I did the latest update of lxp-bridge few seconds ago and now is running.
I'll bother you when something strange appears :1st_place_medal: 
As usual thanks !!

PR #90 (2022-08-29): Add a bit more logging around unhandled tcp_function packets
Trying to resolve #87 by seeing whats in these packets.
PR #92 (2022-08-31): Ignore WriteParam packets, don't disconnect/exit
Not sure what they really do. Just ignoring them for now to avoid lxp-bridge exiting every time it sees one.

Fixes #87