celsworth/lxp-bridge #94: unhandled ReadInput in 0.9.0-dev

jacovt, 2022-09-02:
Hello Chris,

Thank you for this very useful piece of software.

I just started tinkering with it yesterday (v0.7.0) and I am running into the same issue with unhandled ReadInput as described in #88.

I saw that this issue was addressed in PR #89 and merged into the mainline, so I pulled 0.9.0-dev from docker, however, I am still experiencing this issue. I am not 100% sure if #89 made it into 0.9.0-dev that is currently on Dockerhub?

If this issue is supposed to be fixed, but not, I will be very happy to assist with testing / providing any additional information.

Thank you!

Here are the logs in question:
`[2022-09-02T12:52:13.370 INFO lxp_bridge] lxp-bridge 0.9.0-dev starting
[2022-09-02T12:52:13.372 INFO lxp_bridge::lxp::inverter] connecting to inverter BA12200285 at 10.0.0.183:8000
[2022-09-02T12:52:13.373 INFO lxp_bridge::scheduler] scheduler disabled, skipping
...
...
[2022-09-02T12:56:24.602 DEBUG lxp_bridge::lxp::packet_decoder] 19 bytes in: [161, 26, 5, 0, 13, 0, 1, 193, 66, 65, 49, 50, 50, 48, 48, 50, 56, 53, 0]
[2022-09-02T12:56:24.602 DEBUG lxp_bridge::coordinator] RX: Heartbeat(Heartbeat { datalog: BA12200285 })
[2022-09-02T12:56:25.450 DEBUG lxp_bridge::lxp::packet_decoder] 291 bytes in: [161, 26, 5, 0, 29, 1, 1, 194, 66, 65, 49, 50, 50, 48, 48, 50, 56, 53, 15, 1, 1, 4, 50, 50, 54, 51, 48, 51, 56, 48, 49, 51, 0, 0, 254, 17, 0, 0, 0, 0, 0, 52, 12, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 231, 8, 0, 0, 0, 0, 146, 19, 0, 0, 0, 0, 40, 0, 0, 0, 234, 8, 20, 0, 193, 124, 146, 19, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 65, 12, 37, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 28, 0, 24, 0, 26, 0, 28, 0, 0, 0, 247, 15, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 44, 1, 0, 0, 0, 0, 0, 3, 17, 34, 67, 42, 112, 50, 3, 136, 14, 131, 240, 96, 13, 32, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 80, 15, 0, 0, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 5, 1, 0, 0, 50, 50, 54, 51, 48, 51, 56, 48, 49, 51, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 240, 131]
[2022-09-02T12:56:25.451 DEBUG lxp_bridge::coordinator] RX: TranslatedData(TranslatedData { datalog: BA12200285, device_function: ReadInput, inverter: 2263038013, register: 0, values: [17, 0, 0, 0, 0, 0, 52, 12, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 231, 8, 0, 0, 0, 0, 146, 19, 0, 0, 0, 0, 40, 0, 0, 0, 234, 8, 20, 0, 193, 124, 146, 19, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 65, 12, 37, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 28, 0, 24, 0, 26, 0, 28, 0, 0, 0, 247, 15, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 44, 1, 0, 0, 0, 0, 0, 3, 17, 34, 67, 42, 112, 50, 3, 136, 14, 131, 240, 96, 13, 32, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 80, 15, 0, 0, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 5, 1, 0, 0, 50, 50, 54, 51, 48, 51, 56, 48, 49, 51, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] })
[2022-09-02T12:56:25.451 DEBUG lxp_bridge::mqtt] publishing: lxp/BA12200285/inputs/all = {"status":17,"v_pv":312.4,"v_pv_1":0.0,"v_pv_2":0.0,"v_pv_3":312.4,"v_bat":0.0,"soc":0,"soh":0,"p_pv":0,"p_pv_1":0,"p_pv_2":0,"p_pv_3":0,"p_charge":0,"p_discharge":0,"v_ac_r":227.9,"v_ac_s":0.0,"v_ac_t":0.0,"f_ac":50.1,"p_inv":0,"p_rec":0,"pf":0.0,"v_eps_r":228.2,"v_eps_s":2.0,"v_eps_t":3193.7,"f_eps":50.1,"p_eps":0,"s_eps":0,"p_to_grid":0,"p_to_user":0,"e_pv_day":0.0,"e_pv_day_1":0.0,"e_pv_day_2":0.0,"e_pv_day_3":0.0,"e_inv_day":0.0,"e_rec_day":0.0,"e_chg_day":0.0,"e_dischg_day":0.0,"e_eps_day":0.0,"e_to_grid_day":0.0,"e_to_user_day":0.0,"v_bus_1":313.7,"v_bus_2":3.7,"e_pv_all":0.0,"e_pv_all_1":0.0,"e_pv_all_2":0.0,"e_pv_all_3":0.0,"e_inv_all":0.0,"e_rec_all":0.0,"e_chg_all":0.0,"e_dischg_all":0.0,"e_eps_all":0.0,"e_to_grid_all":0.0,"e_to_user_all":0.0,"t_inner":28,"t_rad_1":24,"t_rad_2":26,"t_bat":28,"runtime":4087,"max_chg_curr":0.0,"max_dischg_curr":0.0,"charge_volt_ref":0.0,"dischg_cut_volt":30.0,"bat_status_0":0,"bat_status_1":0,"bat_status_2":768,"bat_status_3":8721,"bat_status_4":10819,"bat_status_5":12912,"bat_status_6":34819,"bat_status_7":33550,"bat_status_8":24816,"bat_status_9":8205,"bat_status_inv":0,"bat_count":0,"bat_capacity":0,"bat_current":0.0,"bms_event_1":0,"bms_event_2":7,"max_cell_voltage":0.0,"min_cell_voltage":0.0,"max_cell_temp":0.0,"min_cell_temp":0.0,"bms_fw_update_state":3920,"cycle_count":0,"vbat_inv":0.4,"time":1662123385,"datalog":"BA12200285"}
[2022-09-02T12:56:25.452 DEBUG rumqttc::state] Publish. Topic = lxp/BA12200285/inputs/all, Pkid = 51, Payload Size = 1398
[2022-09-02T12:56:26.629 DEBUG lxp_bridge::lxp::packet_decoder] 291 bytes in: [161, 26, 5, 0, 29, 1, 1, 194, 66, 65, 49, 50, 50, 48, 48, 50, 56, 53, 15, 1, 1, 4, 50, 50, 54, 51, 48, 51, 56, 48, 49, 51, 127, 0, 254, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 236, 171]
[2022-09-02T12:56:26.629 DEBUG lxp_bridge::coordinator] RX: TranslatedData(TranslatedData { datalog: BA12200285, device_function: ReadInput, inverter: 2263038013, register: 127, values: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] })
[2022-09-02T12:56:26.631 ERROR lxp_bridge] unhandled ReadInput register=127 len=254`

celsworth, 2022-09-02:
Looks like my fix wasn't sufficient :( I'll investigate further - I'm not seeing these packets on my inverter unfortunately but I'll have another look.

celsworth, 2022-09-02:
Ok got it. Ther'es another place I have to handle it ;) Fix shortly

closed 2022-09-02 by celsworth

jacovt, 2022-09-07:
Thank you, @celsworth for the quick update. I can confirm this issue is resolved in 0.9.0-dev.

PR #95 (2022-09-02): Second attempt at ignoring unknown ReadInputs
Fixes #94