celsworth/lxp-bridge #96: Some MQTT input topics don't get published

excieve, 2022-09-27:
I've set up `lxp-bridge` for the Luxpower Eco-Hybrid SNA5000 inverter. I had to use the `dev` tag since the latest version (`v0.8.0`) seems to crash.

It works for the most part, however it only publishes the messages on the `.../inputs/all` topic, while `.../inputs/{1,2,3}` are missing. I noticed it through the Home Assistant integration, which received the discovery configurations, but never the actual values. As I understand the configs point to those missing inputs topics.

There are also frequent ` ignoring unhandled ReadInput` log lines like this:
```
2022-09-27T10:55:53.867 DEBUG lxp_bridge::lxp::packet_decoder] 19 bytes in: [161, 26, 5, 0, 13, 0, 1, 193, 66, 65, 49, 50, 49, 51, 48, 48, 50, 51, 0]
[2022-09-27T10:55:53.867 DEBUG lxp_bridge::coordinator] RX: Heartbeat(Heartbeat { datalog: redacted })
[2022-09-27T10:55:55.071 DEBUG lxp_bridge::lxp::packet_decoder] 291 bytes in: [161, 26, 5, 0, 29, 1, 1, 194, 66, 65, 49, 50, 49, 51, 48, 48, 50, 51, 15, 1, 1, 4, 50, 50, 55, 51, 48, 53, 48, 50, 51, 51, 0, 0, 254, 20, 0, 4, 9, 54, 7, 92, 16, 15, 2, 100, 0, 0, 0, 99, 0, 54, 0, 0, 0, 0, 0, 0, 0, 30, 9, 0, 0, 0, 0, 142, 19, 95, 0, 0, 0, 210, 0, 0, 0, 44, 9, 212, 1, 193, 124, 141, 19, 0, 0, 0, 0, 0, 0, 0, 0, 4, 0, 4, 0, 0, 0, 5, 0, 4, 0, 1, 0, 1, 0, 0, 0, 0, 0, 12, 0, 100, 16, 101, 14, 6, 0, 0, 0, 5, 0, 0, 0, 0, 0, 0, 0, 5, 0, 0, 0, 40, 0, 0, 0, 35, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 56, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 41, 0, 33, 0, 50, 0, 41, 0, 0, 0, 209, 52, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 10, 0, 0, 0, 76, 29, 20, 2, 0, 0, 0, 0, 1, 8, 0, 5, 16, 20, 247, 42, 112, 50, 191, 171, 14, 3, 240, 96, 253, 28, 2, 0, 3, 0, 150, 0, 0, 0, 0, 0, 14, 2, 195, 13, 166, 13, 210, 0, 200, 0, 215, 52, 0, 0, 15, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 1, 0, 0, 50, 50, 55, 51, 48, 53, 48, 50, 51, 51, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 253, 98]
[2022-09-27T10:55:55.071 DEBUG lxp_bridge::coordinator] RX: TranslatedData(TranslatedData { datalog: redacted, device_function: ReadInput, inverter: redacted, register: 0, values: [20, 0, 4, 9, 54, 7, 92, 16, 15, 2, 100, 0, 0, 0, 99, 0, 54, 0, 0, 0, 0, 0, 0, 0, 30, 9, 0, 0, 0, 0, 142, 19, 95, 0, 0, 0, 210, 0, 0, 0, 44, 9, 212, 1, 193, 124, 141, 19, 0, 0, 0, 0, 0, 0, 0, 0, 4, 0, 4, 0, 0, 0, 5, 0, 4, 0, 1, 0, 1, 0, 0, 0, 0, 0, 12, 0, 100, 16, 101, 14, 6, 0, 0, 0, 5, 0, 0, 0, 0, 0, 0, 0, 5, 0, 0, 0, 40, 0, 0, 0, 35, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 56, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 41, 0, 33, 0, 50, 0, 41, 0, 0, 0, 209, 52, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 10, 0, 0, 0, 76, 29, 20, 2, 0, 0, 0, 0, 1, 8, 0, 5, 16, 20, 247, 42, 112, 50, 191, 171, 14, 3, 240, 96, 253, 28, 2, 0, 3, 0, 150, 0, 0, 0, 0, 0, 14, 2, 195, 13, 166, 13, 210, 0, 200, 0, 215, 52, 0, 0, 15, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 1, 0, 0, 50, 50, 55, 51, 48, 53, 48, 50, 51, 51, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] })
[2022-09-27T10:55:55.071 DEBUG lxp_bridge::mqtt] publishing: lxp/redacted/inputs/all = {"status":20,"v_pv":834.2,"v_pv_1":230.8,"v_pv_2":184.6,"v_pv_3":418.8,"v_bat":52.7,"soc":100,"soh":0,"p_pv":153,"p_pv_1":99,"p_pv_2":54,"p_pv_3":0,"p_charge":0,"p_discharge":0,"v_ac_r":233.4,"v_ac_s":0.0,"v_ac_t":0.0,"f_ac":50.06,"p_inv":95,"p_rec":0,"pf":0.0,"v_eps_r":234.8,"v_eps_s":46.8,"v_eps_t":3193.7,"f_eps":50.05,"p_eps":0,"s_eps":0,"p_to_grid":0,"p_to_user":0,"e_pv_day":0.8,"e_pv_day_1":0.4,"e_pv_day_2":0.4,"e_pv_day_3":0.0,"e_inv_day":0.5,"e_rec_day":0.4,"e_chg_day":0.1,"e_dischg_day":0.1,"e_eps_day":0.0,"e_to_grid_day":0.0,"e_to_user_day":1.2,"v_bus_1":419.6,"v_bus_2":368.5,"e_pv_all":1.1,"e_pv_all_1":0.6,"e_pv_all_2":0.5,"e_pv_all_3":0.0,"e_inv_all":0.5,"e_rec_all":4.0,"e_chg_all":3.5,"e_dischg_all":0.2,"e_eps_all":0.0,"e_to_grid_all":0.0,"e_to_user_all":5.6,"t_inner":41,"t_rad_1":33,"t_rad_2":50,"t_bat":41,"runtime":79057,"max_chg_curr":0.0,"max_dischg_curr":75.0,"charge_volt_ref":53.2,"dischg_cut_volt":0.0,"bat_status_0":0,"bat_status_1":2049,"bat_status_2":1280,"bat_status_3":5136,"bat_status_4":10999,"bat_status_5":12912,"bat_status_6":43967,"bat_status_7":782,"bat_status_8":24816,"bat_status_9":7421,"bat_status_inv":2,"bat_count":3,"bat_capacity":150,"bat_current":0.0,"bms_event_1":0,"bms_event_2":526,"max_cell_voltage":35.23,"min_cell_voltage":34.94,"max_cell_temp":2.1,"min_cell_temp":2.0,"bms_fw_update_state":13527,"cycle_count":0,"vbat_inv":52.7,"time":1664276155,"datalog":"redacted"}
[2022-09-27T10:55:55.071 DEBUG rumqttc::state] Publish. Topic = lxp/redacted/inputs/all, Pkid = 78, Payload Size = 1429
[2022-09-27T10:55:56.252 DEBUG lxp_bridge::lxp::packet_decoder] 291 bytes in: [161, 26, 5, 0, 29, 1, 1, 194, 66, 65, 49, 50, 49, 51, 48, 48, 50, 51, 15, 1, 1, 4, 50, 50, 55, 51, 48, 53, 48, 50, 51, 51, 127, 0, 254, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 133, 25]
[2022-09-27T10:55:56.252 DEBUG lxp_bridge::coordinator] RX: TranslatedData(TranslatedData { datalog: redacted, device_function: ReadInput, inverter: redacted, register: 127, values: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] })
[2022-09-27T10:55:56.253 WARN lxp_bridge::coordinator] ignoring unhandled ReadInput register=127 len=254
[2022-09-27T10:55:56.253 WARN lxp_bridge::mqtt] ignoring unhandled ReadInput register=127 len=254
[2022-09-27T10:55:57.432 DEBUG lxp_bridge::lxp::packet_decoder] 291 bytes in: [161, 26, 5, 0, 29, 1, 1, 194, 66, 65, 49, 50, 49, 51, 48, 48, 50, 51, 15, 1, 1, 4, 50, 50, 55, 51, 48, 53, 48, 50, 51, 51, 254, 0, 254, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 217, 87]
[2022-09-27T10:55:57.433 DEBUG lxp_bridge::coordinator] RX: TranslatedData(TranslatedData { datalog: redacted, device_function: ReadInput, inverter: redacted, register: 254, values: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] })
[2022-09-27T10:55:57.433 WARN lxp_bridge::coordinator] ignoring unhandled ReadInput register=254 len=254
[2022-09-27T10:55:57.433 WARN lxp_bridge::mqtt] ignoring unhandled ReadInput register=254 len=254
```

jacovt, 2022-09-27:
> I've set up `lxp-bridge` for the Luxpower Eco-Hybrid SNA5000 inverter. I had to use the `dev` tag since the latest version (`v0.8.0`) seems to crash.
>
> It works for the most part, however it only [...]

Hello @excieve ,

I believe the ReadInput failures are something new in the SNA range of devices' firmware that sends some empty values for that register. I believe most of these were fixed in 0.9.0-dev that is available on Docker Hub. It does look like register 254 is a new failure though, I have also started receiving these the last week and a bit. I believe these are triggered from the comms to the LuxPower portal. I am working around the current crash by using docker compose with a auto restart in case of a container failure.

As far as your missing ../1 ../2 and ../3 topics go, I am working around this by sending a request via MQTT to the command topic for those 3 topics. I do this using node red as I want to receive the values in HomeAssistant every 5 seconds (I like realtime monitoring). It seems that ../all also gets updated once you request values for ../3 using the command topic.

I could publish that Node-Red flow for you in a gist somewhere if you are interested?

It would be nice to get these fixed so that we do not have to use workarounds, but I believe that @celsworth is running different hardware than we are. I am hoping I can get some time this week to write some quick patches (need to brush up on my Rust skills) that I will send to @celsworth for review.

excieve, 2022-09-27:
@jacovt The current `dev` tag doesn't crash anymore — just ignores those unhandled registers, which might be fine.

> I could publish that Node-Red flow for you in a gist somewhere if you are interested?

I'd be very interested, thanks!

> It would be nice to get these fixed so that we do not have to use workarounds

I wonder why do we even need those `../1`, etc. topics if everything seems to be present in the `.../all` topic though? Or is relevant when there are multiple parallel invertors connected?

celsworth, 2022-09-27:
> I wonder why do we even need those `../1`, etc. topics if everything seems to be present in the `.../all` topic though? Or is relevant when there are multiple parallel invertors connected?

So I guess this is a bit evolutionary. As has been suggested above, my inverter (a bit older now, an ACS3600, not a hybrid) publishes 3 different packets a second or so apart, each containing different bits of data. Newer ones publish a single larger packet of data that contains all those 3 combined, and perhaps newer ones still publish yet more larger packets although they contain mostly zeroes (as does yours above) so perhaps its not all used yet.

There were two trains of thought behind introducing the `/all` MQTT message:

* unify the different inverter types so even the older ones like mine would lead to lxp-bridge publishing an `/all` packet, so it doesn't matter which inverter you have, you can always rely on this `/all` being published with all the data we get. So effectively, ignore the older 1/2/3 and just use `/all`, no matter which inverter you have
* when I added InfluxDB capability, I noticed that it was adding 3 different rows to the database, each with their own timestamp, and each with only a third of the data populated, which led to very odd graphs and gaps in the data etc. So I started forming the idea of this merging the data into one super-packet, which was then sent to InfluxDB in one fell swoop, leading to one row with all the data. Much better graphs as a result.

I left the 1/2/3 messages in for backwards compatibility mostly, as people are probably using them already and I didn't want to break that. It could be an option whether to bother publishing them at all now.

excieve, 2022-09-27:
@celsworth Thanks for the clarifications! If `/all` always has the right set of data regardless of the inverter, then would it make sense to change the HA discovery configs to always point to this topic instead of 1/2/3?

excieve, 2022-09-27:
@celsworth I've made a [quick PR](#98) for this — works fine on my setup.

closed 2022-09-29 by celsworth