celsworth/lxp-bridge #97: Add inverter / datalog identification to Home Assistant auto discover topics

jacovt, 2022-09-27:
Hello @celsworth ,

I am starting this issue to discuss an approach for making the Home Assistant sensor metrics a bit more human friendly when you are using multiple inverters / datalogs.

I am running 2 Lux SNA5000's in parallel and when you publish the data from the datalogs to Home Assistant, the names get very confusing in Home Assistant. Sure the unique values are clearly distinguishable as they contain the identifier for the datalog in the names for the sensors, but it becomes tricky when you want to use the sensor data in a graph / calculation because of the identical sensor name that gets and auto incremented sensor name (you end up with sensor_name and sensor_name_1 / sensor_name_2).

I was toying with the idea of using the datalog identifier in the name (as this is easy to override the display name for visualisation when you add a graph). Alternatively, we can include a "friendly name" config option per inverter config that can be used to identify where the metric is coming from that can then be used when publishing the discovery data for the sensors.

And then including a configuration option to turn this on/off for people that choose to use it / not use it.

Thoughts?

Thank you,
---Jaco

celsworth, 2022-09-27:
Hi!

Yeah I am absolutely open to improving this. We did a small rework of the HomeAssistant stuff a while back and I was concerned at the time it might not be too helpful for parallel inverter users but wasn't too sure how to tackle it.

If I remember correctly a concern was raised before that having the inverter datalog in the HA visible labels was a bit redundant if you only had one inverter, which I'd guess most people do. So if we do add the datalog into a label it should be configurable on/off as you mention; or possibly we could only do it if there are multiple inverters configured. If thats satisfied then it sounds good to me :)

celsworth, 2022-12-05:
I believe #120 addressed this, please reopen if anything more needs to be done!

closed 2022-12-05 by celsworth