celsworth/lxp-bridge #101: Further documentation on how to retrieve 'serial' and 'datalog' values

chriscn, 2022-10-07:
This is partly a question and partly a suggestion to improve the documentation. I am not sure where to find the 'serial' and 'datalog' values of the inverter setups. It isn't in the Wiki (as far as I can see).

chriscn, 2022-10-07:
Mainly looking at this error:
```
Oct 07 10:46:47 pi-mqtt lxp-bridge[4239]: [2022-10-07T10:46:47.224 WARN lxp_bridge::lxp::inverter] inverter serial mismatch found; packet=9032001033, config=5555555555 - please check config!
```

celsworth, 2022-10-10:
👍 I thought I had documented this but to be honest maybe this is the kick I need to write a proper Getting Started wiki page. It'll slim down the README a bit as well hopefully.

I'll pick it up soon :)

celsworth, 2022-10-10:
Well effectively the warning message was added to make it easier; its telling you that the serial being sent out by the inverter doesn't match the serial in your configuration. So replace the 5555555 in your config with the number shown in your logline and should all be good.

I was hoping to automate this by replacing the in-memory serial the first time we see the inverter send its true serial but it got difficult for various technical Rust reasons and it was low-reward so I abandoned it for now.

Note that its only important to have the correct serial/datalog in the configuration if you want to control the inverter. If you're only monitoring it then its irrelevant. I'll put this in the wiki page :)

celsworth, 2022-11-03:
Found some time to start the wiki ;)

https://github.com/celsworth/lxp-bridge/wiki/Configuration#serial-numbers :)

I'll do more as time permits but hopefully this is a good start..

Swwils, 2022-11-03:
HOLD_SERIAL_NUM spits out the serial number. Its needed for lots of reads/writes.

celsworth, 2022-11-03:
Yeah, but in order to send the inverter a command to read that register, you need to know the serial number as far as I'm aware.

I'm planning to middle-man-attack the app while its interrogating my inverter to see how they do it at some point :)

Because you can use the phone apps without the app knowing your serial numbers in advance.

Swwils, 2022-11-03:
I think it does it from the "plantId"

closed 2022-11-05 by celsworth