celsworth/lxp-bridge #102: Info log level should log when a MQTT message is published.

chriscn, 2022-10-10:
Would be useful for presentating information. I believe info should log when stuff is sent to MQTT.

celsworth, 2022-10-10:
Sure I could do that. What would you want to see in the logline? A complete dump of the topic and payload is already logged at DEBUG so presumably not that much information?

The mqtt library I use also logs the topic and payload size, but thats also at DEBUG, so I could replicate something similar to that at INFO level.

chriscn, 2022-10-10:
What I would find useful is a log file of the following
`{TIME} sent MQTT for inverter {INVERTER}`

I'd been troubleshooting this morning and had to have an MQTT client setup to see when the messages where being sent as oppossed to just tailing the logs.

PR #108 (2022-11-03): fix: change log level of mqtt publish from debug to info
Would close #102
closed 2022-11-03 by celsworth