celsworth/lxp-bridge #103: No data emitted from inverter without conncting to the luxpower site?

JavonneM, 2022-10-25:
Currently I've setup lxp-bridge with mqtt + mysql db, however no data is being emitted from the datalogger. I have assumed this to be due to no one requesting the data (I do not have much experience in the realm of hardware interfacing/interactions, any guidance on how this is done would be awesome, would love to contribute), I have not created an account on lux power's website or connected the datalogger to their IP (I've actually removed the IP and connected it to a netcat host, really don't want china monitoring my power usage :upside_down_face: ).

Currently the only interaction between the datalogger and lxp-bridge seems to be the heartbeat/ping. Is my understanding correct? Is this behavior expected?

```
lxp-bridge-service  | [2022-10-25T19:51:45.918 DEBUG lxp_bridge::lxp::packet_decoder] 19 bytes in: [<I assume these bytes are sensitive?>]
lxp-bridge-service  | [2022-10-25T19:51:45.918 DEBUG lxp_bridge::coordinator] RX: Heartbeat(Heartbeat { datalog: XXXXXXXX })
lxp-bridge-service  | [2022-10-25T19:52:12.152 DEBUG rumqttc::state] Pingreq,
lxp-bridge-service  |             last incoming packet before 59999 millisecs,
lxp-bridge-service  |             last outgoing request before 60001 millisecs
lxp-bridge-service  | [2022-10-25T19:52:45.680 DEBUG lxp_bridge::lxp::packet_decoder] 19 bytes in: [<I assume these bytes are sensitive?>]
lxp-bridge-service  | [2022-10-25T19:52:45.680 DEBUG lxp_bridge::coordinator] RX: Heartbeat(Heartbeat { datalog: XXXXXXXX })
lxp-bridge-service  | [2022-10-25T19:53:12.154 DEBUG rumqttc::state] Pingreq,
lxp-bridge-service  |             last incoming packet before 60000 millisecs,
lxp-bridge-service  |             last outgoing request before 60001 millisecs
```

From what I can see the only way to trigger a data fetch is via publishing to MQTT as described in the documentation
```
mosquitto_pub -h localhost -t lxp/cmd/{datalogger}/read/inputs/1 -m ""
mosquitto_pub -h localhost -t lxp/cmd/{datalogger}/read/inputs/2 -m ""
mosquitto_pub -h localhost -t lxp/cmd/{datalogger}/read/inputs/3 -m ""
```
Notably it seems that only ` lxp/cmd/{datalogger}/read/inputs/2 or 3` stores to the mysql db and is used by Home Assistant.

If my understanding is correct I will work on adding a auto polling feature and add a config for each inverter with a time interval. Thoughts on this?

## Current Configuration
Image: `celsworth/lxp-bridge:latest`
Inverter config
```
inverters:
- enabled: true
  host: xxxxxxx
  port: 8000
  serial: xxxxxxxxxx
  datalog: xxxxxxx 

databases:
- enabled: true 
  url: mysql://xxxxx:xxxxx@xxxxxx:3306/xxxxx

mqtt:
  enabled: true 
  host: xxxxxxx
  port: 1883
  username:  
  password: 
  namespace: lxp
  homeassistant:
    enabled: true 
    prefix: homeassistant
    sensors: all # or e_pv_all,v_bat,soc etc

```

celsworth, 2022-10-25:
Ooh now this is quite interesting. I think you might be the first person I know of to try operating this inverter without a connection to the Luxpower server. I kind of wondered if it would operate correctly in this case, I guess we're finding out now :)

I don't believe that Luxpower have to request the data every 5 minutes; I've monitored traffic to/from the inverter and there's no "request" that sets off the data broadcasts. But there may be some one-time setup that happens when you first connect the inverter to them, and I completely understand you not wanting to do that.

Quite interesting that you get heartbeats every minute, I only get them every 5 (immediately before the data is broadcast).

Out of interest, what value is stored in param 0 on your inverter? You can get this with:

```
mosquitto_pub -h localhost -t lxp/cmd/{datalogger}/read/param/0 -m ""
```

It should be the number of seconds between data broadcasts. I wonder if yours is 0? If so, setting a value in that register *might* kick it into action. We could explore a way to hackily do that (lxp-bridge doesn't yet support it). If its nonzero though, this idea falls flat.

There is a rudimentary scheduler built into the latest versions of lxp-bridge, it only supports syncing the time at the moment, but perhaps this could be used to send the requests for data to the inverter at set intervals. That's definitely something I'd like to support if its the only way to get the data out without talking to China.

JavonneM, 2022-10-25:
Interesting 
```
mosquitto_pub -h localhost -t lxp/cmd/{datalogger}/read/param/0 -m ""
```
results in a failure, and seems to cause a connection loss
```
xp-bridge-service  | [2022-10-25T20:55:51.869 INFO lxp_bridge::lxp::inverter] connecting to inverter {datalogger} at {IP_Address}:8000
lxp-bridge-service  | [2022-10-25T20:55:51.872 INFO lxp_bridge::lxp::inverter] inverter {datalogger}: connected!
lxp-mqtt-service    | 1666731355: New connection from 127.0.0.1:57956 on port 1883.
lxp-mqtt-service    | 1666731355: New client connected from 127.0.0.1:57956 as auto-66DDD9C3-FBD1-3DB9-5ACE-09AB0E8A6DF1 (p2, c1, k60).
lxp-mqtt-service    | 1666731355: Client auto-66DDD9C3-FBD1-3DB9-5ACE-09AB0E8A6DF1 disconnected.
lxp-bridge-service  | [2022-10-25T20:55:55.785 DEBUG lxp_bridge::mqtt] RX: Message { topic: "cmd/{datalogger}/read/param/0", payload: "" }
lxp-bridge-service  | [2022-10-25T20:55:55.785 DEBUG lxp_bridge::coordinator] parsed command ReadParam(Inverter { enabled: true, host: "{IP_Address}", port: 8000, serial: {serial}, datalog: {datalogger} }, 0)
lxp-bridge-service  | [2022-10-25T20:55:55.785 DEBUG lxp_bridge::lxp::inverter] inverter {datalogger}: TX [{bytes}]
lxp-bridge-service  | [2022-10-25T20:55:56.830 DEBUG lxp_bridge::lxp::packet_decoder] 24 bytes in: [{bytes}]
lxp-bridge-service  | [2022-10-25T20:55:56.831 ERROR lxp_bridge::lxp::inverter] inverter {datalogger}: ReadParam::decode mismatch: values.len()=4, value_length_byte=2
lxp-bridge-service  | [2022-10-25T20:55:56.831 INFO lxp_bridge::lxp::inverter] inverter {datalogger}: reconnecting in 5s
lxp-bridge-service  | [2022-10-25T20:55:56.831 DEBUG lxp_bridge::mqtt] publishing: lxp/result/{datalogger}/read/param/0 = FAIL
lxp-bridge-service  | [2022-10-25T20:55:56.831 DEBUG rumqttc::state] Publish. Topic = lxp/result/{datalogger}/read/param/0, Pkid = 48, Payload Size = 4
lxp-bridge-service  | [2022-10-25T20:56:01.832 INFO lxp_bridge::lxp::inverter] connecting to inverter {datalogger} at {IP_Address}:8000
lxp-bridge-service  | [2022-10-25T20:56:01.835 INFO lxp_bridge::lxp::inverter] inverter {datalogger}: connected!
```
In general the connection to the datalogger drops every now and then, but this consistently resets the connection when the cmd is sent.
So to set this parameter one would 
```
mosquitto_pub -h localhost -t lxp/cmd/{datalog}/set/param/0 -m 60
```

Need to reread the docs and make sure I fully understand `hold` `param` and `inputs`

celsworth, 2022-10-25:
Oh thats weird, thats like lxp-bridge doesn't like the reply coming back:

```
lxp-bridge-service  | [2022-10-25T20:55:56.831 ERROR lxp_bridge::lxp::inverter] inverter {datalogger}: ReadParam::decode mismatch: values.len()=4, value_length_byte=2
```

The line above with the 24 bytes in may contain some clues, but you are right it will also contain your serial number in hex. Don't suppose you fancy emailing me the contents of that line? chris@github.cae.me.uk would be fine.. bed soon though, will look in the morning :)

JavonneM, 2022-10-26:
Heartbeat
```
[161, 26, 5, 0, 13, 0, 1, 193, xx, xx, xx, xx, xx, xx, xx, xx, xx, xx, 0]
```
Result from `lxp/cmd/{datalogger}/read/param/0 `
```
[161, 26, 5, 0, 18, 0, 1, 195, xx, xx, xx, xx, xx, xx, xx, xx, xx, xx, 0, 0, 2, 0, 44, 1]
```
Took a quick look at the decoding code (Not sure what byte is in index 7 :shrug:, probably packet type:shrug:) I dont see an issue with the lengths of the packet data, 13 -> 18 (+5) and there are 5 additional bytes at the end. 
Will take a deeper look into `ReadParam.decode` later

celsworth, 2022-10-26:
Hi,

Just for comparison here is mine;

Oct 26 08:36:57 vps lxp-bridge[3633921]: [2022-10-26T08:36:57.258 DEBUG lxp_bridge::mqtt] RX: Message { topic: "cmd/BA19260024/read/param/0", payload: "" }
Oct 26 08:36:57 vps lxp-bridge[3633921]: [2022-10-26T08:36:57.259 DEBUG lxp_bridge::coordinator] parsed command ReadParam(Inverter { enabled: true, host: "192.168.0.15", port: 8000, serial: 9292006063, datalog: BA19260024 }, 0)
Oct 26 08:36:57 vps lxp-bridge[3633921]: [2022-10-26T08:36:57.260 DEBUG lxp_bridge::lxp::inverter] inverter BA19260024: TX [161, 26, 2, 0, 14, 0, 1, 195, 66, 65, 49, 57, 50, 54, 48, 48, 50, 52, 0, 0]
Oct 26 08:36:58 vps lxp-bridge[3633921]: [2022-10-26T08:36:58.802 DEBUG lxp_bridge::lxp::packet_decoder] 24 bytes in: [161, 26, 2, 0, 18, 0, 1, 195, 66, 65, 49, 57, 50, 54, 48, 48, 50, 52, 0, 0, 2, 0, 44, 1]
Oct 26 08:36:58 vps lxp-bridge[3633921]: [2022-10-26T08:36:58.803 DEBUG lxp_bridge::coordinator] RX: ReadParam(ReadParam { datalog: BA19260024, register: 0, values: [44, 1] })

So the immediate difference I see is you have protocol 5 and I have protocol 2 (the third byte). This could probably do with fixing in lxp-bridge, but its probably not worth it at this stage because we can see the final two bytes of both our responses are [44, 1] - this is the contents of the register which translates to (1 << 8) + 44 = 300.

(actually I think the fix for handling this protocol 5 is probably just to treat it like protocol 2 since it looks identical, lxp-bridge is probably treating it like protocol 1 or something)

So we are back to square 1 really - we don’t know what, if any, initial setup Luxpower do to get the data being broadcast. It seems you already have the 300s interval set.

Lets move discussion back to the Github thread so we have a public record :)

celsworth, 2022-10-27:
Started the scheduler in #104.

I could also try disconnecting my inverter from their servers, either by simply removing the IP in its config or firewalling it off, and see if the packets stop. If they do then perhaps when I reconnect it I'll get some clues about how they request it to start sending.

JavonneM, 2022-10-27:
> 

Sounds like a solid plan, was just about to start learning rust 😅

celsworth, 2022-10-30:
Haven't forgotten this :)

I tried an experiment this morning.. wrote a little socket proxy that logs all traffic going through it, with the intention of using it to sit between a connection from my inverter back to Luxpower.

Came to change the IP that the inverter connects to in their little Chinese "MiCO" web interface and.. it resets the IP of Network Setting 1 every time you change it, back to the Luxpower one. I note in the first post you said you managed to change this, not sure why I can't!

So plan b I guess is to route all wifi traffic from the inverter through a Pi and intercept it with iptables. What a faff. I'll experiment more with this next week.

JavonneM, 2022-10-30:
So yeah I managed to change the settings, initially I disabled the TCP Client and had changed it to TCP Server (That is why my network configuration is switched, ie nc1 is client and nc2 is server) Be sure to click the correct save button :upside_down_face: I was caught a few times clicking the bottom save button when I changed nc1 and then it just resaved nc2's configuration and dropped the new nc1 config :upside_down_face: 

So the current configuration used to get my inverted to connect to my pi, 

[image]

Running netcat server (Just to see what the inverter emits)

[image]

Seems to just be the heart beat, looks like its waiting for the server to send something.

[image]

Once I've setup my server I'll try to capture the handshake between luxpower server and the data logger with a proxy on my local machine.

celsworth, 2022-11-01:
Got a capture of the traffic between the inverter and Luxpower on initial connection. There really isn't that much to it actually.

So the inverter sends the heartbeat as you're observing, then the server sends back a reply that is identical, byte for byte. The server then sends a second reply which is slightly different; the protocol changes from 2 to 1 and there's an extra byte near the end (0x07). Not sure what that means.

I've replaced my serial here with XX.

```
08:04:07.820408 IP MiCO.44352 > 120.79.53.27.4346: Flags [P.], seq 1:20, ack 1, win 8760, length 19
  0x0000:  4500 003b 003b 0000 ff06 4863 c0a8 050c  E..;.;....Hc....
  0x0010:  784f 351b ad40 10fa 0000 196e 7a17 b261  xO5..@.....nz..a
  0x0020:  5018 2238 5c50 0000 a11a 0200 0d00 01c1  P."8\P..........
  0x0030:  XXXX XXXX XXXX XXXX XXXX 00              XXXXXXXXXX.

08:04:08.110217 IP 120.79.53.27.4346 > MiCO.44352: Flags [P.], seq 1:20, ack 20, win 29200, length 19
  0x0000:  4500 003b 7c47 4000 2c06 5f57 784f 351b  E..;|G@.,._WxO5.
  0x0010:  c0a8 050c 10fa ad40 7a17 b261 0000 1981  .......@z..a....
  0x0020:  5018 7210 0c65 0000 a11a 0200 0d00 01c1  P.r..e..........
  0x0030:  XXXX XXXX XXXX XXXX XXXX 00              XXXXXXXXXX.

08:04:08.615489 IP 120.79.53.27.4346 > MiCO.44352: Flags [P.], seq 20:40, ack 20, win 29200, length 20
  0x0000:  4500 003c 7c48 4000 2c06 5f55 784f 351b  E..<|H@.,._UxO5.
  0x0010:  c0a8 050c 10fa ad40 7a17 b274 0000 1981  .......@z..t....
  0x0020:  5018 7210 054f 0000 a11a 0100 0e00 01c3  P.r..O..........
  0x0030:  XXXX XXXX XXXX XXXX XXXX 0700            XXXXXXXXXX..
```

Then nothing until its time for the first data broadcasts, and at this point we again get an identical heartbeat reply

```
08:09:09.090456 IP MiCO.44352 > 120.79.53.27.4346: Flags [P.], seq 1:20, ack 1, win 8721, length 19
  0x0000:  4500 003b 005b 0000 ff06 4843 c0a8 050c  E..;.[....HC....
  0x0010:  784f 351b ad40 10fa 0000 1981 7a17 b288  xO5..@......z...
  0x0020:  5018 2211 5c3d 0000 a11a 0200 0d00 01c1  P.".\=..........
  0x0030:  XXXX XXXX XXXX XXXX XXXX 00              XXXXXXXXXX.
08:09:09.381333 IP 120.79.53.27.4346 > MiCO.44352: Flags [P.], seq 1:20, ack 20, win 29200, length 19
  0x0000:  4500 003b 7c65 4000 2c06 5f39 784f 351b  E..;|e@.,._9xO5.
  0x0010:  c0a8 050c 10fa ad40 7a17 b288 0000 1994  .......@z.......
  0x0020:  5018 7210 0c2b 0000 a11a 0200 0d00 01c1  P.r..+..........
  0x0030:  XXXX XXXX XXXX XXXX XXXX 00              XXXXXXXXXX.

08:09:10.225103 IP MiCO.44352 > 120.79.53.27.4346: Flags [P.], seq 20:137, ack 20, win 8702, length 117
[snip, this is the usual traffic I'd expect to see by this point]
```

There's no replies of substance to these packets, just TCP acks.

So as a starting point I wonder if we knock up a script to basically replace your netcat and reply with this, see if we then start getting data broadcasts?

celsworth, 2022-11-01:
Ok, here's a test. There's two janky Ruby scripts here, one will pretend to be luxpower's servers, one will pretend to be your inverter (just for testing really). Remove the .txt because Github was being awkward. What sort of a git repo doesn't support `.rb` uploads?!

As a proof of concept, run `ruby pretend_to_be_lux.rb`, then in another terminal run `ruby pretend_to_be_inverter.rb`.

The lux part will reply to the inverter with any packet that looks like a heartbeat (6th byte == 193, which is tcp_function Heartbeat). Any others are ignored. Everything is printed as an array of decimal integers so you can see what the inverter sends.

The inverter part sends a heartbeat, then some junk, then repeats. Its just a test that the lux part was doing what I wanted; you should see it print the hex decoded heartbeat replies. If you then connect the real inverter to the Ruby lux's socket, maybe more interesting things will happen.

Haven't tried this with my inverter yet but may do later.

[pretend_to_be_inverter.rb.txt](https://github.com/celsworth/lxp-bridge/files/9909225/pretend_to_be_inverter.rb.txt)
[pretend_to_be_lux.rb.txt](https://github.com/celsworth/lxp-bridge/files/9909226/pretend_to_be_lux.rb.txt)

JavonneM, 2022-11-01:
So this seems to have worked.....
When I say ACK I mean replying to the heart beat with another one as your script does (not a tcp ack)
The setup, lxp-bridge running in a container and then running `pretend_to_be_lux.rb`
upon the heartbeat reply lxp bridge output
- lxp_bridge::coordinator : TranslatedData(TranslatedData { datalog: xxxx, device_function: ReadHold, inverter: xxxx, register: 0, values: [bytes]
- mqtt: publishes hold registers 0 ->79
- TranslatedData(TranslatedData { datalog: xxxx, device_function: ReadInput, inverter: xxx, register: 0, values: [bytes]
- mqtt: publishes `lxp/xxxx/inputs/all`
- TranslatedData(TranslatedData { datalog: xxxx, device_function: ReadInput, inverter: xxxx, register: 127, values: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] })
[2022-11-01T20:49:01.212 ERROR lxp_bridge] unhandled ReadInput register=127 len=254
- lxp bridge crashes.
After the heart beat was sent and lxp_bridge crashed I closed the script, the data logger lights are all green still at this point, so 

So now after restarting lxp-bridge, it seems that the datalogger is constantly emitting the data, I don't need to send the heart beat again. However lxp_bridge is crashing on the register 127 read. So it seems we just need to ack the heart beat with the same payload to tell the data logger to start transmitting.

Just to provide some additional context, its a [Luxpower SNA5000 hybrid](https://luxpowertek.com/luxpower-inverter-sna-3-5k), not too sure what the firmware version its either the displayed properties SOFA + SOFB + SOFC (Software A/B/C) or CLAH + CLAL (Not sure what this is), 

After this I rebooted the datalogger in the web portal to see if the state persists, seems it has reverted to awaiting heart beat

Testing constant heart beat emission (ie not closing the script)
- Socket connected (inverter lights flashing ie disconnected)
- First beat ACK: inverter emits data (Inverter lights all green, connected)
- Second beat ACK: inverter emits data ( I assume in this case the inverter might wait for an ACK to sync the message or completely ignore it)

So it would seem that sending a heart beat only seems to tell the data logger that you may now start transmitting if you have not started already, subsequent.

JavonneM, 2022-11-01:
So I guess from lxp-bridges perspective when the lxp-bridge connects all that needs to be done is simply reply to the heart beats. I'll check tomorrow how long it lasts in this state, the inverter might stop sending data if you don't send an ACK back after a while.

celsworth, 2022-11-01:
Brilliant, thanks for testing this. So this seems like a promising outcome.

Actually the crash is already handled but in the next release: https://github.com/celsworth/lxp-bridge/commit/dbc0afbb2c5038fcec887a090296d00210a1e270

So I guess I should get on with it and get 0.9.0 released soon :) You can try using the `dev` docker tag in the meantime if you want though, that uses the latest `master`. https://hub.docker.com/layers/celsworth/lxp-bridge/dev/images/sha256-cab8487b5c2d8a19ed4d7a03be4aea06f40218d493a2242ccadc7f16d740475b?context=explore

For the heartbeats I think thats pretty easy to add. I'll take a look at doing that tonight/tomorrow. I think its probably preferable to default this to off, and have it enabled per inverter in the configuration, then suggest people only enable it if they're not connected to Luxpower for now; as we don't really know how things will react if an inverter gets two heartbeats :)

And no problem about it stopping after 2 heartbeats, I can have lxp-bridge respond to every heartbeat no problem.

celsworth, 2022-11-01:
The `dev` Docker image referred to above now has a PR merged that will enable heartbeats; set a `heartbeats` key to enable it:

```yaml
inverter:
  datalog: foo
  serial: bar
  heartbeats: true
  <snip>..
```

Let me know if it works :D

JavonneM, 2022-11-02:
Yeah used the image last night and everything seems to be good, data is flowing, thank you so much for the fix :pray:.
Feel free to close this issue whenever you want.

celsworth, 2022-11-02:
👍 brilliant. I'll not bother pursueing the scheduled `read_inputs` then as this seems the much more correct and cleaner solution.

Thanks for the help, I've been wondering for a year or two now what would happen if Luxpower disappeared, guess now we solved that eventual problem :)

closed 2022-11-02 by celsworth

celsworth, 2022-11-22:
> The data is refreshed to the app **every 10 seconds**, _and also to the network server that lxp-bridge is also listening to!_

Now that is pretty interesting, so Luxpower are presumably seeing data every 10 seconds too and logging it. I guess they don't care that much about it.

> I don't know if this is related to the heartbeats or something else? If lxp-bridge can do the same thing without the app, that could be useful.

I've tried a couple of times to see what the app does when directly connecting but I've not yet seen any data packet sent from the app to the inverter that triggers this data broadcast every 10 seconds :( I've tried quite a few things to catch this, including setting up a Pi as an access point with the same IP as the inverter, then listening on port 8000 and logging packets etc.. might have to have another think about this as this is something I'd like to work out.

Presumably if you disconnect the local tablet then the data frequency goes back to once every 5 minutes, immediately?

MrTimbones, 2022-11-22:
> Now that is pretty interesting, so Luxpower are presumably seeing data every 10 seconds too and logging it.

I don't know if data is being uploaded to the cloud every 10s, or just to the local network. Their website still only shows me 5 minute data, so either they're dumping 99% of the data they receive or they never see it.

> Presumably if you disconnect the local tablet then the data frequency goes back to once every 5 minutes, immediately?

Yes, it does. Sometimes I need to poke the tablet back into the right mode to get back to 10s data.

celsworth, 2022-11-23:
Ok, dug up an old Android tablet, installed app, installed "Packet Capture" (installs a fake VPN, quite clever, no root required), and tried Local Connect.

It actually seems a bit dodgy, at least for me it took a couple of attempts to get it going. But once it was receiving data every 10 seconds I have a packet capture so I can see what its doing.

Turns out it is just requesting input data every 10 seconds, which lxp-bridge can already do, so it would be automating that (pretty easy). I'm not sure about why the data doesn't appear on their web portal, I'm pretty sure that if lxp-bridge is seeing it, then they will be seeing it. Probably they just ignore any data that comes too soon after the last logged row.

The other small trick I see they use is in order to get the inverter to reply, you need to know the serial number of it; in lxp-bridge I just do this by configuration file. The app uses the name of the SSID of the WiFi you're connected to, it matches the datalogger stick serial; quite clever.

So in closing, yeah, I can add something that requests data every 10 seconds (can be configurable). It will be off by default and have a bit of a warning but if we're just doing the same thing the app does I don't see how they'd ever really be able to tell the difference.