celsworth/lxp-bridge #109: Do single writes cover everything?

Swwils, 2022-11-03:
I was looking at the lux front end on their website and it looks to me like that several registers wont be covered by single writes.

Is it worthwhile to consider taking a look at writeParam again and sending the entire parameter list like their website does? It looks to me like it reads them out from the inverter that way as it just loads the payload then reads each property in remoteRead.js

celsworth, 2022-11-03:
Do you mean the registers that contain times etc? There's an inverter command WriteMulti for that, but it's not currently supported in lxp-bridge.

I don't think it would be a huge problem to add support though, the main issue is how to expose it externally. Maybe something like the usual `cmd/$datalog/set/hold/X` but then with a payload of comma separated values?

So if you sent this: (don't try this yet, it won't work)

```
mosquitto_pub -t lxp/cmd/all/set/hold/68 -m '0,0'
```

Then lxp-bridge coudl set registers 68 and 69 both to zero (those are AC_CHARGE_START_HOUR / AC_CHARGE_START_MINUTE apparently).

Swwils, 2022-11-03:
No I am use exclusively https://server.luxpowertek.com/WManage/web/maintain/remoteSet

If you inspect this page they have left several js file that contain everything needed to read / write all params.

```
//Read All Parameter...
				if(currentDeviceType != 5) {		//Not LSP 100K
					showReadParamsResult(response);
					
					if(startRegister == 0) {
						setTimeout("readMultiParams(40, 40)", 1000);
					} else if(startRegister == 40) {
						setTimeout("readMultiParams(80, 40)", 1000);
					} else if(startRegister == 80) {
						setTimeout("readMultiParams(120, 40)", 1000);
					} else if(startRegister == 120) {
						setTimeout("readMultiParams(160, 40)", 1000);
					} if(currentDeviceType == 6 && startRegister == 160) {
						setTimeout("readMultiParams(200, 40)", 1000);
					} else {
						if(currentDeviceType == 3 && startRegister == 160 && allowCtrlHold601) {			//Off-Grid
							setTimeout("readMultiParams(600, 40)", 1000);
						} else {
							currentReadMultiParamsStartRegister = -1;
							$('#readAllParamButton').button('reset');
```

and

```
function showLspReadParamsResult(response) {
	if(response.hasOwnProperty('HOLD_MODEL')) {
		$('input[holdParam="HOLD_MODEL"]').val(response.HOLD_MODEL);
	}
	if(response.hasOwnProperty('HOLD_SERIAL_NUM')) {
		$('input[holdParam="HOLD_SERIAL_NUM"]').val(response.HOLD_SERIAL_NUM);
	}
	if(response.hasOwnProperty('HOLD_TIME')) {
		$('input[holdParam="HOLD_TIME"]').val(response.HOLD_TIME);
	}
	if(response.hasOwnProperty('HOLD_COM_ADDR')) {
		$('input[holdParam="HOLD_COM_ADDR"]').val(response.HOLD_COM_ADDR);
	}
	if(response.hasOwnProperty('HOLD_LANGUAGE')) {
		$('select[holdParam="HOLD_LANGUAGE"]').val(response.HOLD_LANGUAGE);
```

etc.

Basically when you press that "read" button you get 5 readMultiParams triggered that spit out a payload of every parameter from register 0 to 160 in blocks of 40.

`{"HOLD_GRID_VOLT_CONN_HIGH":"262.2","HOLD_GRID_VOLT_LIMIT2_LOW_TIME":"26","HOLD_MODEL_lithiumType":2,"HOLD_MODEL_powerRating":5,"FUNC_GRID_ON_POWER_SS_EN":true,"HOLD_RECONNECT_TIME":"30","FUNC_LSP_ISO_EN":false,"valueFrame":"pgoBADEzMzIwMDUwODdhQUFBAhocAQAAFgsDFCMwAQABAAAAAAAAAAQA1fLoAx4AHgDRBz4KjhIeFNEHPgp+ADMAMAexChoAGgAwB7EKGgA=","FUNC_LSP_LCD_REMOTE_DIS_CHG_EN":true,"FUNC_LVRT_EN":false,"FUNC_LSP_FAN_CHECK_EN":true,"FUNC_NEUTRAL_DETECT_EN":false,"FUNC_LSP_BAT_ACTIVATION_EN":true,"FUNC_FEED_IN_GRID_EN":true,"HOLD_LANGUAGE":"1","HOLD_GRID_VOLT_LIMIT1_LOW_TIME":"126","HOLD_MODEL_usVersion":0,"HOLD_MODEL_meterType":0,"HOLD_GRID_FREQ_CONN_HIGH":"51.5","HOLD_GRID_VOLT_LIMIT1_HIGH_TIME":"51","success":true,"HOLD_MODEL_meterBrand":0,"HOLD_MODEL_measurement":1,"FUNC_DCI_EN":true,"HOLD_MODEL_rule":6,"HOLD_GRID_FREQ_CONN_LOW":"47.5","FUNC_AC_CHARGE":true,"HOLD_COM_ADDR":"1","FUNC_OVF_LOAD_DERATE_EN":false,"HOLD_FW_CODE":"aAAA-1A1C","HOLD_GRID_VOLT_LIMIT2_HIGH":"273.7","FUNC_ANTI_ISLAND_EN":true,"FUNC_FORCED_CHG_EN":false,"FUNC_LSP_BYPASS_MODE_EN":false,"HOLD_MODEL":"0x10AA6","HOLD_MODEL_leadAcidType":0,"HOLD_TIME":"2022-11-03 20:35:48","FUNC_DRMS_EN":true,"HOLD_MODEL_ruleMask":0,"HOLD_GRID_VOLT_CONN_LOW":"200.1","HOLD_CONNECT_TIME":"30","HOLD_GRID_VOLT_LIMIT3_HIGH":"273.7","HOLD_GRID_VOLT_LIMIT3_LOW_TIME":"26","HOLD_START_PV_VOLT":"100","HOLD_PV_INPUT_MODE":"4","FUNC_GFCI_EN":true,"FUNC_FORCED_DISCHG_EN":false,"HOLD_GRID_VOLT_LIMIT2_HIGH_TIME":"26","FUNC_SW_SEAMLESSLY_EN":false,"HOLD_MODEL_batteryType":2,"HOLD_GRID_VOLT_LIMIT1_HIGH":"262.2","FUNC_EPS_EN":true,"HOLD_GRID_VOLT_LIMIT3_LOW":"184","HOLD_GRID_VOLT_LIMIT1_LOW":"200.1","HOLD_SERIAL_NUM":"1332005087","INPUT_BATTERY_VOLTAGE":"49.1","FUNC_ISO_EN":true,"FUNC_LSP_AC_CHARGE":false,`

Swwils, 2022-11-03:
And then if you simply HTML edit in a setting with the parameter name correct, you can write that to the inverter as it sends EVERY param no matter which set button you press.

Swwils, 2022-11-03:
For instance a div class with functionparam="FUNC_BUZZER_EN" would then allow you to control the buzzer. Despite this enable not being expose to regular user.

celsworth, 2022-11-03:
Oh nice, I'll have to look at their JS more closely, might learn a few things :)

But I've lost track of what you want lxp-bridge to do, just be able to set things like the buzzer? It can already write to any single register, just need to know which one the buzzer is for example.

Swwils, 2022-11-03:
No its not really a request more of a heads up i've seen the textual description of every register including how all the different model numbers work etc encase it was useful.

celsworth, 2022-11-03:
👍 definitely. JS isn't my strong point but its easy enough to read and as I say, might learn a few things :)

I've documented which registers I know in https://github.com/celsworth/lxp-bridge/blob/master/doc/LXP_REGISTERS.txt (though they have HOLD_ all prepended to them, but you should see some familiar looking ones). Will see if I can find any more.

Thanks!

Swwils, 2022-11-03:
It might be that the functionparams registers are changed by the time you see them.; I haven't yet looked at the correlation.

closed 2022-11-10 by celsworth