celsworth/lxp-bridge #112: Breaking changes planned for 1.0

celsworth, 2022-11-10:
Planned breaking changes for the 1.0 release:

* [ ] remove `v_pv` JSON keys (sum of v_pv_*, bit useless)
* [ ] remove `lxp/cmd/*/inputs/*` in favour of newer `lxp/cmd/*/input/*`

chriscn, 2022-11-11:
Is this likely to affect people who use just the Home Assistant function?

celsworth, 2022-11-11:
The two items planned so far, no probably not. The `v_pv` key isn't one of the more useful ones so I doubt anyone uses it at all, and the second item is just referring to when you publish to `lxp/cmd/all/inputs/1` to get the inverter to publish readings manually.

No need for panic though, I'm just collating ideas, there will be plenty of warning before any of this hits a release :) They might not even all make it.

If any breaking changes are made I'll fully document them in the wiki and release notes at the time.

chriscn, 2022-12-09:
Is there a timeline for v1.0.0?

celsworth, 2022-12-09:
I don't have a timeline for 1.0 at the moment no - still quite a lot of changes going on that revert or improve earlier changes in breaking ways and would be harder to do if we had to guarantee some sort of stability which usually comes with a 1.0 release.

Maybe some time next year if things settle down? (ie not so many new features coming and going all the time)