celsworth/lxp-bridge #113: Add more frequent requesting of input data

celsworth, 2022-11-23:
Expanding from discussions in #103, it's becoming clear that it would be nice to get input data a lot more frequently than the 5 minutes the inverter sends it out automatically.

This isn't a massive job but there a few technicalities that might make this a bit tricky than at first glance.

The mobile app requests data every 10 seconds in Local Connect mode, so we'll emulate this essentially. In order to not create a new data row on the Luxpower portal every 10 seconds, I believe only registers 0-79 (ie not 80-119) are requested by the app, so we'll do the same.

This creates a problem that the MQTT packet `lxp/{datalog}/inputs/all`, and database rows, and InfluxDB, are all only populated when we get those registers 80-119 (aka `inputs/3`). So we need a way to tell all these subprocesses that data is not arriving, and if we want to fill in the various databases, we need a value to put in those columns. MQTT could just use nulls, InfluxDB can be omitted, but the database columns are NOT NULL; perhaps it would be better to use the most recent `inputs/3` data, as those don't change very often.

Beenay25, 2023-03-21:
I might be misinterpreting what you're asking, but the following automation worked for me : 

```
alias: Poll Battery State 1 & 2
trigger:
  - platform: time_pattern
    seconds: /15
action:
  - service: mqtt.publish
    data:
      topic: lxp/cmd/<YOUR INVERTER ID>/read/inputs/1
      retain: true
  - service: mqtt.publish
    data:
      topic: lxp/cmd/<YOUR INVERTER ID>/read/inputs/2
      retain: true
mode: single

```

Replace `<YOUR INVERTER ID>` with the inverter's internal SSID (browse to the inverter's IP and look under **AP Mode Settings**)

It'll send out a MQTT request to the router every 15 seconds and gets a response sent out.

MrTimbones, 2023-03-22:
Thanks for the tip, @Beenay25 - I'm running my own python script to subscribe to messages from lxp-bridge, and I realised that I could add an extra bit of code to periodically request data whenever I wanted. This means I can go rescue my old tablet from continued burn-in. I'm hoping this method is more reliable too, as the official Luxpower app would sometimes fall offline and stop requesting frequent updates.

This also allows me to request inputs/1 (live watts) every 10 seconds and inputs/2 (cumulative kWh) every 5 minutes, as I don't need high resolution on the latter.

excieve, 2023-04-22:
I tried out the automation by @Beenay25, and while it does publish a message to the topic, it doesn't appear to save anything to InfluxDB, from which I'm building my Grafana dashboards.

Reading all 3 inputs works, but I understand it's undesirable since it also causes excessive writes to the remote Luxpower service DB.

@celsworth I wonder if it would be sensible for an MQTT command could be implemented in order to request an InfluxDB/database write? So that the user decides when they want the data to be persisted.

celsworth, 2023-04-23:
I wouldn't be opposed to that, that could work.

Few points to consider..

* if some of the data is stale, do we write it to Influx/database anyway or not? For example, if you're requesting inputs/1 and inputs/2 every few seconds, then requesting a db write, what do we do with the stale inputs/3 data
  * what do we determine as being stale, data not updated since last write? what about if only inputs/1 has been updated? There's maybe a few edge cases to think about.
  * if we want to be able to omit anything that is stale then the PR making some columns NULLable (#114) will need to perhaps be updated to make everything NULLable.. could get a bit messy

I kind of wonder if its trying to add a bit too much flexibility and making the job harder. My preferred solution is probably some sort of single routine to request inputs/1 and inputs/2 then store only that data. This could be hooked up to a new MQTT command or schedule. That's where I was trying to go with the initial PR making inputs/3 optional, so the database wouldn't cry over that missing data (which isn't very interesting to request every few seconds, and also spams Luxpower as you say).

apbarratt, 2023-05-11:
Another vote from me on this task.
I've noticed that when I click "read all" on the Chinese cloud interface, that my MQTT receives lots of updated data for sensor data too. So I'll give @Beenay25's suggestion a go and if all goes well, it'd be good if just prodding that one place will cause the rest to update more regularly :)

Ultimately I'd like to disconnect the cloud side all together.  There's just something that makes me a little nervous about having so much of my house's infrastructure at the mercy of a tech guy in China right now.  Geopolitics there are calm right now but all things solar etc always make me think of backup solutions should I ever need them 😅

apbarratt, 2023-05-12:
May I ask if the automation above by @Beenay25 is still working?  I've set up the same automation with my datalog ID and I'm still only seeing my entities in home assistant being updated every five minutes.  Have I missed a step somewhere?
I tried running a listener on the `lxp/[my ID]/inputs/1` but nothing is showing up there either :/

Beenay25, 2023-05-12:
It definitely still works for me!

> alias: Poll Battery State 1 (W)
> description: ""
> trigger:
> ..\- platform: time_pattern
> ....seconds: /15
> condition: []
> action:
> ..\- service: mqtt.publish
> ....data:
> ......topic: lxp/cmd/\<YOUR INVERTER ID\>/read/inputs/1
> ......retain: true
> mode: single

In the above automation you need to update \<YOUR INVERTER ID\>... so something like BA1234567

Ugh... I can't figure out how to attach code without it messing up the formatting. Don't have time to figure it out so have replaced indents with ...

excieve, 2023-05-13:
@celsworth Sorry for replying late — life took over :)

I totally agree regarding complexity of handling stale data.

> My preferred solution is probably some sort of single routine to request inputs/1 and inputs/2 then store only that data. This could be hooked up to a new MQTT command or schedule. That's where I [...]

Sounds like a good approach to me! I understand this needs #114 merged first. Is there something I could help with?

apbarratt, 2023-05-15:
Okay, I think I've found my problem, though I don't know how to fix it.
From what I can tell, this `retain: true` is important.  However, for some reason, everything I send, whether through a service on home assistant like above, or through a tool like MQTT Analyser on my phone, the `retain: true` flag is being utterly ignored.  Anything with `retain: true` applied to it is being immediately changed to `retain: false` somewhere.  I'm at a bit of a loss for how to fix this but I assume it's some configuration mistake with my mosquito addon :/

apbarratt, 2023-05-15:
One thing I'm a little confused by.  You guys are saying that every five minutes the system is only requesting inputs 1 and 2.  But that's not what I'm seeing when I monitor mine.  What I'm seeing is every five minutes I see data spat out by inputs/1, inputs/2, inputs/3, and most bizarrely, inputs/all.  I say bizarre for the fact that inputs/all contains everything that was included in the other three, making the other messages seem really rather redundant.

I wonder if my inverter is running different software to what has been seen previously and if this could provide clues as to why I'm not getting the behaviour you guys are describing.

When I log into the data module and check the version data it reads:

```
Version Information
Module Name
EMW1290
MAC
84:9D:C2:[hidden for privacy]
SDK Version
MX1290002.032.002
Software Version
ATV1.20@MX1290
Web Version
web_V1.0.0
```

apbarratt, 2023-05-15:
Oh and I've just spotted this on the landing "Run State" page:

```
Command Mode State
Function: Disable
``` 

I wonder if that means commands of any kind are simply not allowed.  I can't seem to find any setting that corresponds to this "Command Mode" to "enable" it :/

apbarratt, 2023-05-17:
I have not been able to reproduce the issue I was having at all.  It just seems to be all working now since restarting the container.  I have gone ahead and closed the other ticket for now.  Please do try restarting your container though if you have the same issue of commands not being respected.  In the meantime, the automation suggested above by [Beenay25](https://github.com/Beenay25) does indeed work perfectly now.

sabelosimelane, 2023-05-29:
> I might be misinterpreting what you're asking, but the following automation worked for me :
> 
> ```
> alias: Poll Battery State 1 & 2
> trigger:
>   - platform: time_pattern
>     seconds: /15
> action:
>   - service: mqtt.publish
>     data:
>       topic: lxp/cmd/<YOUR INVERTER ID>/read/inputs/1
>       retain: true
>   - service: mqtt.publish
>     data:
>       topic: lxp/cmd/<YOUR INVERTER ID>/read/inputs/2
>       retain: true
> mode: single
> ```
> 
> It'll send out a MQTT request to the router every 15 seconds and gets a response sent out.

Thanks, this made my life easier

excieve, 2023-06-10:
@apbarratt So I was a bit puzzled by this automation too. While I do get MQTT responses on the `inputs/1` and `inputs/2` topics, I don't get any updates at the configured interval in the HA entities — they still keep coming at 5 minute intervals when Luxpower portal queries everything.

It seems to be by design, since HA integration in `lxp-bridge` [ties](https://github.com/celsworth/lxp-bridge/blob/b421b717d8ceb65b4af862fca9adb71579675b20/src/home_assistant.rs#L218) into the `inputs/all` topic, which only publishes [after](https://github.com/celsworth/lxp-bridge/blob/b421b717d8ceb65b4af862fca9adb71579675b20/src/coordinator/mod.rs#L379) `inputs/1`, `inputs/2` _and_ `inputs/3` were queried.

So if I understand this right, this isn't really a workaround — either `input/3` needs to be queried often too or partial saves/publishes to `all` should be allowed.

raldred, 2023-10-17:
Sorry, i just wanted to clarify, I'm wondering if its possible to pull data more frequently as the lux app does in local connect mode.

The automation above, is this also triggering updates to be sent to lux? I don't expect them to be too happy about being thrown data every 15 seconds.

celsworth, 2023-10-18:
> Sorry, i just wanted to clarify, I'm wondering if its possible to pull data more frequently as the lux app does in local connect mode.
> 
> The automation above, is this also triggering updates to be sent to lux? I don't expect them to be too happy about being thrown data every 15 seconds.

Yes it does. The inverter sends all its packets to every connected client, no matter who requested that data. So Luxpower servers will see it (and presumably log it) if you request it every few seconds. How much of a problem this is, who knows :)

To get deeper into the detail, they only seem to log a row to their database when they see the third data packet (the one we call `inputs/3`. So if you only request 1 & 2 every 15 seconds (which have most of the interesting details anyway, 3 only has things that change very rarely), they won't log it - this is actually what the local app does. When you have the local app running they're still seeing the 1 & 2 packets, but no 3, so no logging happens.

On a broader note I'd suggest considering firewalling the inverter off so it has no Internet connectivity anyway, a few people are doing that successfully and we can still monitor it via the local heartbeats option.

garthy, 2023-10-26:
So not sure if you are aware of this but it was posted in the Facebook [Lux Power Tek Owners Group - UK](https://www.facebook.com/groups/438696100079367/)

I would be careful in bumping up the refresh to high unless until all is understood

[image]

apbarratt, 2023-10-26:
Indeed, it is quite ridiculous but it's pushed me to kill off the Chinese server access on my dongle and it's been running perfectly for over 12 hours now, confirming that lxp-bridge's hearbeats are indeed keeping things ticking along nicely.
Before
[image]
After
[image]

Beenay25, 2023-10-26:
> Indeed, it is quite ridiculous but it's pushed me to kill off the Chinese server access on my dongle and it's been running perfectly for over 12 hours now, confirming that lxp-bridge's heartbeats are indeed keeping things ticking along nicely. Before [image] After [image]

I've done this and it works nicely. It's a shame as the LXP app is occasionally useful.

celsworth, 2023-10-26:
Also see https://github.com/celsworth/lxp-bridge/discussions/213 for more discussion about this.

The official post is a bit ambiguous, I'm not sure if they mean us or not with "Home Assist App", and presumably "ping settings" means the update frequency. But yes I definitely agree that keeping all the data local is beneficial for multiple reasons - ideally we'd aim to replace the LXP app entirely and be able to do everything locally.

JohnMacrae, 2023-10-26:
> Indeed. I used it the other day to tweak the schedule after Octopus
> increased my cheap electric hours. I think that's doable through bridge
> too, but it was quicker to just spin up the app, though it was the first
> time in a while. It would have been nicer to find a toggle to turn the
> Chinese server off or on instead of having to delete it. I guess I'll just
> need to come back here and check my screenshot should I ever need it again
> 😅

I only keep the app in case, for some reason, I can't get through to my VPN
('cos some numpty has blocked it on their public wifi) - lol.  However, all
this talk of closing down APIs is just scaremongering - there's a very
simple fix if they could be bothered.

The only thing I'm missing in my setup is some decent data logging - anyone
got a decent node-red setup for squirting into influx?

Kind Regards
John

celsworth, 2023-10-26:
> The only thing I'm missing in my setup is some decent data logging - anyone got a decent node-red setup for squirting into influx?

You've seen lxp-bridge can do this directly? https://github.com/celsworth/lxp-bridge/wiki/Configuration#influx

raldred, 2023-10-27:
What's the benefit of logging to influx over what home assistant provides?

JohnMacrae, 2023-10-28:
Yes, Chris - somehow unable to make it work though.

Kind Regards
John

tseaward, 2023-11-01:
> Yes it does. The inverter sends all its packets to every connected client, no matter who requested that data. So Luxpower servers will see it (and presumably log it) if you request it every few seconds. How much of a problem this is, who knows :)

Do they still receive the packets if you configure lxp-bridge to connect directly to the dongle AP, e.g. 10.10.10.1? Similar to using Local Monitor in the app.

raldred, 2023-11-01:
I reckon so @tseaward because if I connect my app to local connect, my HA receives updates about every 10 seconds

celsworth, 2023-11-03:
Yes @tseaward I think @raldred is correct. I think no matter the source of the request, all connected clients on all wifi networks see the responses.

If you're connected to the inverters wifi itself though you could then remove it from your house wifi to keep it entirely isolated if you wanted. lxp-bridge heartbeats would need to be enabled to keep receiving data in that case.

jgulick48, 2024-03-04:
To add to this, I have a local poller that requests data from inputs 1 and 2 but it appears the announce to HomeAssistant has all data being polled from the /all topic. It would be nice if the various components could be updated to point to either inputs 1 or 2 so we could get more frequent updates within Home Assistant itself.

tseaward, 2024-03-04:
I have been running the following setup successfully for about six months now. Raspberry pi with WiFi connected to inverter's WiFi, ethernet connected to internet. 

It polls 1-3 every 15 seconds which home assistant records and updates my dashboard, however Lux online still only receives every 5 minutes.

Here's an example of /1. It's important to repeat the below for 2 AND 3. 

service: mqtt.publish
data:
  topic: lxp/cmd/BA32502751/read/inputs/1
  retain: true

lidgaca, 2024-05-24:
Is it possible for me to request a summary of this thread ... I've been through it and I'm not certain I fully understand all of the points raised. I don't have a background in MQTT and as such some of the terminology and the config files included here are a little cryptic.

What I would like to achieve

- Use influxdb as the recording back end (as I already understand it, and grafana)
- Collect data at a rate of ~ 1 min intervals - faster than the Lux servers provide. 

As I understand the thread, people are able to do this, but use MQTT to schedule reading the data from the inverter ... Due to some clever reverse engineering you have found that if you only ask to read inputs 1 and 2 (NOT inputs 3) this will not cause the Lux servers to log the record.

So is there any way I can accomplish this without using MQTT ? I suspect not, but would like a confirmation of that point. If I have to use MQTT to schedule the read, can I also continue to log data inside influxdb ? I suspect that this is possible, and if so would probably be my way forward.

If as I suspect, I have to use MQTT, I guess I'll need to do some more research ... Looks like the bank holiday is going to be a google-fest :(

-- Chris

jgulick48, 2024-05-24:
> Is it possible for me to request a summary of this thread ... I've been through it and I'm not certain I fully understand all of the points raised. I don't have a background in MQTT and as such some [...]

I have some separate code set up for this specifically. I can put a guide together if needed but https://hub.docker.com/repository/docker/jgulick48/lxp-bridge-statsd/general will subscribe to the mqtt topic, send the correct message to re-trigger the reading, and when a new message is sent to mqtt it will push those to statsd so a telegraf instance can then write to influxdb. Right now it polls ever 10 seconds but I could make that configurable easy enough.

crug80, 2024-06-05:
> I have some separate code set up for this specifically. I can put a guide together if needed but https://hub.docker.com/repository/docker/jgulick48/lxp-bridge-statsd/general will subscribe to the [...]

A guide is appreciated, thx! :)

lidgaca, 2024-06-05:
Hiya

I started to look at this but life (birthdays, trips to the vet, new tyres,
etc) got in the way.

Thanks  for the information ... It does look rather more complicated than I
thought.

Im going to do the stop-and-think thing for a while as I consider the extra
work involved (both for me, and the pi data logger).

Thanks.

-- Chris