celsworth/lxp-bridge #116: Homeassistant auto-discovery commands for common operations

lupine, 2022-11-24:
I'm running my inverter without the dubious benefit of the vendor's application, and it's not allowed to connect to anything outside my house, so I'm having to figure out how to control it via the MQTT messages lxp-bridge supports.

To get started more easily, it would be neat to encode this knowledge into homeassistant autodiscovery messages, so toggles and gauges magically appear to do the simple tasks. Right now, the only messages set up sensor topics, but HA *does* support command topics as well: https://www.home-assistant.io/integrations/mqtt/#entities-with-command-topics

A toggle to enable/disable AC charging and forced discharge, and some fields to allow the charging periods to be configured, would get us 90% of the way there, I think.

Happy to work on this if you'd accept a PR for it, just let me know!

The other option is a bunch of shell scripts that wrap `mosquitto_pub` :trollface:

celsworth, 2022-11-25:
This is definitely something I'd like to see added.

Obviously you've probably seen that enabling/disabling AC charge and forced discharge are fairly trivial already via dedicated cmd topics, so some of the work is really just constructing a new payload that references though, like the existing ones in `src/home_assistant.rs`.

It looks like the hardest part will be the state topic; the main problem is that the toggles such as AC charge and forced discharge are in a register that uses individual bits to represent whether they're on/off: https://github.com/celsworth/lxp-bridge/blob/master/doc/LXP_REGISTERS.txt#L37 (not moved this doc to wiki yet)

So currently the state topic would be `lxp/{datalog}/hold/21` and you'd have to `& 128` the result to check if the relevant bit is set, and I'm guessing thats not really within the realms of HA (not a regular user of it, so not really sure).

This is most of the reason why the dedicated cmd topics (`lxp/cmd/{datalog}/set/ac_charge`) exist, so the user doesn't  have to worry too much about the individual bits for the most common operations. (Retrieving, updating, and then re-setting register 21 with the new value would have exactly the same effect).

I'm just considering breaking out the individual bits and sending an MQTT message for each of them (or a JSON hash of them). Need to think of a suitable topic for it to live under.. always the hardest part!

celsworth, 2022-11-25:
@lupine offtopic for this but would also be interested to confirm something we've discovered in a recent issue here; if you don't allow your inverter to connect to China at all, then it doesn't send regular power data until we respond to a heartbeat. Did you find the same thing and did enabling the new `heartbeats` configuration option spring it all into life?

Or did you not have this problem, or have an alternate solution?

lupine, 2022-11-25:
Yeah, I was thinking this would be an easy way to get started with the codebase. I don't use homeassistant much either, but I've got a docker container for it and it'd be a quick and easy way to get started. It's also kind of an unofficial interop standard these days.  I'll see what I can put together. There are bitfield operators in HA's templating: https://www.home-assistant.io/docs/configuration/templating/#using-templates-with-the-mqtt-integration but I agree, individual topics would be easier.

Yes, I'm using `heartbeats: true`, and it's working fine. Without it, I wasn't getting any data updates. A five-minute update period is a bit long for me ;) but I can see you're working on that already. Writing to param register 0 isn't known / doesn't work?

celsworth, 2022-11-25:
> Yes, I'm using `heartbeats: true`, and it's working fine. Without it, I wasn't getting any data updates. A five-minute update period is a bit long for me ;) but I can see you're working on that already. Writing to param register 0 isn't known / doesn't work?

👍 good to get confirmation. Writing to datalog param register 0 hasn't been tried. WriteParam isn't hooked up yet as it feels like quite a risky operation.. want to be a guinea pig? :) I'll get it wired in if so.

As a whole the params are a bit unknown. The only one that even responds on my (older) inverter is 0. I've seen the app and web portal requesting various other ones, presumably for newer inverters or firmwares.

lupine, 2022-11-25:
Yeah, happy to give it a go. At worst, I have to buy a new datalogger, I guess ^^. 

Something else I'm struggling with is working out how to configure the inverter to use PV to cover house load, then send any excess to battery. Right now it's in 0x04 status, "pv supporting load first; surplus into grid". I don't see an obvious status value for the mode I'd like to be the default. It seems like such an obvious configuration - does it not exist?

celsworth, 2022-11-25:
I thought that was the default mode to be honest. I don't have a hybrid inverter, only a 3600ACS, so there's no PV, just battery. So my experience here is a bit limited.

The only configurable thing around this that I remember seeing though is "Charge Priority" (again in register 21, as it happens), which means charge the batteries before supplying the house load. When that's off (the default) it supplies the house load first, then charges the battery.

In both modes it should only export to grid when both sinks are satisfied. Are you saying its not charging your battery but still exporting? Any chance its not happy with the battery for any reason? Does it use the battery for discharging when PV isn't producing?

lupine, 2022-11-25:
Well, the inverter was in mode 4 / `00000100`, and generation was less than consumption.

Generation has just gone above consumption, and it's now switched to mode  0x0c / 12 / `00001100` - and the battery is charging! So I guess this is the status I was expecting, and I was getting misled by the text assigned to 0x04. 

Happy to help out exploring and improving the project for the pv+battery case. 

For the `WriteParam` PR, the code does seem to have worked to set the value, and I'm now seeing updates every minute, rather than every 5. It did report:

```
Nov 25 09:35:31 ur.gs lxp-bridge[905741]: [2022-11-25T09:35:31.929 INFO lxp_bridge::mqtt] publishing: lxp/result/{{datalogger}}/set/param/0 = FAIL
```

but subsequent reads of the param show the value is now 60:

```
Nov 25 09:42:51 ur.gs lxp-bridge[905741]: [2022-11-25T09:42:51.411 INFO lxp_bridge::mqtt] publishing: lxp/{{datalogger}}/param/0 = 60
```

and I'm seeing a minute between `inputs/all` messages, rather than 5 minutes.

celsworth, 2022-11-25:
Awesome. Don't worry about the FAIL too much, that just means we didn't get back the expected response, but if it's set it then all good. I might actually remove those `result` topics, they're a pain in the ass code-wise. If anyone wants to check if something worked, just read the param back again.

lupine, 2022-11-25:
It parsed this response to the WriteParam request, if I'm interpreting it right:

```
Nov 25 09:35:21 ur.gs lxp-bridge[905741]: [2022-11-25T09:35:21.558 DEBUG lxp_bridge::coordinator] RX: WriteParam(WriteParam { datalog: {{datalogger}}, register: 0, values: [0, 0] })
```
There's a line for the raw packet too, but I think that includes the serial :sweat_smile:

celsworth, 2022-11-25:
Ah right, so values 0, 0 is wrong. It would have expected to see 60, 0 there.

Yes the entire packet does include serial. I'm not sure how much of a danger posting those is but I tend to try and keep mine private anyway :)

I just based this on the existing set_hold implementation; when you write a regular holding register, the inverter replies with the new value so we use that to be sure the value was accepted.

Seems like this setting params might use a different schema. What are the last 6 or so bytes of your response packets? That should be safe to paste; the entire packet is something like:

2 bytes of prefix
2 bytes of protocol version
2 bytes of length
1 byte of address
1 byte of "tcp function"
10 bytes of serial
then after this I'm unsure, this is what its not parsing properly, probably

celsworth, 2022-11-25:
> Well, the inverter was in mode 4 / `00000100`, and generation was less than consumption.
>
> Generation has just gone above consumption, and it's now switched to mode 0x0c / 12 / `00001100` - and the [...]

👍 Apologies for the text then, it's more or less directly copied from a "Chinglish" inverter datasheet I got hold of! We can probably improve the translation :)

lupine, 2022-11-25:
OK, here's a redacted successful set of the param to 60 seconds:

```
Nov 25 09:35:20 ur.gs lxp-bridge[905741]: [2022-11-25T09:35:20.924 DEBUG lxp_bridge::mqtt] RX: Message { topic: "cmd/{{datalogger}}/set/param/0", payload: "60" }
Nov 25 09:35:20 ur.gs lxp-bridge[905741]: [2022-11-25T09:35:20.924 DEBUG lxp_bridge::coordinator] parsed command WriteParam(Inverter { enabled: true, host: "{{host}}", port: 8000, serial: {{serial}}, datalog: {{datalogger}}, heartbeats: Some(true) }, 0, 60)
Nov 25 09:35:20 ur.gs lxp-bridge[905741]: [2022-11-25T09:35:20.924 DEBUG lxp_bridge::lxp::inverter] inverter {{datalogger}}: TX [161, 26, 2, 0, 18, 0, 1, 196, XX, XX, XX, XX, XX, XX, XX, XX, XX, XX, 0, 0, 2, 0, 60, 0]
Nov 25 09:35:21 ur.gs lxp-bridge[905741]: [2022-11-25T09:35:21.558 DEBUG lxp_bridge::lxp::packet_decoder] 21 bytes in: [161, 26, 2, 0, 15, 0, 1, 196, XX, XX, XX, XX, XX, XX, XX, XX, XX, XX, 0, 0, 0]
Nov 25 09:35:21 ur.gs lxp-bridge[905741]: [2022-11-25T09:35:21.558 DEBUG lxp_bridge::coordinator] RX: WriteParam(WriteParam { datalog: {{datalogger}}, register: 0, values: [0, 0] })
Nov 25 09:35:31 ur.gs lxp-bridge[905741]: [2022-11-25T09:35:31.542 DEBUG lxp_bridge::lxp::packet_decoder] 19 bytes in: [161, 26, 2, 0, 13, 0, 1, 193, XX, XX, XX, XX, XX, XX, XX, XX, XX, XX, 0]
Nov 25 09:35:31 ur.gs lxp-bridge[905741]: [2022-11-25T09:35:31.929 INFO lxp_bridge::mqtt] publishing: lxp/result/{{datalogger}}/set/param/0 = FAIL
Nov 25 09:35:31 ur.gs lxp-bridge[905741]: [2022-11-25T09:35:31.929 DEBUG rumqttc::state] Publish. Topic = lxp/result/{{datalogger}}/set/param/0, Pkid = 28, Payload Size = 4
```

And here's a redacted failed set:

```
Nov 25 10:00:40 ur.gs lxp-bridge[905741]: [2022-11-25T10:00:40.276 DEBUG lxp_bridge::mqtt] RX: Message { topic: "cmd/{{datalogger}}/set/param/0", payload: "10" }
Nov 25 10:00:40 ur.gs lxp-bridge[905741]: [2022-11-25T10:00:40.277 DEBUG lxp_bridge::coordinator] parsed command WriteParam(Inverter { enabled: true, host: "10.0.2.102", port: 8000, serial: {{serial}}, datalog: {{datalogger}}, heartbeats: Some(true) }, 0, 10)
Nov 25 10:00:40 ur.gs lxp-bridge[905741]: [2022-11-25T10:00:40.277 DEBUG lxp_bridge::lxp::inverter] inverter {{datalogger}}: TX [161, 26, 2, 0, 18, 0, 1, 196, XX, XX, XX, XX, XX, XX, XX, XX, XX, XX, 0, 0, 2, 0, 10, 0]
Nov 25 10:00:42 ur.gs lxp-bridge[905741]: [2022-11-25T10:00:42.139 DEBUG lxp_bridge::lxp::packet_decoder] 21 bytes in: [161, 26, 2, 0, 15, 0, 1, 196, XX, XX, XX, XX, XX, XX, XX, XX, XX, XX, 0, 0, 1]
Nov 25 10:00:42 ur.gs lxp-bridge[905741]: [2022-11-25T10:00:42.140 DEBUG lxp_bridge::coordinator] RX: WriteParam(WriteParam { datalog: {{datalogger}}, register: 0, values: [0, 1] })
Nov 25 10:00:51 ur.gs lxp-bridge[905741]: [2022-11-25T10:00:51.279 INFO lxp_bridge::mqtt] publishing: lxp/result/{{datalogger}}/set/param/0 = FAIL
Nov 25 10:00:51 ur.gs lxp-bridge[905741]: [2022-11-25T10:00:51.279 DEBUG rumqttc::state] Publish. Topic = lxp/result/{{datalogger}}/set/param/0, Pkid = 31, Payload Size = 4
```

Just let me know if they don't make sense, I've tried to remove unrelated ReadInput and Heartbeat-related lines and might have been overenthusiastic!

lupine, 2022-11-25:
(Sorry, I guess the WriteParam discussion would have been better off in #22!)

celsworth, 2022-11-25:
Thanks for that, that does make sense, not at all what I expected to see though! Does seem correct that a success just returns 0 and a fail returns 1. Difficult to tell if that `0, 0, 1` should be interpreted as `0, 0` then `1` or `0` then `0, 1` though. Ultimately it doesn't really matter, the important thing is we can set the register and it has the desired effect.

Yeah we're a bit offtopic but never mind, I think we're done with it now. I'll have a think about whether to just remove the `result` topics (I'd like to, not sure if anyone will even notice).

lupine, 2022-11-25:
The first time I did a set command, I was watching `mosquitto_sub` and the result topic did give me some immediate relief that I'd done it right :sweat_smile:

Just a pic of the grafana dashboard showing the improved (20-second) data resolution:

[image]

I was getting 5 second resolution out of the Solax, so there's still scope to enhance this via the polled ReadInputs approach you're working on.

celsworth, 2022-11-25:
> I was getting 5 second resolution out of the Solax, so there's still scope to enhance this via the polled ReadInputs approach you're working on.

Be interesting to see how this plays with lower param 0 setting. The polled approach is pretty rudimentary; basically just send a "read inputs" packet every x seconds. You can bodge this in right now by putting a couple of mosquitto_pub in a bash loop.

The complication (there's always one) is only because ideally we should only request registers 0-79 and not 80-119 to exactly copy what the app does, and the MQTT `inputs/all` topic is only published when we get registers 80-119. More discussion in #113, lets not derail again :D

celsworth, 2022-11-26:
I made a start on this today. First step is decoding some registers so the state topics in HA have something to listen to. It looks a bit like this:

```
# mosquitto_pub -t lxp/cmd/all/read/hold/21 -m ''
# mosquitto_pub -t lxp/cmd/all/read/hold/110 -m ''

<in another window>
#mosquitto_sub -t lxp/# -v

lxp2/cmd/all/read/hold/21 (null)
lxp2/BA00000000/hold/21 62421
lxp2/BA00000000/hold/21/bits {"eps_en":true,"ovf_load_derate_en":false,"drms_en":true,"lvrt_en":false,"anti_island_en":true,"neutral_detect_en":false,"grid_on_power_ss_en":true,"ac_charge_en":true,"sw_seamless_en":true,"set_to_standby":true,"forced_discharge_en":false,"forced_charge_en":false,"iso_en":true,"gfci_en":true,"dci_en":true,"feed_in_grid_en":true}
lxp2/result/BA00000000/read/hold/21 OK

lxp2/cmd/all/read/hold/110 (null)
lxp2/BA00000000/hold/110 2
lxp2/BA00000000/hold/110/bits {"ub_pv_grid_off_en":false,"ub_run_without_grid":true,"ub_micro_grid_en":false}
lxp2/result/BA00000000/read/hold/110 OK
```

Not exactly sure what all of them mean to be honest (more datasheet copying). But I did try toggling the AC Charge one and got the correct result reflected in the new JSON.

celsworth, 2022-12-06:
Were you still interested in working on this @lupine? If not I might make a start at it. I've added a couple more MQTT topics recently that should assist in the creation of this.

lupine, 2022-12-06:
@celsworth yes, but I don't know when I'd be able to get to it, so I won't be sad if you want to do it before I can put something together :sweat_smile:

lupine, 2022-12-07:
Thanks for working on this!

Hmm, not far off, it seems. The switch shows up in HA just fine:

[image]

[image]

But it has status "unknown" - I didn't see any messages to `lxp/XXXXXXXXXX/hold/21/bits` being published throughout.

Pressing the `off` state causes a message to be published:

```
lxp/XX12345678/set/ac_charge OFF
```
and pressing the "on" state causes:

```
lxp/XX12345678/set/ac_charge ON
```
I didn't see any response to this, or any indication that lxp-bridge had received the message from mqtt, in the logs, even at debug level.

I just use a tiny docker script to run HA locally, btw:

```sh
#!/bin/sh

mkdir -p /home/lupine/ha/config

docker run -d \
  --name homeassistant \
  --privileged \
  --restart=unless-stopped \
  -e TZ=Europe/London \
  -v /home/lupine/ha/config:/config \
  --network=host \
  ghcr.io/home-assistant/home-assistant:stable
```

There's a short setup process, then you can add the mqtt integration and see it come through.

celsworth, 2022-12-07:
Oh I'm an idiot. The `command_topic` should be `lxp/cmd/XX12345678/set/ac_charge`. I've pushed a fix to the PR.

I'll have a look at getting HA running tomorrow.

celsworth, 2022-12-07:
For initial state, I thought it might request the `state_topic` on startup.. if not we might need to think of a workaround for that.

lupine, 2022-12-07:
OK, now we're cooking with gas. State is still unknown, but pressing the on switch gets:

```
lxp/cmd/XX12345678/set/ac_charge ON
lxp/XX12345678/hold/21 62164
lxp/XX12345678/hold/21/bits {"eps_en":false,"ovf_load_derate_en":false,"drms_en":true,"lvrt_en":false,"anti_island_en":true,"neutral_detect_en":false,"grid_on_power_ss_en":true,"ac_charge_en":true,"sw_seamless_en":false,"set_to_standby":true,"forced_discharge_en":false,"charge_priority_en":false,"iso_en":true,"gfci_en":true,"dci_en":true,"feed_in_grid_en":true}
lxp/XX12345678/inputs/1 {"status":32,"v_pv":5.5,"v_pv_1":5.3,"v_pv_2":0.0,"v_pv_3":0.2,"v_bat":52.5,"soc":54,"soh":100,"p_pv":0,"p_pv_1":0,"p_pv_2":0,"p_pv_3":0,"p_charge":0,"p_discharge":0,"v_ac_r":243.8,"v_ac_s":128.1,"v_ac_t":0.0,"f_ac":49.97,"p_inv":0,"p_rec":0,"pf":1.0,"v_eps_r":243.8,"v_eps_s":-1357.5,"v_eps_t":10.6,"f_eps":49.97,"p_eps":0,"s_eps":0,"p_to_grid":0,"p_to_user":2426,"e_pv_day":3.6,"e_pv_day_1":3.6,"e_pv_day_2":0.0,"e_pv_day_3":0.0,"e_inv_day":3.1,"e_rec_day":0.0,"e_chg_day":0.7,"e_dischg_day":0.5,"e_eps_day":0.0,"e_to_grid_day":0.1,"e_to_user_day":23.9,"v_bus_1":341.9,"v_bus_2":303.5,"time":1670444125,"datalog":"XX12345678"}
lxp/XX12345678/hold/21 62164
lxp/XX12345678/hold/21/bits {"eps_en":false,"ovf_load_derate_en":false,"drms_en":true,"lvrt_en":false,"anti_island_en":true,"neutral_detect_en":false,"grid_on_power_ss_en":true,"ac_charge_en":true,"sw_seamless_en":false,"set_to_standby":true,"forced_discharge_en":false,"charge_priority_en":false,"iso_en":true,"gfci_en":true,"dci_en":true,"feed_in_grid_en":true}
lxp/result/XX12345678/set/ac_charge OK
```

My dashboard shows that the power is flowing.

Pressing off:

```
lxp/cmd/XX12345678/set/ac_charge OFF
lxp/XX12345678/hold/21 62164
lxp/XX12345678/hold/21/bits {"eps_en":false,"ovf_load_derate_en":false,"drms_en":true,"lvrt_en":false,"anti_island_en":true,"neutral_detect_en":false,"grid_on_power_ss_en":true,"ac_charge_en":true,"sw_seamless_en":false,"set_to_standby":true,"forced_discharge_en":false,"charge_priority_en":false,"iso_en":true,"gfci_en":true,"dci_en":true,"feed_in_grid_en":true}
lxp/XX12345678/hold/21 62036
lxp/XX12345678/hold/21/bits {"eps_en":false,"ovf_load_derate_en":false,"drms_en":true,"lvrt_en":false,"anti_island_en":true,"neutral_detect_en":false,"grid_on_power_ss_en":true,"ac_charge_en":false,"sw_seamless_en":false,"set_to_standby":true,"forced_discharge_en":false,"charge_priority_en":false,"iso_en":true,"gfci_en":true,"dci_en":true,"feed_in_grid_en":true}
lxp/result/XX12345678/set/ac_charge OK
```

Charge power drops back down to 0W as you'd expect.

Interesting that even with a hold/21/bits message being published, HA leaves the state as "unknown".

celsworth, 2022-12-07:
Could be expecting `"ON"` and `"OFF"` as values rather than `true`/`false`.

I see there's https://www.home-assistant.io/integrations/switch.mqtt/#state_off (and on) but not sure if they have to be strings or an actual `true`/`false` will work.. Worth a try I suppose, I'll push to the PR and we can see if it works..

celsworth, 2022-12-07:
Or, if that doesn't work, an alternative would be to just change the `true` in the `21/bits` message to `"ON"`. Nothing says it has to be a JSON bool I guess, and its so new nobody will be relying on it yet (I hope).

lupine, 2022-12-07:
Well, the control still works:

```
lxp/cmd/XX12345678/set/ac_charge true
lxp/XX12345678/hold/21 62036
lxp/XX12345678/hold/21/bits {"eps_en":false,"ovf_load_derate_en":false,"drms_en":true,"lvrt_en":false,"anti_island_en":true,"neutral_detect_en":false,"grid_on_power_ss_en":true,"ac_charge_en":false,"sw_seamless_en":false,"set_to_standby":true,"forced_discharge_en":false,"charge_priority_en":false,"iso_en":true,"gfci_en":true,"dci_en":true,"feed_in_grid_en":true}
lxp/XX12345678/hold/21 62164
lxp/XX12345678/hold/21/bits {"eps_en":false,"ovf_load_derate_en":false,"drms_en":true,"lvrt_en":false,"anti_island_en":true,"neutral_detect_en":false,"grid_on_power_ss_en":true,"ac_charge_en":true,"sw_seamless_en":false,"set_to_standby":true,"forced_discharge_en":false,"charge_priority_en":false,"iso_en":true,"gfci_en":true,"dci_en":true,"feed_in_grid_en":true}
lxp/result/XX12345678/set/ac_charge OK
```

but the state still shows as unknown in HA.

I found https://community.home-assistant.io/t/understanding-state-topic/185757/11 which seems to be grappling with some of the same problems, but I'm out of time to fiddle with this particular thing today, sadly.

lupine, 2022-12-07:
Hmm, now the control has vanished from HA altogether. Very peculiar :thinking:

celsworth, 2022-12-07:
Seems like that last commit is a step back then, I'll revert that and have a play tomorrow with a local HA. Looking into converting the bools into strings in the `bits` message.

celsworth, 2022-12-07:
Done, another commit pushed to PR. I'll have a go at testing it tomorrow if you've not beaten me to it.

Also noticed I could set an `availability` object so HA knows when lxp-bridge is running or not, so I'll get that added soon too.

celsworth, 2022-12-08:
I think this is working :) It's still unknown at HA startup, but after the first `21/bits` message now using `"ON"`, it's working.

celsworth, 2022-12-08:
I've merged this, let me know if anything needs tweaking.

closed 2022-12-08 by celsworth

lupine, 2022-12-08:
Just https://github.com/celsworth/lxp-bridge/pull/131 from my side. Lovely to see it working. Thanks!

PR #127 (2022-12-08): Add homeassistant switch discovery messages
To get this working, its been necessary to change the new `bits` messages to have `"ON"` and `"OFF"` as values, rather than JSON booleans.

Closes #116