celsworth/lxp-bridge #134: Get & set times (particularly AC charge start / stop) directly, including homeassistant

lupine, 2022-12-09:
The approach in https://github.com/celsworth/lxp-bridge/pull/132 works badly for the AC charge start/stop time fields because the hour and minute is encoded together into a single field.

How about having new `set` endpoints that allow you to pass strings like `0800` or `17:45` ? It *might* be possible to express the mapping from time-of-day to bits via `value_template`, but I'm... not keen :D. So we'd also need matching read endpoints.

In terms of homeassistant, I see two options:

* [MQTT Select](https://www.home-assistant.io/integrations/select.mqtt/)
* [MQTT Text](https://www.home-assistant.io/integrations/text.mqtt/)

For the former, we'd provide a list of possibilities (maybe half-hourly or hourly resolution?) and the user would get a dropdown to choose between them. The set endpoint could support one-minute resolution, but we wouldn't be able to *expose* that in HA very nicely.

For the latter, they get a free-form text entry field, although we can pass a regexp to validate that against. This would allow them to type in `13:58` if that's what they really wanted.

Weird that there isn't a clock integration, though.

celsworth, 2022-12-09:
I sort of had a go at making MQTT messages that would ease this in https://github.com/celsworth/lxp-bridge/pull/123, but that assumed we'd want to set start/end at the same time.

If thats no use then I'm happy to replace it with something similar that only handles one time per message. I'd probably want to remove the previous effort then though, having both would be confusing and unnecessary I think.

So a lot of the grunt work to convert `17:45` into an int for a register is done.  Its just a mapping of an MQTT message to the command (which could be refactored slightly to do just one register per call no problem).

And what to call all these new MQTT messages :)

I'd like to stay away from the `set/hold` namespace, keeping that for more raw register writes.

Maybe `set/time/ac_charge/1/start` and so on for writing, and `read/time/ac_charge/1/start` for reading, where we can specify 1-3 and start or end.

lupine, 2022-12-09:
Oh, I didn't notice those go in :sweat_smile:

The combinatorial explosion means setting both start and end times in a single field wouldn't work well with an MQTT Select, but it might be fine with the text box. I'll have a play after #132 is in and see what I can come up with :thumbsup:

celsworth, 2022-12-10:
Well don't worry too much about making it work with the previous PR - if its not suitable then I'm completely open to changing it to operate with one register at a time, and just doing the HH:MM to int conversion internally.

If it turns out to be more than a few minutes work to mess about with the JSON format as it is, let me know and we can simplify it.

closed 2023-03-27 by celsworth