celsworth/lxp-bridge #139: IoT noob usage questions

SamFarrington, 2023-01-19:
Firstly this project and the energy stats website are really cool, can tell a lot of work has gone into it 👍 

# Problem Statement

_I am a Lux Power Battery owner and I want to dynamically charge the batteries in a cost effective way, utilising Agile Octopus pricing_

# Background

I stumbled across https://github.com/celsworth/octolux which seemed to meet the need, however that project is archived and this one seems to supercede  it.

Is there any documentation on how to create a system using `lxp-bridge` to address the above problem statement?

If not, would it be useful to others if I write some as I figure out how to do it?

I've only got experience of consumer IoT setups, Ring and Hive etc so think there's a gap in my understanding in regards to where this bridge sits and what else is required to perform the task.

I am a software engineer but don't know Rust or MQTT. 

# Errors

In terms of the bridge, I'm having issues with MQTT config, Influx Config and the database config.

## MQTT

MQTT repeats the following error:

```
[2023-01-19T13:21:46.892 ERROR lxp_bridge::mqtt] I/O: Cannot assign requested address (os error 99)
[2023-01-19T13:21:46.892 INFO lxp_bridge::mqtt] reconnecting in 5s
```

## Influx

Influx also seems to be struggling

```
[2023-01-19T13:28:23.853 DEBUG reqwest::connect] starting new connection: http://localhost:8086/
[2023-01-19T13:28:23.856 ERROR lxp_bridge::influx] push failed: ReqwestError(reqwest::Error { kind: Request, url: Url { scheme: "http", cannot_be_a_base: false, username: "", password: None, host: Some(Domain("localhost")), port: Some(8086), path: "/write", query: Some("db=lxp"), fragment: None }, source: hyper::Error(Connect, ConnectError("tcp connect error", Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" })) }) - retrying in 10s
```

## Database

I've tried creating the database using both the Postgres option and the sqllite option to no avail.

I get a log saying `Initializing database` but then nothing happens, e.g. no table is created.

I've checked the connect string using a little Node app and it works successfully, including creating a table.

# Current Architecture

Here is my current architecture.

Got a feeling I'm missing some things.

```mermaid

graph LR;
    LuxPowerInverterHardware --> lxp-bridge;
    lxp-bridge --> postgres-db;

```

celsworth, 2023-01-19:
Hi :)

Lets address your connection problems first. Just to confirm your setup, are you running this in Docker? And presumably the services you want to connect to (influx, mqtt, database) are running on the host natively?

If so, I'm guessing you have `localhost` in the lxp-bridge config, but this will be trying to connect locally inside lxp-bridge's container, and obviously mqtt/influx/database aren't running in there. Instead you need to tell it to connect to the host machine. This can vary depending on your exact setup (Linux, Mac, etc), have a check of the answer in https://stackoverflow.com/a/24326540/1054419 - the short version is probably that replacing `localhost` with `host.docker.internal` might work, but it might need extra Docker configuration as well.

I think all 3 of your problems are the same, once you fix that hopefully it will all magically start to work (assuming you've created the database; lxp-bridge will then create the table itself).

You're right this does supercede octolux and indeed that was originally created to address your exact problem, scheduling charging times based on external factors like Agile prices. The problem was I felt it was trying to shoehorn too much into one project. So instead this project really just wants to be a bridge (hence the name) between your inverter and MQTT/Influx etc.

So the missing bit you need is something to talk to MQTT and send/receive messages based on external criteria, that lxp-bridge can respond to. Generally, that missing bit is node-red and/or HomeAssistant. You probably want node-red, then feed Agile prices to that and make decisions about when to charge, then send an MQTT message that will instruct lxp-bridge accordingly. I'm not a big user of it myself so can't advise too much on the specifics of that I'm afraid. But hopefully thats enough of a starting point?

SamFarrington, 2023-01-19:
Thanks so much for the detailed reply!

So firstly setting `localhost` to `host.docker.internal` has sorted out the Postgres connection 🥳 

In my ignorance I didn't know that I also needed some other services for MQTT and Influx.

My updated req'd architecture as I understand things:

```mermaid
graph LR;
    LuxPowerInverterHardware --> lxp-bridge;
    lxp-bridge --> postgres-db;
    lxp-bridge --> flux-db
    lxp-bridge --> mqtt
    mqtt --> my-node-red-app
    my-node-red-app --> mqtt
    mqtt --> lxp-bridge
    lxp-bridge --> LuxPowerInverterHardware
```

Assuming `my-node-red-app` handles listening to octopus agile pricing and firing off requests via MQTT to start and stop the charging, does this updated diagram look about right?

Will update again when I get chance to spend more time on this.

Will also write up a PR with documentation as a sort of example usage guide

chriscn, 2023-01-19:
Something of note, if you are running your database in docker, you can just use it's container name if you want to access it. (You may need to ensure they're on the same network however).

Your docker-compose.yml
```yaml
lxp-bridge:
  container_name: lxp_bridge
  ...

postgres:
  container_name: postgres

networks:
  default:
    name: docker
```

You can test it by `docker exec -it lxp-bridge /bin/bash` and try `ping postgres` to see if you get a response.

SamFarrington, 2023-01-19:
Got it - so I'm thinking of making a `docker-compose.yml` with `postgres`, `influxdb`, `eclipse-mosquitto` and `lxp-bridge`.

Get them all talking to each other without issue and then moving onto using `node-red` to interact with the inverter rather than purely observing and recording it.

Does that sound like a sensible approach?

celsworth, 2023-01-19:
Yep that all sounds good.

One thing to note is that you'll need to setup a persistent volume if you want your database to be maintained between Docker restarts, or you'll be starting fresh every time. And the same for InfluxDB. MQTT is less important as thats mostly transitory state.

closed 2023-02-17 by celsworth