celsworth/lxp-bridge #140: Intervel time

teicors, 2023-01-23:
Hi Chris,
I know that is not an issue of you sw:, but I have an issue with the interval time : two of three inverters send  the data every 5 minutes, but one  sends the data every 20 seconds. My question is easy: is there any commands to fix the issue (i.e. send the data every 5 minutes) ?
I§t would be interesting to execute a sequence of commands at the start of the sw (ie. changing the interval of time, set a different threshold for  the battery and so on).

celsworth, 2023-01-24:
Hi,

Sorry for delay in replying here :)

So yes, there is a parameter that sets the interval time for an inverter, but I am surprised you have one set at 20 seconds! I thought they were all 5 minutes now.

You can check it with the following:

```
mosquitto_pub -t lxp/cmd/all/read/param/0 -m ''
```

If you have 3 inverters this should get you 3 replies; one of them 20 and the other two 300?

You can then set the odd one out to 300 like this:

```
mosquitto_pub -t lxp/cmd/XXXXXXXXXX/set/param/0 -m 300
```

Where XXXXXXX is the datalog of the inverter to set.

Hopefully that gets them all at 5 minutes?

teicors, 2023-01-25:
[2023-01-25T17:20:40.014 lxp/BA10430542/inputs/1
[2023-01-25T17:21:00.505 lxp/BA10430542/inputs/1
[2023-01-25T17:21:21.019 lxp/BA10430542/inputs/1
[2023-01-25T17:21:39.513 lxp/BA10430542/inputs/1
[2023-01-25T17:22:00.023 lxp/BA10430542/inputs/1
[2023-01-25T17:22:20.514 lxp/BA10430542/inputs/1

no luck at all :(

what may I expect from the command ?
Can I check if the inverter can receive my command?

celsworth, 2023-01-27:
Hi :)

Ok lets take it one step at a time; did you run the first commands, and what responses did you get?

In one terminal, just to watch whats going on, run:

```
mosquitto_sub -t lxp/# -v
```

Now in another terminal:

```
mosquitto_pub -t lxp/cmd/all/read/param/0 -m ''
```

What does the first terminal give you? I'm expecting to see two responses with 300 and one with 20?

celsworth, 2023-01-28:
Ok, thats bad news :(

If you're sure the serial numbers in configuration are right then that should get a response. If not we can't change the interval I guess. Might be worth asking Luxpower for a firmware upgrade? And double check all the serial numbers in your config match each inverter, if not then the inverter won't respond to our requests.

I'll close this issue as there's no bug or anything to fix here but we can continue the discussion.. I should probably open the discussion feature on Github.

closed 2023-01-28 by celsworth

teicors, 2023-03-07:
Hi Chris,
few days ago the electrician did an update of fw and the  command worked:
[2023-03-07T09:00:51.430 DEBUG lxp_bridge::coordinator] parsed command WriteParam(Inverter { enabled: true, host: "10.10.20.242", port: 8000, serial: xxx, datalog: xx, heartbeats: None }, 0, 300)
[2023-03-07T09:00:51.456 DEBUG lxp_bridge::lxp::inverter] inverter xxx: TX [161, 26, 2, 0, 18, 0, 1, 196, 66, 65, 49, 48, 52, 51, 48, 53, 52, 50, 0, 0, 2, 0, 44, 1]
[2023-03-07T09:00:53.066 DEBUG lxp_bridge::lxp::packet_decoder] 21 bytes in: [161, 26, 2, 0, 15, 0, 1, 196, 66, 65, 49, 48, 52, 51, 48, 53, 52, 50, 0, 0, 0]
[2023-03-07T09:00:53.066 DEBUG lxp_bridge::coordinator] RX: WriteParam(WriteParam { datalog: xxx, register: 0, values: [0, 0] })
[2023-03-07T09:00:53.069 INFO lxp_bridge::mqtt] publishing: lxp/result/xxx/set/param/0 = OK
[2023-03-07T09:00:53.069 DEBUG rumqttc::state] Publish. Topic = lxp/result/xxx/set/param/0, Pkid = 100, Payload Size = 2
[2023-03-07T09:00:55.066 DEBUG lxp_bridge::lxp::packet_decoder] 19 bytes in: [161, 26, 2, 0, 13, 0, 1, 193, 66, 65, 49, 48, 52, 51, 48, 53, 52, 50, 0]
[2023-03-07T09:00:55.068 DEBUG lxp_bridge::coordinator] RX: Heartbeat(Heartbeat { datalog: xxx })

[2023-03-07T08:58:56.321
[2023-03-07T08:59:17.622
[2023-03-07T08:59:36.933
[2023-03-07T08:59:56.225
[2023-03-07T09:00:17.522
[2023-03-07T09:00:36.823
[2023-03-07T09:00:36.823
[2023-03-07T09:05:57.208
[2023-03-07T09:10:56.323
[2023-03-07T09:15:55.427

Thanks,
Angelo

celsworth, 2023-03-07:
👍 good to know, glad its sorted :)