celsworth/lxp-bridge #141: Set logging levels

Beenay25, 2023-01-28:
I could be completely missing something obvious, but docker, portainer and lxp-bridge are all pretty new to me.

Is there a way to set the logging level for lxp-bridge?

At the moment all the logs I see are full of DEBUG messages. I want to quieten them down a little.

celsworth, 2023-01-28:
Hi,

Yes there's an environment variable `RUST_LOG` that will be honoured. Set this when starting lxp-bridge.

It's set in the example systemd unit file: https://github.com/celsworth/lxp-bridge/blob/master/systemd/lxp-bridge.service#L9

```
Environment="RUST_LOG=info"
```

I guess a configuration file option for this wouldn't go amiss. And maybe it can be mentioned in the docs, will take a look at adding that.

Beenay25, 2023-01-28:
Perfect. Thank you so much for the fast response.

I just set an environmental variable up in Portainer and it worked first time.

closed 2023-01-28 by Beenay25