celsworth/lxp-bridge #148: Home assistant integration not appearing as a device

Leepee, 2023-03-01:
Hi, 

First up, the usual thanks. People like you make the nerd world go around <3

I have this MQTTing to my HA, and it's not appearing as a grouped "device" but rather a ton of sensor inputs. I did read about the changes made in 0.1 that were to resolve this - is there somewhere I need to enable that?

The initial discovery messages in the log look like this: 

`[2023-03-01T23:22:30.360 DEBUG rumqttc::state] Publish. Topic = homeassistant/sensor/MYSERIAL_soc/config, Pkid = 3, Payload Size = 223`

And my config.yaml is like this: 

`inverters:
- enabled: true
  host: 10.20.0.4
  port: 8000
  serial: MYSERIAL
  datalog: MYDATALOG
  heartbeats: false
- enabled: false
  host: 192.168.0.163
  port: 8000
  serial: 9999999999
  datalog: 3333333333
  heartbeats: false

databases:
- enabled: false
  url: postgres://lxp:lxp@localhost/lxp
  # url: mysql://lxp:lxp@localhost/lxp
  # url: sqlite://lxp.db

mqtt:
  enabled: true
  host: 10.0.0.5
  port: 1883
  username: MYUSERNAME
  password: MYPASSWORD
  namespace: lxp
  homeassistant:
    enabled: true
    prefix: homeassistant
`
Thanks for any help :)

celsworth, 2023-03-03:
Hi,

Yeah v0.10 makes a lot of HA improvements but it's not released yet. Although that shouldn't matter too much, if you're running this via Docker then you can just checkout the `dev` tag rather than `latest` and that will get you a compiled master branch, which is what v0.10 will end up being.

I'm a bit busy with other stuff at the mo so not giving this repo the love it needs :( I'll try and get the outstanding PRs merged and a v0.10 release made soon.

Leepee, 2023-03-04:
Hey, I just logged on to mention that I realised that at another time. Thanks for the work and no worries that it's not getting priority - I'll install the dev tag :)

closed 2023-03-04 by Leepee