celsworth/lxp-bridge #163: Client lxp-bridge already connected, closing old connection.

kebaotu, 2023-05-15:
I use Mosquitto mqtt on my home assistant. When I change the wifi router, I get the following error on the log of Mosquitto mqtt and the sensor parameters on the lxp-bridge are all unknown:
 New client connected from 192.168.1.167:40896 as lxp-bridge (p2, c1, k60, u'mqtt').
New connection from 192.168.1.167:54382 on port 1883.
Client lxp-bridge already connected, closing old connection.
I tried to remove lxp-bridge using mqtt explorer software but still not getting the above error. I don't know setting unique id or clearing all data about lxp-bridge on mosquitto mqqt on home assistant.
I tried cmd: mosquitto_pub -h localhost -p 1883 -t 'home/lxp-bridge' -m 'disconnect' but it didn't work either

apbarratt, 2023-05-15:
I wonder if it's in fact confused on the module itself, perhaps try killing all power to the inverter and rebooting that before starting the bridge again :)

kebaotu, 2023-05-15:
Restart doesn't work.

kebaotu, 2023-05-15:
I think the problem is because before changing wifi router, lxp-bridge was on mosquitto broker and when changing router mosquitto doesn't accept connection due to same name. I don't know how to add unique id on config file

closed 2023-05-15 by kebaotu