celsworth/lxp-bridge #164: MQTT commands not being received

apbarratt, 2023-05-15:
Having spent a lot of time discussing why I can't get more frequent data in https://github.com/celsworth/lxp-bridge/issues/113, I've come to realise that my instance simply doesn't respond to any kind of command whatsoever.

I am seeing the regular output of data on the `lxp/BA12345678` topic including `ac_charge`, `charge_priority`, `forced_discharge`, `hold`, and `inputs` with a regular update every five minutes to `inputs/1`, `inputs/2`, `inputs/3`, and `inputs/all` (making the previous 3 redundant I'd have thought), but I cannot get any other kind of message out of the system manually.

For example, if I run `lxp/cmd/BA12345678/read/input/1`, I can see this message appearing if I'm listening to `lxp/#`, but I don't see any response in the corresponding `lxp/BA12345678/inputs/1` topic.

The same applies to other commands.  If I run `lxp/cmd/BA12345678/read/hold/1`, then nothing at all appears under `lxp/result/#`.  The same applies if I try `all` instead of my data logger ID.

Basically, running any `lxp/cmd/...` command is no different for me than if I'd run `lxp/nonsenseCommand`.  It goes out, but nothing appears to be listening or handling the message.

Logging into the module to see version data says the following:
```
Version Information
  Module Name
    EMW1290
  MAC
    84:9D:C2:[hidden for privacy]
  SDK Version
    MX1290002.032.002
  Software Version
    ATV1.20@MX1290
  Web Version
    web_V1.0.0
```

celsworth, 2023-05-15:
The first thing to check, and it's absolutely critical, is that the datalog/serial numbers in the configuration match your actual serial numbers (and the ones in the MQTT message you're sending). If they don't, then the inverter will ignore the packets we create and send to it.

This is by far the most common reason I've found for the inverter not responding to messages thus far.

apbarratt, 2023-05-15:
I shall check the serial number now :)
I did see another closed ticket suggesting this but they'd reported seeing "FAIL" messages in their container console and I'm not seeing those so I'd assumed it wasn't the same thing.  Checking now...

apbarratt, 2023-05-15:
Okay... this is odd. It's working. I didn't change anything, but I restarted the container and suddenly it's all working. So clearly something had just got stuck.

I did have my electrics turned off the other day and I've restarted the inverter a couple of times. I wonder if this is an issue caused by power loss at some point causing the command listener to lock up? :/

When I've a moment to test, I'll try killing the inverter and see what happens to this, cooking right now so not a good time to do that though.

apbarratt, 2023-05-17:
After fiddling with my consumer box and network, I'm afraid I simply have not been able to reproduce this bug.  Restarting the docker container is clearly what fixed it but I cannot seem to reproduce the behaviour at all.  I'm going to go ahead and close it for now, though I'll reopen if it should happen again and I can figure out some reproduction steps.

closed 2023-05-17 by apbarratt