celsworth/lxp-bridge #165: Crash when trying to use InfluxDB

reinhardholl, 2023-05-16:
Hi, trying to use influxDB to send some data to a Grafana instance. When enabling Influx the bridge crashes with the following:

```
thread 'main' panicked at 'cannot represent -16384 as u64 for bat_status_2', /usr/src/lxp-bridge/src/influx.rs:88:33
stack backtrace:
   0: rust_begin_unwind
             at ./rustc/9eb3afe9ebe9c7d2b84b71002d44f4a0edac95e0/library/std/src/panicking.rs:575:5
   1: core::panicking::panic_fmt
             at ./rustc/9eb3afe9ebe9c7d2b84b71002d44f4a0edac95e0/library/core/src/panicking.rs:64:14
   2: <futures_util::future::maybe_done::MaybeDone<Fut> as core::future::future::Future>::poll
   3: <futures_util::future::poll_fn::PollFn<F> as core::future::future::Future>::poll
   4: <futures_util::future::maybe_done::MaybeDone<Fut> as core::future::future::Future>::poll
   5: <futures_util::future::poll_fn::PollFn<F> as core::future::future::Future>::poll
   6: <tokio::future::poll_fn::PollFn<F> as core::future::future::Future>::poll
   7: tokio::runtime::park::CachedParkThread::block_on
   8: tokio::runtime::scheduler::multi_thread::MultiThread::block_on
   9: lxp_bridge::main
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

My influx settings are: 

```
influx:
  enabled: true
  url: http://192.168.0.101:8086
  username: xxxx
  password: xxxxxxxx
  database: lxp
```

reinhardholl, 2023-05-17:
Awesome, will wait for the latest docker to arrive 👌 Will give it a test

lupine, 2023-08-18:
Looks like we can close this issue @celsworth  ?

celsworth, 2023-08-18:
👍