celsworth/lxp-bridge #166: Ability to set log-level (info, warn, error, debug, etc) in configuration

zatie12, 2023-05-19:
The ability to control the logging level would be great as lxp-bridge is currently writing a significant amount of info and debug data to the journal/syslog. Apologies if this has already been discussed or I missed this in the Wiki.

Thanks as always!

celsworth, 2023-05-22:
There is an environment variable you can set, `RUST_LOG`, which will change this. It's used in the example systemd unit file: https://github.com/celsworth/lxp-bridge/blob/master/systemd/lxp-bridge.service#L9

I'd be fine with adding a commandline argument to set this as well though if that would be easier. Also open to suggestions for log messages being moved between different loglevels.

zatie12, 2023-05-23:
That's great, thanks Chris. I'll set the environment variable in the systemd unit file.

johnl, 2023-05-29:
This is probably a touch more important now that there is a home assistant addon, and needs to be settable within the config really for that. It's common to run home assistant on a raspberry pi with an sd card, so it's good to be able to write as little data as possible so as not to wear out the sdcard. And need 

I'm wondering if the "INFO lxp_bridge::mqtt] publishing: lxp/..." messages should be at debug level too, to allow reducing it even further without losing more important info messages?

celsworth, 2023-05-29:
This seems to be cropping up a bit more now, and I've no objections so I've added a `loglevel` to the configuration in #168 which will be in v0.11

The second point though, moving the loglevel to debug, I'm not too sure about as it was recently moved from debug to info: https://github.com/celsworth/lxp-bridge/commit/6402f900e15ee5b0c15e48055a0de5fd46bb1ac5

So I'm wary of just moving things back and forth. I think this probably does belong at info. Maybe another way of looking at it is anything remotely important should perhaps be at warn instead, so are there any candidates you'd like to see moved from info to warn perhaps? Which messages are you thinking of that you wouldn't want to lose?

closed 2023-10-28 by celsworth