celsworth/lxp-bridge #173: Remote connection

Boegebjerg, 2023-06-19:
Is it possible to host lxp-bridge on a remote server and connect to the datalogger?

celsworth, 2023-06-19:
Mostly the limitation here is connectivity. Generally speaking you don't want your datalogger exposed to the Internet, so unless you setup a VPN, lxp-bridge running on a remote server cannot see it.

Luxpower get around this by having the datalogger actually connect outwards to a listening port. We could also do this - lxp-bridge could listen on a port and then we configure the datalogger to connect to that.

Since there is no authentication in the communications with the datalogger it would still be your responsibility to secure lxp-bridge, ie by firewalling it. But it would offer a bit more flexibility for cases like these.

Would that be useful?

garthy, 2023-09-29:
I suspect a better case for remote management is to use the luxpower HTTP API to mange the parameters.

celsworth, 2023-10-28:
Bit of a thorny issue at the moment, that HTTP API, I think :) I'm staying well clear ;)

As the OP hasn't returned to make any further comments I'll just close this for now.

closed 2023-10-28 by celsworth