celsworth/lxp-bridge #176: EG4 18kPV

dbainbridge, 2023-07-07:
Any ideas if this will work with the EG4 18kPV which is based on a LUX-12K hybrid inverter?

celsworth, 2023-07-08:
Good question, in all honest I have absolutely no idea :) The EG4 is an american-based product right? So I don't think I'll be able to get my hands on one to try it out. It may be possible to add support but we'll need to know the protocol it speaks first.

If I had to guess, I'd suspect that if its based on a Luxpower inverter then the internal registers will all be pretty much the same, but perhaps the wifi dongle may be slightly different so there may be a different TCP wrapper around the packets. 

Do you have one, are you able to try it? Does the EG4 talk to Luxpower's servers? What does the web interface of the wifi dongle look like, can you open a listening port for lxp-bridge to connect to?

Most usefully, some packet dumps of what's going on when it sends data out would be a good start, maybe we can deciphering it.

dbainbridge, 2023-07-08:
Thanks.  I have ordered one so will be able to investigate further.  It is an "Americanized" version of the LUX-12K hybrid inverter so maybe it will work.  It does not however talk to Luxpower servers.  Instead it is supposed to talk to AWS servers  in USA region.   On page 41 of this PDF it shows interface setting up the Wi-fi dongle:
https://eg4electronics.com/wp-content/uploads/2023/06/EG4-18KPV-12LV-Manual-1.3.0.pdf

celsworth, 2023-07-08:
Well well, look what I found ;) https://eg4electronics.com/wp-content/uploads/2023/06/EG4-18KPV-12LV-Modbus-Protocol.pdf

The registers correspond exactly with what we have documented in https://github.com/celsworth/lxp-bridge/wiki/Inputs and https://github.com/celsworth/lxp-bridge/wiki/Holdings so I think there's a good chance it will work.

This is supremely helpful, until now Luxpower have been extremely cagey about releasing documentation like this, a lot of what we know is reversed engineered or guessed. In fact there's more registers in that doc than I knew about so we could add those to lxp-bridge's output at some point. I see there's older/smaller models in the EG4 range that also have Modbus Protocol documentation which look completely different (https://eg4electronics.com/downloads/), so probably only the 18kPV will work with this software.

The documentation is focused on UART though rather than over TCP so there's still a bit of an unknown there over TCP encapsulation. But I think that should be easy to work out, just by watching what the inverter sends for a while.

The Wi-fi dongle images in the manual look pretty much identical to what's on Lux inverters as well, so the instructions will be much the same, just open a listening port to connect to. The AWS servers you mention look like simple reskins (literally just a different colour scheme) by the look of it.

dbainbridge, 2023-07-08:
Excellent.  It will be a while before I can test this since haven't received it yet.  Maybe someone else who has one may be able to test beforehand.

Sboshoff76, 2023-07-17:
> Well well, look what I found ;) https://eg4electronics.com/wp-content/uploads/2023/06/EG4-18KPV-12LV-Modbus-Protocol.pdf

Oooh... this makes things exciting :)

joshs85, 2023-08-18:
> Any ideas if this will work with the EG4 18kPV which is based on a LUX-12K hybrid inverter?

It works.  I'm using it with mine.  It locks up occasionally and stops updating the values.  Not sure why yet.. but for the most part it works.  I'd love it if the rest of the registers (the ones the distributor has access to that customers don't) would be published tho.

celsworth, 2023-08-18:
> I'd love it if the rest of the registers (the ones the distributor has access to that customers don't) would be published tho.

Any ones in particular? Since we have the modbus protocol document now we can add a lot more, but since I don't have access to an inverter that populates all these new registers, I'm not really sure which ones are useful (and even with the Chinglish doc, understanding what half of them mean is tricky sometimes)

joshs85, 2023-08-18:
I'd really like to see the fault codes and status.. i mean really all of them that make sense.  I'd be happy to provide some data if needed.  I have my system ac coupled.  I don't have anything in the PV ports.  just a huge AC Coupled array.