celsworth/lxp-bridge #178: thread 'main' panicked

NicLogix, 2023-07-16:
I have been following the amazing work you are doing with lxp-bridge for quite some time, and finally yesterday got my system up and running.
LXP-5000WPV with SHOTO SDC10 BOX 5

I'm running the Home Assistant addon of lxp-bridge.

I noticed on mqtt that the lxp/cmd/BAxxxxxxxx/read/inputs/1 2 3 are being requested, but nothing gets published on lxp/BAxxxxxxxx/inputs/all, and therefore no values updated in home assistant.
Looking at the logs:

[2023-07-16T13:58:28.826 DEBUG lxp_bridge::lxp::packet_decoder] 19 bytes in: [161, 26, 5, 0, 13, 0, 1, 193, 66, 65, 50, 52, 55, 48, 50, 51, 52, 57, 0]
[2023-07-16T13:58:28.826 DEBUG lxp_bridge::coordinator] RX: Heartbeat(Heartbeat { datalog: BAxxxxxxxx })
[2023-07-16T13:58:30.046 DEBUG lxp_bridge::lxp::packet_decoder] 291 bytes in: [161, 26, 5, 0, 29, 1, 1, 194, 66, 65, 50, 52, 55, 48, 50, 51, 52, 57, 15, 1, 1, 4, 50, 53, 49, 51, 53, 51, 48, 48, 48, 55, 0, 0, 254, 16, 0, 0, 0, 0, 0, 67, 16, 30, 2, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 14, 0, 220, 8, 0, 0, 1, 8, 121, 19, 0, 0, 46, 0, 110, 0, 0, 0, 226, 8, 16, 20, 74, 16, 122, 19, 0, 0, 0, 0, 0, 0, 21, 2, 0, 0, 0, 0, 0, 0, 0, 0, 14, 0, 7, 0, 8, 0, 5, 0, 0, 0, 56, 0, 66, 16, 187, 14, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 33, 0, 0, 0, 22, 0, 0, 0, 8, 0, 0, 0, 5, 0, 0, 0, 0, 0, 0, 0, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 26, 0, 26, 0, 34, 0, 25, 0, 0, 0, 240, 22, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 10, 0, 0, 0, 28, 37, 62, 2, 0, 0, 0, 0, 0, 0, 0, 192, 0, 0, 0, 0, 112, 50, 191, 171, 14, 3, 240, 96, 157, 16, 2, 0, 1, 0, 50, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 23, 2, 2, 0, 23, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 1, 0, 0, 50, 53, 49, 51, 53, 51, 48, 48, 48, 55, 0, 0, 0, 0, 0, 0, 253, 251, 0, 5, 193, 124, 212, 1, 104, 188]
[2023-07-16T13:58:30.046 DEBUG lxp_bridge::coordinator] RX: TranslatedData(TranslatedData { datalog: BAxxxxxxxx, device_function: ReadInput, inverter: 25xxxxxxxx, register: 0, values: [16, 0, 0, 0, 0, 0, 67, 16, 30, 2, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 14, 0, 220, 8, 0, 0, 1, 8, 121, 19, 0, 0, 46, 0, 110, 0, 0, 0, 226, 8, 16, 20, 74, 16, 122, 19, 0, 0, 0, 0, 0, 0, 21, 2, 0, 0, 0, 0, 0, 0, 0, 0, 14, 0, 7, 0, 8, 0, 5, 0, 0, 0, 56, 0, 66, 16, 187, 14, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 33, 0, 0, 0, 22, 0, 0, 0, 8, 0, 0, 0, 5, 0, 0, 0, 0, 0, 0, 0, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 26, 0, 26, 0, 34, 0, 25, 0, 0, 0, 240, 22, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 10, 0, 0, 0, 28, 37, 62, 2, 0, 0, 0, 0, 0, 0, 0, 192, 0, 0, 0, 0, 112, 50, 191, 171, 14, 3, 240, 96, 157, 16, 2, 0, 1, 0, 50, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 23, 2, 2, 0, 23, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 1, 0, 0, 50, 53, 49, 51, 53, 51, 48, 48, 48, 55, 0, 0, 0, 0, 0, 0, 253, 251, 0, 5, 193, 124, 212, 1] })
[2023-07-16T13:58:30.046 INFO lxp_bridge::mqtt] publishing: lxp/BAxxxxxxxx/inputs/all = {"status":16,"v_pv_1":0.0,"v_pv_2":0.0,"v_pv_3":416.3,"v_bat":54.2,"soc":100,"soh":0,"p_pv":0,"p_pv_1":0,"p_pv_2":0,"p_pv_3":0,"p_charge":0,"p_discharge":14,"v_ac_r":226.8,"v_ac_s":0.0,"v_ac_t":204.9,"f_ac":49.85,"p_inv":0,"p_rec":46,"pf":0.0,"v_eps_r":227.4,"v_eps_s":513.6,"v_eps_t":417.0,"f_eps":49.86,"p_eps":0,"s_eps":0,"p_to_grid":0,"p_to_user":533,"e_pv_day":0.0,"e_pv_day_1":0.0,"e_pv_day_2":0.0,"e_pv_day_3":0.0,"e_inv_day":0.0,"e_rec_day":1.4,"e_chg_day":0.7,"e_dischg_day":0.8,"e_eps_day":0.5,"e_to_grid_day":0.0,"e_to_user_day":5.6,"v_bus_1":416.2,"v_bus_2":377.1,"e_pv_all":0.0,"e_pv_all_1":0.0,"e_pv_all_2":0.0,"e_pv_all_3":0.0,"e_inv_all":0.0,"e_rec_all":3.3,"e_chg_all":2.2,"e_dischg_all":0.8,"e_eps_all":0.5,"e_to_grid_all":0.0,"e_to_user_all":10.0,"t_inner":26,"t_rad_1":26,"t_rad_2":34,"t_bat":25,"runtime":71408,"max_chg_curr":0.0,"max_dischg_curr":95.0,"charge_volt_ref":57.4,"dischg_cut_volt":0.0,"bat_status_0":0,"bat_status_1":0,"bat_status_2":-16384,"bat_status_3":0,"bat_status_4":0,"bat_status_5":12912,"bat_status_6":-21569,"bat_status_7":782,"bat_status_8":24816,"bat_status_9":4253,"bat_status_inv":2,"bat_count":1,"bat_capacity":50,"bat_current":0.0,"bms_event_1":0,"bms_event_2":0,"max_cell_voltage":0.0,"min_cell_voltage":0.0,"max_cell_temp":0.0,"min_cell_temp":0.0,"bms_fw_update_state":535,"cycle_count":2,"vbat_inv":53.5,"time":1689508710,"datalog":"BAxxxxxxxx"}

thread 'main' panicked at 'cannot represent -16384 as u64 for bat_status_2', /usr/src/lxp-bridge/src/influx.rs:88:33

note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

The addon succesfully stops and restarts, but obviously this is not ideal (as everything in home assistant goes to 'Unavailable'.

celsworth, 2023-07-16:
Hi :)

This is actually already fixed in #161, but it hasn't made it into a release yet. There's no hold up on making a new one so I'll get onto that now. Try v0.11 when its built, should fix it for you :)

NicLogix, 2023-07-16:
Great, will have a lookout for v0.11

NicLogix, 2023-07-16:
Confirmed v0.11 working, and add-on is kept running.
lxp/BAxxxxxxxx/inputs/all is getting published, and entities updated in Home Assistant.

The add-on does not pick up there is an update available.  I manually uninstalled, and reinstalled to get v0.11.

Keep up the good work.

closed 2023-07-16 by NicLogix

celsworth, 2023-07-17:
> Confirmed v0.11 working, and add-on is kept running. lxp/BAxxxxxxxx/inputs/all is getting published, and entities updated in Home Assistant.

👍 thanks for the confirmation!

> The add-on does not pick up there is an update available. I manually uninstalled, and reinstalled to get v0.11.

Hmm this *might* have happened eventually. There is some CI to update the addon version automatically but it broke so I need to have a closer look and try to fix that. I did end up just manually updating the addon yaml about 6 hours after the release and I'd expect that's what should prompt HA to auto-update. But there is probably some time delay between it checking for updates, possibly a day or two.

Be interested to see if anyone else manages to get the auto update (I'm not a HA user personally, in fact my Luxpower's not even wired up at the mo).