celsworth/lxp-bridge #184: Occasionally "PV Generation (All time)" goes out of whack

Sboshoff76, 2023-08-19:
[image]

celsworth, 2023-08-20:
Ah, floating point maths :) I assume this is the only key that does this? It will be because of this:

https://github.com/celsworth/lxp-bridge/blob/df38f1c51122fba212c2e9999277c94d7fb2adb7/src/lxp/packet.rs#L793

All of those are `f64`. `e_pv_day` will probably do the same thing occasionally.

I think this is mostly cosmetic to be fair, the rounding errors are pretty insignificant. I could work around it by doing maths in integer then converting back, will have a think.

closed 2023-08-20 by celsworth

Sboshoff76, 2023-08-20:
Yeah wasnt a biggie, just thought I'd mention. Thanks for the fix!

PR #185 (2023-08-20): Avoid floating point maths oddities
Fixes #184 

Actually managed to get the broken behaviour into a test so remove that too, oops.