celsworth/lxp-bridge #188: Would it be possible to pull Status and Error codes

Sboshoff76, 2023-08-21:
Attached doc might not be universal but probably a good start.

[SNA Eco-Hybrid Status and Error Codes2.pdf](https://github.com/celsworth/lxp-bridge/files/12398157/SNA.Eco-Hybrid.Status.and.Error.Codes2.pdf)

celsworth, 2023-08-24:
#189 has added `internal_fault`, `warning_code`, and `fault_code` keys to the JSON output (`status` was already there)

I'll get the wiki updated with these later.

Sboshoff76, 2023-08-24:
@celsworth thanks for including these. Because of your frequent patches I learnt how to use docker just so that I can get the latest version. :)

I've recreated my container from the latest dev image and even deleted my device from mqtt so it can import it afresh but it's still not showing these new entities. Is there something else I need to do to see them?

celsworth, 2023-08-24:
Ah, I missed the bit that exposes these to HomeAssistant. I'll need to dig through their documentation to work out what sort of entity to expose it as though, since its just a number that doesn't mean much on its own..

celsworth, 2023-08-24:
Merged a PR in so the dev build should get updated within the next half hour or so. Not had chance to test it (I'll need to setup HA tomorrow to give it a try), and I'm a bit unsure if the "None" device_class will work for this, so let me know if anything breaks..

There might be a nicer way to expose it to HA (some string constants describing exactly what the status/fault/warning is for instance) but for now the integer is hopefully enough to start with..

Sboshoff76, 2023-08-25:
Thank Chris. I deployed the image from last night but still a no go I'm afraid. I can't find them and the entity count remains on 72.

celsworth, 2023-08-25:
Took the time to test it with HA this morning and I can see its not working; I'll start a new PR shortly to get this going properly.

celsworth, 2023-08-25:
https://github.com/celsworth/lxp-bridge/pull/191 is underway :)

Just deciding what to do about fault/warning. I think technically speaking the inverter can present multiple faults at once, but HA can only show one, so I'll probably just show the first one I find thats set. I think its probably quite rare there will be >1 anyway so it won't matter much.

celsworth, 2023-08-29:
#191 is merged, could you give this another try (give it an hour from this comment for Docker `dev` build to update)

Sboshoff76, 2023-08-29:
Thanks! 3 additional entities have come in but all showing unknown.

[image]

Sboshoff76, 2023-08-29:
Brilliant... Added to the config and it's reporting in as it should.
[image]

Thanks Chris!

celsworth, 2023-08-29:
Hurrah, got there eventually :)

I'll close this as completed then. I'll update some docs to make it clear that this config option needs turning on, and probably default it to on sooner rather than later actually.

closed 2023-08-29 by celsworth