celsworth/lxp-bridge #198: Inverter Gaining Time

jlwainwright, 2023-09-18:
Hi Guys

I have a client with a LXP12K and the inverter has recently started gaining time on the clock causing all the set schedules to be of no use.

The distributor wants the inverter to be removed and sent in for diagnostics and repairs but in the meantime I am thinking to make use of a HA automation to update/sync the time say every 10 minutes with that of the HASS host. 

Is the clock registers available and will we be able to write to it?

celsworth, 2023-09-18:
The clock registers are just like any other and are available at holding registers 12-14.

However if you just want to sync the time on the inverter then lxp-bridge has an experimental system already built in that can do that; add something like this to your config if its not there already:

```
scheduler:
  enabled: true
  timesync:
    enabled: true
    cron: "*/10 * * * *"
```

This will set the time on the inverter to the time of the system lxp-bridge is running on, every 10 minutes. It used to have some issues particularly around DST changes, although I did try to fix these (#107, #153) so it might be ok now :)

jlwainwright, 2023-09-18:
Amazing! Thanks Chris

jlwainwright, 2023-10-14:
Hi Chris.

Not sure if it is good practice to open an older ticket but it is very much related.

I have configured the scheduler but I get this message when the Add-On starts up and the time to the inverter is not synchronising:

INFO lxp_bridge::scheduler] scheduler config not found, skipping

[6db19dfb_lxp-bridge_2023-10-14T18-46-20.520Z.log](https://github.com/celsworth/lxp-bridge/files/12907473/6db19dfb_lxp-bridge_2023-10-14T18-46-20.520Z.log)

[config.txt](https://github.com/celsworth/lxp-bridge/files/12907478/config.txt)

celsworth, 2023-10-14:
hmm, that's a little odd, as if I use your config (all I had to do was change the MQTT password not to have `*` in it), I get:

```
[2023-10-14T20:41:01.146 INFO lxp_bridge::scheduler] scheduler starting
[2023-10-14T20:41:01.147 INFO lxp_bridge::scheduler] next timesync at 2023-10-14 20:50:00 +01:00, sleeping for PT538.852738S
```

So that indicates there's nothing wrong with the config, it should be starting the scheduler. Are you absolutely sure that the correct config file is being used?

jlwainwright, 2023-10-14:
I am surely doing something wrong. I am getting the same on another site with another inverter.

I have noticed that the scheduler config does not get displayed in the GUI config (see attached screenshot) but is present when editing in YAML.

In what section can I add the scheduler settings when editing in GUI? 

[image]

celsworth, 2023-10-14:
Oh! I hadn't twigged this was related to the HA schema. I think its because this section is missing in https://github.com/celsworth/lxp-bridge/blob/master/addon/config.yaml

I'll make a commit which should fix this (you may need to reinstall the addon). I'll post back when done.

celsworth, 2023-10-14:
Done in https://github.com/celsworth/lxp-bridge/commit/a7ecc684ddae1c4f4f82bcf3a3424040fc68a1e5, how does that look now? As per above you might need to reinstall the addon to get the latest schema.

jlwainwright, 2023-10-14:
Strange. Not seeing any change in the schema and still getting same warning. Even after Uninstall and fresh install. 
Will try rebooting HA between removal and reinstallation.

jlwainwright, 2023-10-14:
No change on either of my two systems. That is after uninstalling and reinstalling the add-on again.

celsworth, 2023-10-15:
Hm :( Not sure what else to suggest at the moment. I was under the impression that reinstalling the addon would re-read the `config.yaml` that HA uses (which contains an lxp-bridge config). Wonder if its cached somewhere in your systems.

I don't have a HA installation setup at the moment to test but I'll see if I can get one setup later.

celsworth, 2023-10-15:
Falling at the first hurdle here - added the lxp-bridge repo to a new HA install on a Pi4, but lxp-bridge isn't showing as an available addon to install. No errors in the logs or anything.

I'm not a regular HA user so might be doing something wrong. (someone else actually contributed the HA addon configs). I'll keep messing with it.

jlwainwright, 2023-10-15:
I can confirm that I can no longer install the add-on on HA yes.

celsworth, 2023-10-15:
Ok, not really sure what's going on. I've reset the configs back to the known working ones from last week (albeit with no scheduler config in the UI) and I still can't install it on my test HA so I'm a bit stumped at the moment.

celsworth, 2023-10-15:
Okay, think I worked it out. The app is installable again, it just took a few minutes after reverting to the known-good config files to reappear.

The issue thats making it disappear seems to be that I have nested configuration under the scheduler. So adding this bit is fine:

```
scheduler:
  enabled: true
```

but if I change it to this, it disappears from HA:

```
scheduler:
  enabled: true
  timesync:
    enabled: true
```

And I suspect its because HA doesn't support two levels of nesting like that? Not sure what we can do about this apart from rework the config to just use `timesync_enabled` and `timesync_cron` perhaps.

jlwainwright, 2023-10-15:
Great!! Add-on is back and working again. 

Wish I could chime in with advice but that is way above my paygrade. 

Will meantime log into inverter twice a day to set the time manually.  Caveman style! 

Thanks for all your trouble and assistance Chris

NicLogix, 2023-10-15:
@jlwainwright @celsworth 
To run any network with more than two devices communicating either to each other or to the internet, time is of utmost importance.
Most routers worth their salt will have a NTP section to configure.  
Best practice is to sync the DHCP device to the internet NTP pool, and set DHCP lease to 12 hours.  
After 6 hours devices will (at the lease) re-request ip and get time sync, i.e. four times a day.

HA has an accurate and lightweight add-on to manage this called Chrony. As default NTP packets is broadcast on the network every half hour.  Very easy config and setup.
Set the routers to get time from Chrony, and your time will be synced to a 1/1000 of a second.

I run HA with Chrony, and my DHCP router sync time.  Chrony and router sync from internet, and each other.  Never any time issues.  Just my 2c.

Great work as always with this code.

celsworth, 2023-10-28:
0.13 released which has the new config structure

closed 2023-10-28 by celsworth

jlwainwright, 2023-12-10:
> 0.13 released which has the new config structure

Just to check Chris. Scheduling is still not intended to be working in HA Add-On even with 0.13?

Still getting:

"INFO lxp_bridge::scheduler] scheduler config not found, skipping"

This with my config looking like this:
GUI:
[image]

YAML:
[image]