celsworth/lxp-bridge #204: Loglevel is missing from the Home assistant schema and config.

garthy, 2023-09-29:
https://github.com/celsworth/lxp-bridge/blob/master/addon/config.yaml

celsworth, 2023-09-29:
It being missing isn't a problem in itself, it should default to DEBUG, but are you saying it should be present as INFO?

garthy, 2023-09-29:
So I think if is part of the schema it will show up as a field you can edit in the configuration for the addon. (I'm fairly new to Home Assistant and haven't quite worked out all the integration points)

celsworth, 2023-10-01:
Ah I see. I'm happy to add that then, I don't think it'll even need a new release, any new install after `master` is updated will see the new schema.

I think it probably makes sense to move towards using `info` as the default loglevel too these days, so I'll add it at that.

celsworth, 2023-10-02:
Added in https://github.com/celsworth/lxp-bridge/commit/dbd093310d1f9eb7b4083c1dd12f868dd63d4ece

Would you be able to try this and verify it has the intended behaviour?

NicLogix, 2023-10-07:
I am by no means s coder, but I can copy and paste.
I tried this version, but it lists a radio button to select, which adds the whole list.
Config cannot be saved, as the yaml does not compile.
Should it not look like this:

"schema": {
    "log_level": "list(trace|debug|info|notice|warning|error|fatal)?",
  
Credit to https://github.com/zhloba/hassio-wireguard/tree/main

Let me know when it is changed, I will happily test again.
I run a backup every day, so to rollback is a breeze.

celsworth, 2023-10-07:
Ah sorry, yes you're right, it should use `|` and not `,`.

I've made the change, would you mind trying again?

Thanks!

NicLogix, 2023-10-08:
Great!
Now working as intended.
GUI gives a drop-down to select, and yaml compiles 100%.

Thank you again for a great piece of software.

I am an electrician, and install about 3 luxpowers per month.  Lxp-bridge gives me peace of mind that if the lux cloud disappears, I can migrate my clients to something else.

Keep up the good work.

celsworth, 2023-10-09:
Awesome, thanks for confirming!

Glad to hear lxp-bridge is useful to you, that's exactly why I first started tinkering with this stuff, having ability to tweak things locally without relying on China :)

closed 2023-10-09 by celsworth