celsworth/lxp-bridge #206: [Question/Not an issue] Can we disable the tcp client altogether?

ishdemon, 2023-10-11:
Hi, can we disable the tcp client altogether & only keep the tcp server running? also how come the local mode works with app..without those heartbeat acknowledgment..I am yet to get my hands on this inverter, so was curious.

celsworth, 2023-10-11:
In short, yes you can, although you need to enable the heartbeats configuration flag as you've probably seen. This is because the inverter won't send any data out until it gets a response from the other side of its TCP connection that something is listening and wants data from it.

The app works because it sends heartbeats of its own I think.

The only thing to be aware of is the wifi dongle can be quite persistent in trying to add back the TCP Client IP, so you might have to persist or find the right magic sequence of saving it. An alternative is just to firewall off the inverter, or remove its default gateway or similar sort of thing.

NicLogix, 2023-10-13:
D

Nic du Plessis
Logic Services & Consulting
0729871980