celsworth/lxp-bridge #207: Force Discharge timeslots only updated upon restart

raldred, 2023-10-12:
Hi, I've noticed that if values for discharge timeslots when changed the lux app/website don't get published to home assistant.
Other options do, seems to be list the timeslots.

I changed the endtime1 to 18:10 on lux website, it didn't change and remains at 18:59 on home assistant.

[image]

[image]

I think this is the package coming in when it gets saved....

```
[2023-10-12T14:51:11.210 DEBUG lxp_bridge::lxp::packet_decoder] 38 bytes in: [161, 26, 5, 0, 32, 0, 1, 194, 66, 65, 50, 52, 55, 48, 49, 51, 54, 48, 18, 0, 1, 6, 51, 48, 50, 50, 51, 53, 48, 49, 48, 54, 85, 0, 18, 10, 46, 119]
[2023-10-12T14:51:11.211 DEBUG lxp_bridge::coordinator] RX: TranslatedData(TranslatedData { datalog: <REDACTED>, device_function: WriteSingle, inverter: <REDACTED>, register: 85, values: [18, 10] })
[2023-10-12T14:51:11.211 INFO lxp_bridge::mqtt] publishing: lxp/<REDACTED>/hold/85 = 2578
[2023-10-12T14:51:11.211 DEBUG rumqttc::state] Publish. Topic = lxp/<REDACTED>/hold/85, Pkid = 7, Payload Size = 4
```

Any ideas?

celsworth, 2023-10-12:
I think this is a limitation of the current implementation, so it's working as expected at the moment.

However it would be nice to improve this. The problem is for HA to update this field, it needs to see the message topic `lxp/<serial>/forced_discharge/1`. But we only send these messages when we initially connect to the inverter. It's actually marked as a FIXME for improvement here: https://github.com/celsworth/lxp-bridge/blob/b90d0edd0b248b9275c5e2b617030877e3f90895/src/coordinator/mod.rs#L451-L477

I think a better solution would actually be to recognise the appropriate registers in the more generic "we got a holding register's details" code, such as is done for register 21 here: https://github.com/celsworth/lxp-bridge/blob/b90d0edd0b248b9275c5e2b617030877e3f90895/src/mqtt.rs#L43-L50 - and then construct and publish the `lxp/<serial>/forced_discharge/1` topic (and others as appropriate)

I might have a go at doing this over the weekend. Might just need some refactoring of the message creation for these special types as its a bit hidden away at the moment at https://github.com/celsworth/lxp-bridge/blob/b90d0edd0b248b9275c5e2b617030877e3f90895/src/coordinator/commands/time_register_ops.rs#L93-L101

Also tagging @lupine as the original creator of some of this code in case of any better ideas or thoughts :)

celsworth, 2023-10-12:
@lupine Would we better splitting up the HA time-range widget into completely separate start/end times? Is that possible? Would that simplify pretty much everything about this integration (as well as making the above feasible)?

celsworth, 2023-10-12:
Yeah I'm inclined to agree that splitting them into single times makes sense.

I'm unable to find which MQTT device class to use for a time input though, do you happen to know? I'm looking at the list at https://www.home-assistant.io/integrations/#search/mqtt but not seeing it there. I think the existing time range ones just use a text type but then we don't get any special UI handling.

lupine, 2023-10-12:
:wave: 

There isn't a specific MQTT type for time. Text box is the best you get.

I put start and end time in the same text box so that edge cases like setting a start time later than an end time were easier to handle - I don't have a super-strong opinion though. Just remember that we need to delete the MQTT retained message for the old setting if we're introducing a new setting that replaces it.

IIRC I was hoping for https://github.com/celsworth/lxp-bridge/issues/151 to fix this - i.e. we'd publish all holding registers, and all the combinations thereof, whenever one of them changed, and ha could subscribe to that for updates. In the end, I've stopped using home-assistant, so don't have much of a motivation to work on this at the moment :/

celsworth, 2023-10-28:
Just to circle back to this with an update.

So I do agree it should be done, but unfortunately it's pretty low priority right now for a few reasons (not much time to work on this, the fix isn't trivial, and it's a bit of a niche case that doesn't occur too often, hopefully) so I'm not sure when I'll get to it.

raldred, 2023-10-28:
Totally understand. Thanks for taking the time to consider.

PR #261 (2024-03-15): Improve time register handling
Aiming to send new MQTT messages with JSON start/end so HA can be updated when registers change externally.

Fixes #207