celsworth/lxp-bridge #210: Not always I'm getting data

teicors, 2023-10-17:
Hi Chris,
how are you ? I hope as myself : well !!
I have another strange issue: let me explain it with the logs:
grep publishing lux-stderr---supervisor-b3ibbghj.log | cut -f 1 -d {

[2023-10-17T13:03:39.453 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/3 = 
[2023-10-17T13:05:57.928 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/2 = 
[2023-10-17T13:05:59.072 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/3 = 
[2023-10-17T13:07:24.330 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/1 = 
[2023-10-17T13:07:25.429 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/2 = 
[2023-10-17T13:07:26.760 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/all = 
[2023-10-17T13:07:26.760 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/3 = 
[2023-10-17T13:08:37.236 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/1 = 
[2023-10-17T13:08:38.337 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/2 = 
[2023-10-17T13:08:39.440 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/all = 
[2023-10-17T13:08:39.440 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/3 = 
[2023-10-17T13:10:58.179 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/2 = 
[2023-10-17T13:10:59.323 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/3 = 
[2023-10-17T13:12:23.644 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/1 = 
[2023-10-17T13:12:24.729 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/2 = 
[2023-10-17T13:12:25.832 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/all = 
[2023-10-17T13:12:25.832 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/3 = 
[2023-10-17T13:13:39.248 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/1 = 
[2023-10-17T13:13:40.351 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/2 = 
[2023-10-17T13:13:41.493 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/all = 
[2023-10-17T13:13:41.493 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/3 = 
[2023-10-17T13:16:00.434 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/2 = 
[2023-10-17T13:16:01.539 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/3 = 
[2023-10-17T13:16:28.382 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/3 = 
[2023-10-17T13:16:34.379 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/3 = 
[2023-10-17T13:17:26.070 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/2 = 
[2023-10-17T13:17:27.168 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/3 = 
[2023-10-17T13:18:40.399 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/2 = 
[2023-10-17T13:18:41.503 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/3 = 
[2023-10-17T13:21:01.434 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/2 = 
[2023-10-17T13:21:02.537 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/3 = 
[2023-10-17T13:22:25.377 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/2 = 
[2023-10-17T13:22:26.474 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/3 = 
[2023-10-17T13:23:40.407 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/2 = 
[2023-10-17T13:23:41.547 INFO lxp_bridge::mqtt] publishing: lxp/BA10430020/inputs/3 = 
[2023-10-17T13:26:01.749 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/2 = 
[2023-10-17T13:26:02.745 INFO lxp_bridge::mqtt] publishing: lxp/BA10430568/inputs/3 = 
[2023-10-17T13:27:24.681 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/2 = 
[2023-10-17T13:27:25.791 INFO lxp_bridge::mqtt] publishing: lxp/BA10430542/inputs/3 = 

I have three inverter, but most items of the first input are missed: do you have any suggestion? I pull the sources, refreshed rust and cargo, but only lxp/XXX/inputs/1 is missed.

Cheers,
Angelo

teicors, 2023-10-17:
hi,
looking at the log I found something interesting:

thread 'main' panicked at src/lxp/packet.rs:817:31:
attempt to subtract with overflow
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[2023-10-17T13:31:01.770 INFO lxp_bridge] lxp-bridge 0.12.0 starting

thread 'main' panicked at src/lxp/packet.rs:817:31:
attempt to subtract with overflow
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[2023-10-17T13:36:04.015 INFO lxp_bridge] lxp-bridge 0.12.0 starting

thread 'main' panicked at src/lxp/packet.rs:817:31:
attempt to subtract with overflow
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[2023-10-17T13:56:06.879 INFO lxp_bridge] lxp-bridge 0.12.0 starting

thread 'main' panicked at src/lxp/packet.rs:817:31:
attempt to subtract with overflow
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[2023-10-17T14:01:07.126 INFO lxp_bridge] lxp-bridge 0.12.0 starting

and more...
May this helps you toi find the root cause ?

BR,
Angelo

celsworth, 2023-10-17:
😱 looks like it could be related to the recent change to convert values to unsigned integers (#196) but I thought I added sufficient tests for this kind of thing. I'll take another look this evening :)

Do you happen to know if your inverter is charging or discharging the battery when these crashes occur? The problematic line is https://github.com/celsworth/lxp-bridge/blob/60436373a0bfd01d26b9cbb555f0eeafa6f87d6c/src/lxp/packet.rs#L817

So it'll be related to an overflow in that maths somehow. Hopefully not too hard to reproduce later.

teicors, 2023-10-17:
Hi,
with
grep BA10430568/inputs/1 lux-stderr---supervisor-b3ibbghj.log | grep DEBUG
I'm getting
[2023-10-17T11:40:42.787 DEBUG rumqttc::state] Publish. Topic = lxp/BA10430568/inputs/1, Pkid = 37, Payload Size = 678
[2023-10-17T13:41:03.236 DEBUG rumqttc::state] Publish. Topic = lxp/BA10430568/inputs/1, Pkid = 47, Payload Size = 676
and in this time I'm only discharging the battery 
es.
p_charge = 0
and
p_discharge = 1224
or
p_discharge = 2619
or also
p_discharge = 169
And I have also more examples with always p_charge=0 and a number in p_discharge.

BR,
Angelo

teicors, 2023-10-17:
Hi, 
latest data in the db is related at 16:41 and the p_charge is 0 and the same for p_discharge.
after this time the p_charge is 0 and the p_discharge as a value: it seems really a rust bug during the conversion or a wrong interpretation of the value from inverter.

BR,
Angelo

celsworth, 2023-10-17:
I made a commit to master, are you able to try the `dev` docker tag once it updates?

(ps: I recently made it possible to install the dev version as a HA addon so refresh the HA repo if you use that)

closed 2023-10-17 by celsworth

teicors, 2023-10-17:
Hi, I installed the patch and It seems tò work as excpected:
2023-10-17T16:41:32.256 DEBUG rumqttc::state] Publish. Topic = lxp/BA10430568/inputs/1, Pkid = 66, Payload Size = 673
[2023-10-17T21:52:16.434 DEBUG rumqttc::state] Publish. Topic = lxp/BA10430568/inputs/1, Pkid = 41, Payload Size = 676
[2023-10-17T21:57:18.477 DEBUG rumqttc::state] Publish. Topic = lxp/BA10430568/inputs/1, Pkid = 53, Payload Size = 681
[2023-10-17T22:02:18.535 DEBUG rumqttc::state] Publish. Topic = lxp/BA10430568/inputs/1, Pkid = 66, Payload Size = 6

Tomorrow I'll check Better.

BR,
Angelo

PR #211 (2023-10-17): Hopefully fix i16 overflow in maths
Hopefully fixes #210