celsworth/lxp-bridge #214: Docker Compose Example

JohnMacrae, 2023-10-26:
The contents of the docker compose example (https://github.com/celsworth/lxp-bridge/wiki/docker-compose.yaml.example) seem more like the config.yaml. Is there another version available please?

celsworth, 2023-10-26:
Hm yes think someone might have had a fat finger edit of that, I'll fix it up :)

But there's an actual example file in the repo at https://github.com/celsworth/lxp-bridge/blob/master/docker-compose.yaml.example

JohnMacrae, 2023-10-26:
You're right - thank you :)

(I'm an idiot)

JohnMacrae, 2023-10-26:
FWIW, I store my config files in /opt/lxp-bridge/config.yaml. I've been unable to map source and target to allow lxp-bridge to run under docker compose. When run manually, it starts just fine from the /opt/lxp-bridge directory. Any hints much appreciated.

celsworth, 2023-10-26:
I think it should just be this?

```
   volumes:
   - type: bind
     source: /opt/lxp-bridge/config.yaml
     target: /etc/config.yaml
```

All we're doing is telling Docker to make the file that's at `/opt/lxp-bridge/config.yaml` in the host, and make it available at `/etc/config.yaml` in the container (this is just where lxp-bridge looks for its config by default, which is a pretty bad default but thats another story).

If that's not working you'll have to give more details, what's the error? Is lxp-bridge not able to read it?

JohnMacrae, 2023-10-26:
Yes - that's exactly what I had set up but consistently get the error: 2023-10-26T16:53:22.760124409Z Error: error reading /etc/config.yaml: No such file or directory (os error 2)

Weird.

My Compose file:

  lxp-bridge:
    container_name: lxp-bridge
    image: celsworth/lxp-bridge:latest
    restart: unless-stopped
    init: true
    volumes:
      - type: bind
        source: /opt/lxp-bridge/config.yaml
        target: /etc/config.yaml

celsworth, 2023-10-26:
Hm I can't reproduce this. If I have a config file at that path and use this docker-compose, it works:

```
version: '3.4'

services:
 lxp-bridge:
   container_name: lxp-bridge
   image: celsworth/lxp-bridge:latest
   restart: unless-stopped
   init: true
   volumes:
     - type: bind
       source: /opt/lxp-bridge/config.yaml
       target: /etc/config.yaml
```

Since I'm running it on Docker Desktop for Mac I did have to add /opt to the list of allowed filepaths in the Preferences, but other than that its fine..

jmrush97, 2023-10-26:
Chris,

Just a thought , but be careful about permissions on the file areas
especially with Mac's . It could be that the docker process does not have
permission to read the location being used.

John