celsworth/lxp-bridge #219: AC Charge Switch seems to be 'read only'

JohnMacrae, 2023-11-11:
At some time in the last few weeks, the AC charge switch (and the discharge switch) seem to have become 'read only' in HA. What I mean by that is that they correctly reflect the state of the switch in the Lux app, but when activated, they switch on or off but then revert after a few seconds.  All other status information is correct. Can anyone point me in the right direction please?

JohnMacrae, 2023-11-11:
It seems that, as usual, I found a solution immediately after posting the problem. I restarted the docker container (for the nth time) and this time is seems to have worked. Who knows for how long.

closed 2023-11-11 by JohnMacrae

celsworth, 2023-11-11:
Hm intriguing. Let us know if it happens again, could be good to get some logs to see what's going on. My best guess is lxp-bridge had lost it's subscription to MQTT (when you press that toggle in HA, it sends an MQTT message that lxp-bridge has to receive in order to action the request). So the first step would be "does that appear in the logs".