celsworth/lxp-bridge #225: Home Assistant: error loading config

dbainbridge, 2023-12-16:
Just installed fresh copy of HA and also added lxp-bridge but get this error when trying to start it up.  Also note the log level is listed as INFO even though I have the config set as TRACE.

```
[20:24:31] INFO: Creating lxp-bridge config from options...
Done.
[20:24:31] INFO: Starting lxp-bridge...
Error: inverters[0]:  must be exactly 10 characters at line 3 column 5
```

```
loglevel: trace
inverters:
  - enabled: true
    host: 192.168.20.223
    port: 8000
    serial: ""
    datalog: ""
    heartbeats: false
    publish_holdings_on_connect: false
databases:
  - enabled: false
    url: postgres://lxp:lxp@localhost/lxp
mqtt:
  enabled: true
  host: localhost
  port: 1883
  username: ""
  password: ""
  namespace: lxp
influx:
  enabled: false
  url: http://localhost:8086
  username: ""
  password: ""
  database: lxp

```

celsworth, 2023-12-17:
It looks like a bit of crucial detail is missing in that error, it should be referring to the serial/datalog - they need to be 10 characters for the config parser to give it a pass.

So currently you have:

```
    serial: ""
    datalog: ""
```

They need 10 chars in the strings. I think the documentation might actually say you don't need to set this if you don't want to control the inverter but thats obviously wrong, I'll see if I can improve that actually. The correct advice should probably be "if you don't want to control the inverter, just monitor it, you can put ten zero's here" or 10 of anything really.

As to the logging level, so you've set `trace` which means you'll get full debugging, but those lines having `INFO` just means those particular lines are logged at INFO level.

nemirow2, 2024-02-17:
I installed lxp-bridge - and when filling in the serial number of the inverter, an error pops up; if you change one digit to a letter, it is saved - what could it be? (I enter the real serial numbers from the inverter
[image]

celsworth, 2024-02-17:
That looks like it's  because it's interpreting it as a number not a string (no letters in it). just surrounding the value with quotes should fix it :)

nemirow2, 2024-02-17:
Thanks for the quick response!!! It’s just that
sensor.lxp_ba35204810_active_eps_power always shows only 0 in the ON,
although in the application the power consumption is always shown well
- I thought this was the problem... maybe you know what this problem
could be? (I put the serial number in quotes - the configuration is
saved)

celsworth, 2024-02-19:
EPS power being 0 I think is normal when AC is present. I don't think it has a way to measure it since its basically bypass.

If you yank AC power so it starts acting like a UPS then I think it starts to show EPS power.